ZoneMinder/zoneminder #1560: first class JS calendar

onlyjob, 2016-07-13:
It would be great if eventually we could find a better calendar than abandoned/unmaintained _jscalendar_. Anything really... Some alternatives were [suggested here](https://github.com/ZoneMinder/ZoneMinder/pull/1429#issuecomment-215939963).

alager, 2016-08-25:
Sorry for my ignorance, but I couldn't find the page/area where there are dates to pick.  Where is it used?

connortechnology, 2016-09-01:
Normally on the date fields in the filters.  Only shows up if jscalendar is available.

alager, 2016-09-01:
Yeah, mine (1.29.0) is just a text input field:
[image]
`<input name="filter[terms][0][val]" id="filter[terms][0][val]" value="now">`

onlyjob, 2018-06-16:
No activity doesn't mean that the problem is gone.

knight-of-ni, 2018-07-11:
ZoneMinder is using jquery-ui-timepicker for filter calendar functions now which effectively resolves this issue.

closed 2018-07-11 by knight-of-ni

alager, 2018-07-11:
@knight-of-ni What version was this fixed in?  I'm on 1.31.1 and it's still a text input field.
[image]

connortechnology, 2018-07-11:
Please update to latest master for all kinds of wonderful updates.

This new timepicker was added in this commit: https://github.com/ZoneMinder/zoneminder/commit/d7532b0340b323d1f867e5627bdeb60e69eb3675 

It was incorporated into my tree and only merged to master a few months ago.