ZoneMinder/zoneminder #1569: Error when install ZoneMinder

eduardomunizbarreto, 2016-07-29:
Hi,
i got this error:  Error: Package: Zoneminder-1.30.0-3.el6.x86_64 (zm repo) Requires: perl (Data :: Dump), when i execute this command: sudo yum install zoneminder.

I may have forgot something?

knnniggett, 2016-07-29:
The package perl-Data-Dump is missing from zmrepo.
You can wait until I've got a moment to figure out why and fix it, or you can head on over to the EPEL 6 repo and download and installed it manually (it is the exact same package).  

https://fedoraproject.org/wiki/EPEL#What_packages_and_versions_are_available_in_EPEL.3F

knnniggett, 2016-07-29:
The missing package has been successfully synced to all the zmrepo mirrors.
You either need to wait for your local Yum cache to expire (min 6 hours), or clear it out yourself followed by a normal "yum update".

closed 2016-07-29 by knnniggett