ZoneMinder/zoneminder #1584: build failed

zhuykovkb, 2016-08-07:
I following next steps:

```

ubuntu@vm-kzhuykov:~/dev$ git clone https://github.com/ZoneMinder/ZoneMinder.git ZMT
Cloning into 'ZMT'...
remote: Counting objects: 62294, done.
remote: Compressing objects: 100% (44/44), done.
remote: Total 62294 (delta 17), reused 0 (delta 0), pack-reused 62249
Receiving objects: 100% (62294/62294), 50.13 MiB | 1.65 MiB/s, done.
Resolving deltas: 100% (45555/45555), done.
Checking connectivity... готово.

ubuntu@vm-kzhuykov:~/dev$ cd ZMT
ubuntu@vm-kzhuykov:~/dev/ZMT$ git submodule update --init --recursive
Submodule 'web/api/app/Plugin/Crud' (https://github.com/FriendsOfCake/crud.git) registered for path 'web/api/app/Plugin/Crud'
Cloning into 'web/api/app/Plugin/Crud'...
remote: Counting objects: 47112, done.
remote: Compressing objects: 100% (32/32), done.
remote: Total 47112 (delta 21), reused 0 (delta 0), pack-reused 47078
Receiving objects: 100% (47112/47112), 10.49 MiB | 4.72 MiB/s, done.
Resolving deltas: 100% (34601/34601), done.
Checking connectivity... done.
Submodule path 'web/api/app/Plugin/Crud': checked out 'c3976f1478c681b0bbc132ec3a3e82c3984eeed5'

ubuntu@vm-kzhuykov:~/dev/ZMT$ docker build --no-cache .
Sending build context to Docker daemon 33.52 MB
Step 1 : FROM ubuntu:trusty
 ---> 0ccb13bf1954
Step 2 : MAINTAINER Kyle Johnson <kjohnson@gnulnx.net>
 ---> Running in 55b531aa641c
 ---> b41ed7ef6d23
Removing intermediate container 55b531aa641c

...................................................................................................

Step 7 : RUN make && make install
 ---> Running in 352b393100ea
Scanning dependencies of target zm
[  2%] Building CXX object src/CMakeFiles/zm.dir/zm_box.cpp.o
[  4%] Building CXX object src/CMakeFiles/zm.dir/zm_buffer.cpp.o
[  6%] Building CXX object src/CMakeFiles/zm.dir/zm_camera.cpp.o
[  8%] Building CXX object src/CMakeFiles/zm.dir/zm_comms.cpp.o
[ 10%] Building CXX object src/CMakeFiles/zm.dir/zm_config.cpp.o
[ 12%] Building CXX object src/CMakeFiles/zm.dir/zm_coord.cpp.o
[ 14%] Building CXX object src/CMakeFiles/zm.dir/zm_curl_camera.cpp.o
[ 16%] Building CXX object src/CMakeFiles/zm.dir/zm.cpp.o
[ 18%] Building CXX object src/CMakeFiles/zm.dir/zm_db.cpp.o
[ 20%] Building CXX object src/CMakeFiles/zm.dir/zm_logger.cpp.o
[ 22%] Building CXX object src/CMakeFiles/zm.dir/zm_event.cpp.o
[ 24%] Building CXX object src/CMakeFiles/zm.dir/zm_exception.cpp.o
[ 26%] Building CXX object src/CMakeFiles/zm.dir/zm_file_camera.cpp.o
[ 28%] Building CXX object src/CMakeFiles/zm.dir/zm_ffmpeg_camera.cpp.o
/ZoneMinder/src/zm_ffmpeg_camera.cpp: In member function 'int FfmpegCamera::OpenFfmpeg()':
/ZoneMinder/src/zm_ffmpeg_camera.cpp:241:67: error: 'av_dict_parse_string' was not declared in this scope
   ret = av_dict_parse_string(&opts, Options().c_str(), "=", ",", 0);
                                                                   ^
make[2]: *** [src/CMakeFiles/zm.dir/zm_ffmpeg_camera.cpp.o] Error 1
make[1]: *** [src/CMakeFiles/zm.dir/all] Error 2
make: *** [all] Error 2
The command '/bin/sh -c make && make install' returned a non-zero code: 2
ubuntu@vm-kzhuykov:~/dev/ZMT$
```

anyone can test this?

knnniggett, 2016-08-07:
You appear to be running ubuntu, but we don't know what version, nor do we know the version of ffmpeg installed on your system, which is key to understanding the cause of the problem.

What we can say is the version of ffmpeg installed on your system does not contain the av_dict_parse_string function call

connortechnology, 2016-08-07:
@knnniggett you are so fast!

So it builds fine for me on 16.04 with standard ffmpeg.  
14.04 exhibits this behaviour which has ffmpeg 2.4.3

connortechnology, 2016-08-07:
Looks like we are going to have to copy this code from newer ffmpeg. I have a PR coming.

knnniggett, 2016-08-07:
Well, I was getting bored watching another project build in one of my vm's. Like watching grass grow, so I thought I'd answer forum questions.

connortechnology, 2016-08-07:
Uh... thought this would be simple, but I am grabbing too much code from ffmpeg.

zhuykovkb, 2016-08-07:
@knnniggett i'm using that Dockerfile from the project and run:

```
ubuntu@vm-kzhuykov:~/dev/ZMT$ docker build --no-cache .
Sending build context to Docker daemon 33.52 MB
Step 1 : FROM ubuntu:trusty
```

it's trusty!

But, if i set HEAD to commit 004b651d61842a7f95fa2d9815c7595c1ae876db
i have a succesfull build!

knnniggett, 2016-08-07:
@zhuykovkb I'm not sure any of us here use the dockerfile. It has largely been maintained by the community.

PR #1585 (2016-08-14): Fixes #1584. I've just copied the relevant functions from ffmpeg sour…
Fixes #1584. I've just copied the relevant functions from ffmpeg source.  Please review carefully before merging, as I may have screwed up the version check or something.
commit 177c2c74 (2016-08-14): Fixes #1584. I've just copied the relevant functions from ffmpeg source.  Please review carefully before merging.
commit cd55c8b2 (2016-08-14): Merge pull request #1585 from ZoneMinder/copy_av_dict_parse_string_from_ffmpeg

Fixes #1584. I've just copied the relevant functions from ffmpeg sour…
closed 2016-08-14 by knnniggett