ZoneMinder/zoneminder #1588: ONVIF authentication for profiles command not working with Trendnet IP862-IC

connortechnology, 2016-08-10:
Probe finds the camera, but the profiles command gives an auth error.

connortechnology, 2016-08-10:
@altaroca 

/usr/bin/zmonvif-probe.pl -v profiles http://192.168.9.247/onvif/device_service 1.1 admin password
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"><faultcode>SOAP-ENV:Server</faultcode><faultstring>Error deserializing message: Bad top node html at /usr/share/perl5/ONVIF/Deserializer/MessageParser.pm line 89.
 at line 1 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 82.
. 
Message was: 
&lt;html&gt;
&lt;body&gt;
&lt;h1&gt;Unauthorized&lt;/h1&gt;
Please enter correct account/password.
&lt;/body&gt;
&lt;/html&gt;
</faultstring><faultactor>urn:localhost</faultactor></Fault>

knnniggett, 2016-08-10:
We need to poke @altaroca to get his attention.

connortechnology, 2016-08-10:
I did.

altaroca, 2016-08-10:
The obvious first question is: Are you sure you are using the correct credentials?
Some thoughts: 
- the camera might have different sets of credentials for web ui and for onvif.
- the camera might need a password change on the very first login.
- can you authenticate with the same credentials using some other onvif tool?

connortechnology, 2016-08-10:
So far I havn't found a tool that will speak to any of my cameras.  

The lingodigit onvif tool detects my DCS-2310L at least.  

Yes I am sure about the password.  I have also tried the device's default password.

knnniggett, 2016-08-10:
Try using onvif device manager. It should work with that tool if the camera is truly onvif compliant.
https://sourceforge.net/projects/onvifdm/

If it doesn't work with onvif device manager then the camera may not be onvif compliant, despite any marketing claims to the contrary.

connortechnology, 2016-08-10:
onvifdm just hangs in my vm.

knnniggett, 2016-08-10:
Make sure the real network card you bind your vm to has promiscuous mode enabled under the vm settings.

connortechnology, 2016-08-10:
Ok, so odm sees both cameras, and the auth works with the dcs-2310L. Everything works with it.

Auth does not work for the TV-IP862-IC under odm, using either the set credentials or the devices defaults. (Defaults are supposed to be admin/admin.).

SteveGilvarry, 2016-08-10:
http://www.onvif.org/ConformantProducts/ProfileProducts.aspx
Not very much listed here under Trendnet and to claim conformance they should be.

knnniggett, 2016-08-10:
The Onvif standard allows one to browse for cameras on the local network, with no authentication, but everything following that requires it. That of course explains why the first step of the onvif discovery probe worked.  When one digs into how one authenticates via onvif, one encounters a time element which is needed to create the nonce. So if the clock in the camera is not set or is not close to the time on the machine you are on, then, by design, the authentication will not work.  e.g. check the camera clock

knnniggett, 2016-10-04:
@connortechnology Did you make any onvif changes today that might address this?

connortechnology, 2016-10-05:
No.   This camera responds intermittently and has auth problems.   Nowhere does it say that it is onvif compliant... so... this is one of those odd cameras that we may never support.

knnniggett, 2017-05-18:
Closing all the Onvif issues out since the latest release contains improvements to the Onvif probe. Please create a new issue if problems persist.

closed 2017-05-18 by knnniggett