ZoneMinder/zoneminder #1590: http cURL source with no content-size header fails

wilsonwaters, 2016-08-11:
Hi all,
I have a camera source which provides a jpeg snapshot through an http URL. Unfortunately it doesn't include the content-size http header with the data.
Zoneminder gives the following error when trying to read 

> Unable to get the size of the image

I had a quick look through zm_curl_camera.cpp and it seems after reading the data you're trying to find the size by calling `curl_easy_getinfo(c, CURLINFO_CONTENT_LENGTH_DOWNLOAD, &dSize);`. I suggest you should use `databuffer.size()` instead.

I don't know the codebase well enough so there's probably a good reason you do it this way, but it would be nice if we could have support for curl based sources which don't sent the content-size header.

Thanks!

connortechnology, 2016-08-17:
I'm kindof curious what happens if you just use the remote http monitor type.

I'm not sure if we are going to maintain the curl type.  It was an interesting experiment, which ultimately did not solve the problem we were trying to solve.

That being said... databuffer.size may not be the entire image, as in, it may not all have arrived yet.  The logic was really not designed to handle this case, although obviously it should be.

What camera make/model is this?  Would you be able to expose it to me for testing?

connortechnology, 2016-08-17:
No worries, I have that app.  I'll do some testing when I find a moment.

mattdurant, 2016-09-01:
The problem is that without Content-Length, you have no choice but to read until the other end closes the socket.  In ReadData() of zm_remote_camera_http.cpp around line 193, it's assumed that if there is no data to read that it's because of an exception with the socket, no because the other end is finished sending the response in the case of a single image.

mattdurant, 2016-09-01:
I tested this before and after this PR, using the same app (I use it personally as well).  This fix resolved the issue for me.

wilsonwaters, 2016-09-01:
Nice. Thanks @mattdurant. I'll see if I can work out how to build Zoneminder and test it sometime soon.

Just had a quick look through the change. Would it be safer to test if the actual read returns 0, rather than what ioctl(FIONREAD) says is currently available on the socket?
Could there be a situation where the socket is open but the remote end doesn't immediately start writing data? Or if there is a pause half way though the transfer for some reason, wouldn't this cause the image to be truncated?

I'm not being very helpful sorry, I should build Zoneminder and test it myself but just wanted to mention it now.

mattdurant, 2016-09-01:
The request for bytes available will block until bytes ARE available, or if the socket is closed, it will return 0 because no bytes will ever be available to read.  Pauses or delays in network traffic (as long as the connection is still valid) will not cause this to fail.

wilsonwaters, 2016-09-01:
yea, but ioctl isn't a blocking call.

mattdurant, 2016-09-01:
No, but the select above it is a synchronous call that should block until data is available or something happened to the socket.  The logic was already there to return an error if the socket closed, but was assuming behavior as if the stream had a Content-Length and the socket closing was unexpected.  The check I added for a single image is needed to ignore remote socket closings because it's expected behavior if no Content-Length is available that the remote side closes the socket when it's finished.  It may not be the optimal solution, but it just piggybacked on what was already there and shouldn't interfere with any normal operation or cause any issues that I can see.

wilsonwaters, 2016-09-01:
oh yea, there's a select just above! Missed that. My bad.
All looks sensible to me now. Thanks again for the fix.

connortechnology, 2017-02-11:
The fix merged for this is bogus and breaks a lot of cameras. We will have to keep working on this.

mastertheknife, 2017-02-11:
Back at the day, i used the curl's function to get the content-length because it seemed easier and wiser to let curl get the size for us.
I suppose we can add a check that if it fails, we will use the Content-Length header instead.

> I'm not sure if we are going to maintain the curl type. It was an interesting experiment, which ultimately did not solve the problem we were trying to solve

AFAIK the current "Remote" source type doesn't support SSL (HTTPS), so i wouldn't go removing it yet :)

mattdurant, 2017-02-11:
Have you pinpointed where it was blocking infinitely or looping? What
make/model cameras are you seeing it with?

mastertheknife, 2017-02-11:
@mattdurant 
> Have you pinpointed where it was blocking infinitely or looping? What
make/model cameras are you seeing it with?

I was refering to the "cURL" source type.

Regarding the "Remote" source code, i think we can safely continue without a content-length in case of a single image, because the socket will simply close right after (we might need to send "Connection: Close" header if the server requested keep-alive)
I am also not sure why there is a timeout implemented. Without the timeout, the select() could be substituted with read() which will block for data, and return 0 when the socket closed, slightly simpler.
Getting the current buffer size with ioctl could also be removed. In case there is no Content-Length header, we just need a loop to read from the socket until it returns zero (closed) or less than zero (error)

PR #1608 (2016-09-01): Added check for SINGLE_IMAGE to ignore socket closed by remote side
The issue is that with SINGLE_IMAGE requests, if a Content-Length is not specified in the headers on the response, appropriate behavior is to read until the socket is closed by the camera.  Resolves #1590
closed 2018-06-22 by stale[bot]