ZoneMinder/zoneminder #1595: Authentication via LDAP / AD

peruchi, 2016-08-17:
it would be interesting to integrate with LDAP / AD, making bind, filters for groups, is a suggestion.

knnniggett, 2016-08-17:
Per our posting guidelines, we only accept feature requests where you are willing to either a) write the code yourself or b) assign a sizeable bounty to the feature request to pay someone else to do the work.

Unless you plan on meeting one of those criteria soon, I am going to close this issue.

This does not mean we do not appreciate your idea. We get a lot of feature requests, but most won't get worked on due to lack of manpower.

peruchi, 2016-08-17:
OK, you can close, I am studying the language and try to carry out this implementation and if successful, will make a pull request.

closed 2016-08-17 by peruchi

knnniggett, 2016-08-17:
We are in the #zoneminder irc channel daily. That would be the best place to get to know the development team and ask code related questions.