ZoneMinder/zoneminder #1597: PurgeWhenFull deletes too many events as zmaudit can't delete fast enough when OPT_FAST_DELETE enabled

belidzs, 2016-08-21:
I'm using Ubuntu 16.04.1LTS with ZoneMinder 1.30.0.

Consider the following scenario:
  1. `OPT_FAST_DELETE` is enabled
  1. `zm_audit.pl` is running daemonized as per default config
  1. Disk has only 5% free space left
  1. Default `PurgeWhenFull` filter kicks in, deleting the 100 oldest event
  1. `zmaudit.pl` doesn't delete the files immediately (this might be the bug)
  1. `PurgeWhenFull` runs again, deleting another 100 event until disk space eventually freed up

I myself lost around 600 events because of this.

As a workaround, I have disabled `OPT_FAST_DELETE`. I probably might also use cron to run `zm_audit.pl` instead of daemonizing it, but I think the root of the problem should be addressed instead.

If you need any data reproducing the issue please let me know!

mattdurant, 2016-09-01:
The PurgeWhenFull filter runs in the background the delete, if OPT_FAST_DELETE is enabled only deletes the database records involved.  Only when zmaudit runs does it match the missing records to image files that need to be delete and they are removed from disk.  The problem is that the current method of deleting those images takes a really long time, hence the need for OPT_FAST_DELETE in the first place.

I've seen a solution using rsync to delete files instead that operates MUCH faster, even on a system with hundreds of thousands of events and even more files.  Incorporating that change would mean less reliance on OPT_FAST_DELETE and that each filter run would cause a change in the disk space instead of only when zmaudit runs.

connortechnology, 2016-09-01:
I have seen that rsync thing too... meant to look more into it. 

The whole OPT_FAST_DELETE thing is a bad idea.  A very bad idea.  It needs to go away.

My recommendation is to turn it off and run zmaudit less frequently.

In future, we will be deleting fewer files because each event will consist of a single mp4 and some snapshot jpgs and thumbnails.

mattdurant, 2016-09-01:
I've used it in practice in a few installations that had some issues with recording new events faster than could be deleted and it performed very well.  I can make a pull request if you guys want to test it out?

I know we're moving away from stored individual frames, but it wasn't but a few lines that had to be changed and it was orders of magnitude faster than what we have now.

connortechnology, 2016-09-01:
I'd like to see it/test it.

jamiemaher, 2016-10-19:
Just lost all my events with OPT_FAST_DELETE turned on, during a very stormy day generating lots of events, with my purge filter set to 96% disk used.

The filesystem filled up completely and I had to stop zoneminder and then manually run zmaudit.pl to cleanup some events to free up some disk space and bring the disk utilization down.

I adjusted the AUDIT_CHECK_INTERVAL down to about 240 secs from the default of 900 seconds (15 minutes) to keep up with the data being saved, otherwise it was constantly loosing ground.

knnniggett, 2016-10-19:
Part of this argument is flawed, and won't be fixed since there is nothing to fix.

There is no such thing as a PurgeWhenFull filter that works for every system. Bump it too high and it deletes too much (and we get bug reports). Keep it too low and it doesn't keep up (and we get bug reports).

The fact is that it is working exactly as intended. However, it must be tuned for each system it is running on. The end user has to monitor their system to make sure it is deleting the right amount of events within the check interval. The simplest thing to do would be to avoid messing with AUDIT_CHECK_INTERVAL altogether and just adjust the number of matches in the filter.

However, what I will do is create a PR that changes the default setting of OPT_FAST_DELETE from ON to OFF.  This will not affect existing databases, and this will have no effect on the zoneminder packages in zmrepo since they have been shipping with this option turned off for a couple years now.

OPT_FAST_DELETE is one of the features in ZoneMinder which was designed back when hard drives were slow and a VGA camera was considered crazy high resolution. Under those conditions, OPT_FAST_DELETE works as intended. However, with today's drives and high resolution cameras, it has become a lot easier to get zoneminder recording data to disk faster than the data is removed from the disk. To avoid this, the best recommendation is to turn OPT_FAST_DELETE to OFF as it is no longer relevant with modern hardware.

PR #1657 (2016-10-20): change fast_delete to off
fixes #1597
commit 191efed8 (2016-10-20): change fast_delete to off - fixes #1597
closed 2016-10-20 by connortechnology