ZoneMinder/zoneminder #1600: ZM_UPDATE_CHECK_PROXY

rebell3x6, 2016-08-26:
Hi,

I have a proxy server in my environment to connect to the internet. I setup everything right in the config, but update check sill failed.

I changed the following line in zmupdate.pl and now it works.

--$ua->proxy( "http", $Config{ZM_UPDATE_CHECK_PROXY} );
++$ua->proxy( "https", $Config{ZM_UPDATE_CHECK_PROXY} );

connortechnology, 2016-09-01:
Um... so this person's proxy requires https.  That seems totally weird to me.  @rebell3x6 So you use a proxy for https... but not http?

Code wise I suppose we could add another setting... like ZM_UPDATE_CHECK_PROXY_PROTOCOL

kylejohnson, 2016-09-02:
Either adding a `ZM_UPDATE_CHECK_PROXY_PROTOCOL` option, or removing the prepended `http` from that line. 

Would require some testing to see if the proxy setting works with `http` removed and then with the user entering a proxy without `http` (a protocol-less proxy setting).

knnniggett, 2016-09-02:
Why don't we just parse the first part of url? Simple enough. No need for an additional ZM parameter.

I would like @rebell3x6 to answer the question @connortechnology  asked before we make any changes so we know the full picture, though.

rebell3x6, 2016-09-15:
This was the error is have seen in the ZM Update Logfile, so I thought https could bring the solution.

Error check failed: '500 Can't connect to update.zoneminder.com:443'

closed 2018-06-22 by stale[bot]