ZoneMinder/zoneminder #1603: Wrong paths in zmaudit

fri-K, 2016-08-29:
Hi
Zmaudit doesn't clean my disk when I delete event in WebUI and in logs I can find multiple following error messages:
```
zmaudit[5588].ERR [Can't chdir to 'events/1/1/16/08/26': No such file or directory]
zmaudit[5588].ERR [delete_empty_directories: Can't open directory '/var/cache/zoneminder/events/1/16/07/271': No such file or directory]
```

In first there is two times monitor number in path, what is wrong. In second there is strange 271 number after month, when there should be a day.
When running it by hand I get lots of:
```
$ sudo -u www-data zmaudit.pl -r
Use of uninitialized value in multiplication (*) at /usr/share/perl5/ZoneMinder/Event.pm line 345.
```

Is this problem of my setups (two mochines), or is there bug in zmaudit?

OS: Ubuntu 16.04 [i386,x86_64]
ZM: zoneminder.1.30.4-xenial-2016082201 (from http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial main)

knnniggett, 2016-08-29:
This appears to be a packaging problem.
The Event.pm in our source tree is not the same as the one you've got in your package:
https://github.com/ZoneMinder/ZoneMinder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/Event.pm

Notice it is only 330 lines long. The error message you received is in a subroutine called age, which isn't in our source tree. Please contact the package maintainer directly to resolve this. You can create a thread in our user forum if you need a place to discuss this.

closed 2016-08-29 by knnniggett

SteveGilvarry, 2016-08-29:
You have picked the wrong ppa, 1.30.4 is only available in the storageareas ppa, which is a bleeding edge experimental branch.
https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder-storageareas

connortechnology, 2016-08-29:
@fri-K Feel free to continue with me personally on this. iconnor@connortechnology.com.