ZoneMinder/zoneminder #1612: Selection for what sources show in montage review

tommycw1, 2016-09-02:
I would like to have an option to choose which sources show up in the montage review view.  I'm thinking a check box in the source to exclude from that view would be great!

mattdurant, 2016-09-03:
The current expectation is that, like the Montage view, you change your Group first.  Whatever group you have selected is used to limit which cameras show up in Cycle, Montage, and Montage View.

tommycw1, 2016-09-03:
That's cool.  I honestly never knew this existed!

knnniggett, 2016-09-03:
Guys, I'm closing this out. 

We have far more feature requests than volunteers willing to work on them. If you want a new feature added then you must be willing to write it yourself or assign a bounty to it to encourage someone else to write it for you. 

Feel free to have discussions in our user forum or irc channel.

closed 2016-09-03 by knnniggett