ZoneMinder/zoneminder #1613: in feature-h264-videostorage zmc instances sometimes stop with no errors, but do not start

bhaal2, 2016-09-03:
In record mode, zmc instances sometimes stop and do not restart until the next event time, zma still appears to be running as it's recording the frames being recorded.  The video files are closed properly and no error is generated.  I have turned on debug logging to try and capture this again but it's hard to tell unless I am looking at all events...

2016-09-03 23:53:33.064048	zmc_m6		31226	INF	Deleting videoStore instance	zm_ffmpeg_camera.cpp	625
2016-09-03 23:53:33.063179	zms		10674	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:32.954328	zms		10671	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:32.860564	zmu		10666	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:32.852424	zmc_m5		31210	INF	Deleting videoStore instance	zm_ffmpeg_camera.cpp	625
2016-09-03 23:53:32.717499	zmu		10664	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:32.672834	zmc_m7		31243	INF	Deleting videoStore instance	zm_ffmpeg_camera.cpp	625
2016-09-03 23:53:32.625065	zms		10662	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:32.572924	zmc_m4		31197	INF	Deleting videoStore instance	zm_ffmpeg_camera.cpp	625
2016-09-03 23:53:32.561452	zmu		10661	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:32.489634	zms		10658	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:32.489051	zms		10659	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:32.468296	zmu		10657	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:32.358895	zmc_m2		8880	INF	Deleting videoStore instance	zm_ffmpeg_camera.cpp	625
2016-09-03 23:53:32.332689	zmu		10653	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:32.055225	zms		10643	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:31.961247	zms		10642	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:31.140087	zms		10636	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61
2016-09-03 23:53:31.140047	zms		10635	INF	Loaded Storage area 1 '2TB'	zm_storage.cpp	61

connortechnology, 2016-09-03:
On my todo list is to make the Loaded Storage area code into a debug instead of INF.  

You can increase the debugging level using kill -SIGUSR1 pid to find out what a zmc instance is doing.  Or set your debug settings higher to being with.  

The other thing for other readers is that I think this is storageareas branch, not feature-h264-videostorage, right @bhaal2 ?

bhaal2, 2016-09-03:
Yep, I'm using storageareas branch at the moment, but I also experienced the same problem with feature-h264-videostorage before making the change which is why I mentioned this for feature-h264 :) I just kept forgetting to create an issue.  So here we are :)  I have set ZM to debugging, so hopefully will catch the bug in action soon, otherwise I will scan for files under a certain size to find them and get to you with corresponding logs..

bhaal2, 2016-09-04:
Small update ... and I cannot confirm this at this stage, but turning on debug seemed to break the h264 videos ... They were still being written but were otherwise corrupted or such...  I will confirm when I have time to sit down again..

bhaal2, 2016-09-09:
Hi ic0n,

Ok attached is logfile for when the cameras start corrupting the .mp4 files.  The files that are saved at 3am are not corrupted, but the files saved at 3:10am onwards are corrupted...  I've included an hours worth of logs, which is large, but hopefully enough information..

[brokendebug.zip](https://github.com/ZoneMinder/ZoneMinder/files/463337/brokendebug.zip)

SteveGilvarry, 2016-09-12:
@pliablepixels Narrowed this down to zmNinja doing something different when you load it when authentication is on. So off works fine and we keep recording, but turn auth on and it causes our zmc based recording to stop. Got any hints on where to look for what zmninja is doing when auth on that it doesn't when off?

pliablepixels, 2016-09-12:
Based on what @bhaal2 and I looked at in #irc today, this _might_ be **load** related. 
Summary: This issue occurs only if montage view loads up. This issue does not occur in other screens, nor does it occur in zmN 'low-bandwidth' mode 

What we tried and observed:

With 'low bandwidth mode' **disabled** in zmN:
a) This only happens in montage screen (auth or no auth) of zmN.
b) If you switch to events view in zmN and move it to the background and move it back to foreground, then zmN loads events view (not montage view) and this behavior is not observed.

With 'low bandwidth mode' **enabled** I'm zmN:
a) Behavior does not occur in montage as well

What low bandwidth mode does differently in zmN: 

a) Reduces quality of images
b) Instead of polling for snapshots at 2 secs for each image, it does 8 secs
c) Disables checking for "alarm status" via alarm APIs for each monitor, which occurs once in 5 seconds

SteveGilvarry, 2016-09-12:
I will have to catch you on irc sometime. I am able to replicate it in Live view for one monitor. But updated to latest version and low bandwidth does seem to help it. It doesn't need to affect anything else just flags recording to stop.

pliablepixels, 2016-09-12:
@SteveGilvarry - I'm traveling all of this week, my IRC logins might be at odd times. Give this a quick try: modify getVersion function in api/Controller/HostController.php to return 1.29 instead of 1.30 and then try the live view. Without low bandwidth. Let me know if you see a problem.

If that solves it, then the problem lies here https://github.com/ZoneMinder/ZoneMinder/blob/master/web/api/app/Controller/MonitorsController.php#L200

SteveGilvarry, 2016-09-13:
@pliablepixels @bhaal2 and I arrived at the command `zmu -mX -s` is doing it, thanks for the help finding what was being called, I will debug that.
On a side note did you PR anything to fix that verbose hack you have in there?

pliablepixels, 2016-09-13:
Ok great. No, I never did PR a verbose fix.

closed 2016-09-21 by bhaal2