ZoneMinder/zoneminder #1615: Hikvision DS-2CD2020F-I no picture

belmontpier, 2016-09-04:
Dear Support,

I have installed zoneminder on ubuntu 14.04 and it ONVIF search found my DS-2CD2020F-I and all the settings were filled in.
But when I want to watch the cam, there is nothing.

DS-2CD2020F-I works well with VLC and rtsp.

Please let me know if you want to test more I would be glad to make this cam work with zoneminder.

knnniggett, 2016-09-04:
You've not provided us any data to indicate this is a bug. Please take this to the user forum, and provide the contents of your zoneminder log.

closed 2016-09-04 by knnniggett

belmontpier, 2016-09-04:
where are the logs??? I m not  dev but a user...

belmontpier, 2016-09-04:
2016-09-04 16:45:02.590717	zmc_m3		3875	ERR	Unable to get response, disconnecting	zm_remote_camera_http.cpp	1141
2016-09-04 16:45:02.586422	zmc_m3		3875	ERR	Unable to read header	zm_remote_camera_http.cpp	628
2016-09-04 16:44:55.034792	web_php		3673	ERR	socket_sendto( /var/run/zm/zms-694187s.sock ) failed: No such file or directory	/usr/share/zoneminder/www/includes/functions.php	1993
2016-09-04 16:44:54.483444	zmc_m3		3875	ERR	Unable to get response, disconnecting	zm_remote_camera_http.cpp	1141
2016-09-04 16:44:54.476031	zmc_m3		3875	ERR	Unable to read header	zm_remote_camera_http.cpp	628
2016-09-04 16:44:46.360647	zmc_m3		3875	INF	Starting Capture version 1.30.0	zmc.cpp	250
2016-09-04 16:44:46.337640	zmdc		3356	INF	'zmc -m 3' starting at 16/09/04 16:44:46, pid = 3875	zmdc.pl	
2016-09-04 16:44:46.337420	zmdc		3875	INF	'zmc -m 3' started at 16/09/04 16:44:46	zmdc.pl	
2016-09-04 16:44:46.227410	zmwatch		3419	INF	Restarting capture daemon for HIKVISION20DS-2CD2020F-I, shared data not valid	zmwatch.pl	
2016-09-04 16:44:43.992258	zmc_m1		3391	INF	HIKVISION1: 10000 - Capturing at 25.64 fps	zm_monitor.cpp	3116
2016-09-04 16:44:43.003292	web_js		3673	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-694187s.sock ) failed: No such file or directory - checkStreamForErrors()	?view=watch	
2016-09-04 16:44:42.979870	web_php		3673	ERR	socket_sendto( /var/run/zm/zms-694187s.sock ) failed: No such file or directory	/usr/share/zoneminder/www/includes/functions.php	1993
2016-09-04 16:44:36.216770	zmdc		3356	INF	'zmc -m 3' exited normally	zmdc.pl	
2016-09-04 16:44:36.201413	zmc_m3		3847	ERR	Unable to get response, disconnecting	zm_remote_camera_http.cpp	1141
2016-09-04 16:44:36.197472	zmc_m3		3847	ERR	Unable to read header	zm_remote_camera_http.cpp	628
2016-09-04 16:44:36.193562	zmc_m3		3847	ERR	Select error: Interrupted system call	zm_remote_camera_http.cpp	175
2016-09-04 16:44:36.189603	zmc_m3		3847	INF	Got signal 15 (Terminated), exiting	zm_signal.cpp	40
2016-09-04 16:44:36.185140	zmdc		3356	INF	'zmc -m 3' sending stop to pid 3847 at 16/09/04 16:44:36	zmdc.pl	
2016-09-04 16:44:36.075210	zmwatch		3419	INF	Restarting capture daemon for HIKVISION20DS-2CD2020F-I, time since last capture 6 seconds (1473000276-1473000270)	zmwatch.pl	
2016-09-04 16:44:30.956040	web_js		3675	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-694187s.sock ) failed: No such file or directory - checkStreamForErrors()	?view=watch	
2016-09-04 16:44:30.932631	web_php		3675	ERR	socket_sendto( /var/run/zm/zms-694187s.sock ) failed: No such file or directory	/usr/share/zoneminder/www/includes/functions.php	1993
2016-09-04 16:44:30.441649	zmc_m3		3847	ERR	Unable to get response, disconnecting	zm_remote_camera_http.cpp	1141
2016-09-04 16:44:30.435132	zmc_m3		3847	ERR	Unable to read header	zm_remote_camera_http.cpp	628
2016-09-04 16:44:22.332309	zmc_m3		3847	ERR	Unable to get response, disconnecting	zm_remote_camera_http.cpp	1141
2016-09-04 16:44:22.326400	zmc_m3		3847	ERR	Unable to read header	zm_remote_camera_http.cpp	628
2016-09-04 16:44:18.919918	web_js		3675	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-694187s.sock ) failed: No such file or directory - checkStreamForErrors()	?view=watch	
2016-09-04 16:44:18.903796	web_php		3675	ERR	socket_sendto( /var/run/zm/zms-694187s.sock ) failed: No such file or directory	/usr/share/zoneminder/www/includes/functions.php	1993
2016-09-04 16:44:14.230010	zmc_m3		3847	ERR	Unable to get response, disconnecting	zm_remote_camera_http.cpp	1141
2016-09-04 16:44:14.225047	zmc_m3		3847	ERR	Unable to read header	zm_remote_camera_http.cpp	628
2016-09-04 16:44:06.887139	web_js		3675	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-694187s.sock ) failed: No such file or directory - checkStreamForErrors()	?view=watch	
2016-09-04 16:44:06.875539	web_php		3675	ERR	socket_sendto( /var/run/zm/zms-694187s.sock ) failed: No such file or directory	/usr/share/zoneminder/www/includes/functions.php	1993
2016-09-04 16:44:06.078692	zmc_m3		3847	INF	Starting Capture version 1.30.0	zmc.cpp	250
2016-09-04 16:44:06.060210	zmdc		3847	INF	'zmc -m 3' started at 16/09/04 16:44:06	zmdc.pl	
2016-09-04 16:44:06.060210	zmdc		3356	INF	'zmc -m 3' starting at 16/09/04 16:44:06, pid = 3847	zmdc.pl	
2016-09-04 16:44:05.956230	zmwatch		3419	INF	Restarting capture daemon for HIKVISION20DS-2CD2020F-I, shared data not valid	zmwatch.pl	
2016-09-04 16:44:04.025400	zmc_m1		3391	INF	HIKVISION1: 9000 - Capturing at 25.00 fps	zm_monitor.cpp	3116
2016-09-04 16:43:55.948600	zmdc		3356	INF	'zmc -m 3' exited normally	zmdc.pl	
2016-09-04 16:43:55.932341	zmc_m3		3821	ERR	Unable to get response, disconnecting	zm_remote_camera_http.cpp	1141
2016-09-04 16:43:55.920928	zmc_m3		3821	ERR	Unable to read header	zm_remote_camera_http.cpp	628
2016-09-04 16:43:55.918987	zmc_m3		3821	ERR	Select error: Interrupted system call	zm_remote_camera_http.cpp	175
2016-09-04 16:43:55.915053	zmc_m3		3821	INF	Got signal 15 (Terminated), exiting	zm_signal.cpp	40
2016-09-04 16:43:55.910550	zmdc		3356	INF	'zmc -m 3' sending stop to pid 3821 at 16/09/04 16:43:55	zmdc.pl	
2016-09-04 16:43:55.806990	zmwatch		3419	INF	Restarting capture daemon for HIKVISION20DS-2CD2020F-I, time since last capture 7 seconds (1473000235-1473000228)	zmwatch.pl	
[... cut ...]
2016-09-04 16:43:40.144227	zmc_m3		3821	ERR	Unable to read header	zm_remote_camera_http.cpp	628
2016-09-04 16:43:32.037832	zmc_m3		3821	ERR	Unable to get response, disconnecting	zm_remote_camera_http.cpp	1141
2016-09-04 16:43:32.033529	zmc_m3		3821	ERR	Unable to read header	zm_remote_camera_http.cpp	628
2016-09-04 16:43:30.801160	web_js		3675	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-694187s.sock ) failed: No such file or directory - checkStreamForErrors()	?view=watch

belmontpier, 2016-09-04:
Not solved at all...

belmontpier, 2016-09-04:
my mistake, sorry Sir