ZoneMinder/zoneminder #1619: Event API BasePath is incorrect?

annunaki2k2, 2016-09-12:
Just checking out the zoneminder API as I'm thinking of using it, but when looking at an event, the "BasePath" attribute doesn't seem to seem to reflect reality.

Running the following curl command on my server: `curl -s http://localhost/zm/api/events/4023.json` returns this following Event info: `
{
    "event": {
        "Event": {
            "Id": "4023",
            "MonitorId": "5",
            "Name": "Event-front14023",
            "Cause": "Motion",
            "StartTime": "2016-09-12 19:37:42",
            "EndTime": "2016-09-12 19:37:53",
            "Width": "1920",
            "Height": "1080",
            "Length": "11.50",
            "Frames": "199",
            "AlarmFrames": "36",
            "TotScore": "1281",
            "AvgScore": "35",
            "MaxScore": "70",
            "Archived": "0",
            "Videoed": "0",
            "Uploaded": "0",
            "Emailed": "0",
            "Messaged": "0",
            "Executed": "0",
            "Notes": "Motion: All",
            "BasePath": "\/usr\/share\/zoneminder\/www\/events\/5\/16\/09\/12\/19\/37\/42\/",
            "Next": "4024",
            "Prev": "4022",
            "NextOfMonitor": null,
            "PrevOfMonitor": "4021"
        },` 
_\<snip\>_

You see that the `BasePath` returned is `/usr/share/zoneminder/www/events/5/16/09/12/19/37/42/`, but there is no such directory on the filesystem. In fact, of that entire path, just the base directory with the monitor ID (`/usr/share/zoneminder/www/events/5`) exists, and inside that are directories with event IDs. 

I.e. `ls /usr/share/zoneminder/www/events/5/` shows these directories:
`4008  4010  4012  4014  4016  4018  4019  4021  4023`

Is this a bug? Or could it just be that somehow my configuration is broken?

Thanks in advance.

pliablepixels, 2016-09-13:
I think your configuration is broken somehow. basePath does not return absolute paths.

Here is the result of my server:
```
https://myserver/zm/api/events/11902.json
```

Produces:
```
{
    "event": {
        "Event": {
            "Id": "11902",
            "MonitorId": "5",
            "Name": "New Event",
            "Cause": "Continuous",
            "StartTime": "2016-08-15 17:10:00",
            "EndTime": null,
            "Width": "2304",
            "Height": "1296",
            "Length": "319.81",
            "Frames": "1600",
            "AlarmFrames": "0",
            "TotScore": "0",
            "AvgScore": "0",
            "MaxScore": "0",
            "Archived": "0",
            "Videoed": "0",
            "Uploaded": "0",
            "Emailed": "0",
            "Messaged": "0",
            "Executed": "0",
            "Notes": "",
            "BasePath": "events\/5\/16\/08\/15\/17\/10\/00\/",
            "Next": "11903",
            "Prev": null,
            "NextOfMonitor": "11912",
            "PrevOfMonitor": null
        },
        "Monitor": {
<deleted more stuff>
```

knnniggett, 2017-05-18:
closing this issue as it has gone stale

closed 2017-05-18 by knnniggett

skrussell, 2017-05-18:
Can I please have this re-opened?

I still have this - the basepath being given by the API for events is incorrect - exactly as above. I am now using 1.30.4.

If you think it is a configuration error, can you please tell me where to start looking?

My set-up is and has been working well for a very long time, it's just this one issue that I seem to be encountering.

SteveGilvarry, 2017-05-18:
What is your config value for ZM_DIR_EVENTS? And also if you don't have date and time based folders then you probably have a very old install that never had deep storage turned on. I think the option still exists but is not visible in the gui, but looks like maybe the api doesn't check for that. Have a look in your config table in db see if you can find a reference to Deep Storage.
I never ran without it so not sure if there is a conversion process from non Deep Storage, but see what you find first.

skrussell, 2017-05-19:
Thanks for the help.

Indeed, my install is quite a well established one (initially set-up with version 1.22.something many many moons ago). Been upgrading frequently ever since and have had no real issues apart from this one.

ZM_DIR_EVENTS is currently set to '/usr/share/zoneminder/www/events' (which, if it's relevant, is a symlink to '/var/lib/zoneminder/events').

I also see that there is a setting for 'ZM_USE_DEEP_STORAGE', and that is currently set to 0.

So, the question is: Should the API respect the deep storage setting, or do I have no choice to but to convert if I wish to utilise the API.

SteveGilvarry, 2017-05-19:
Might be something going on with ZM_DIR_EVENTS, it is supposed to be relative to www, so should just be events (symlinked to /var/cache/zoneminder/events in Debian/Ubuntu these days).  Maybe something in your apache config.

Given the API is new(ish) and Deep Storage = 1 is now the default and hidden, not sure anyone will be jumping to add support. Funny thing is we are moving away from the large numbers of files that prompted Deep Storage, but no discussion on changing back.

Confirmation that Deep Storage doesn't get a mention here.
https://github.com/ZoneMinder/ZoneMinder/blob/master/web/api/app/Controller/Component/ImageComponent.php#L97

If you do want to convert I found some discussion here, but please take a back up, and shutdown ZoneMinder while you do it.
https://github.com/ZoneMinder/ZoneMinder/issues/783

pliablepixels, 2017-05-19:
Interestingly, https://github.com/ZoneMinder/ZoneMinder/blob/c859f7291c105ec21fe3562ea770ebec621b873c/web/includes/functions.php#L444-L450 has a function that does honor zm_dir_events  - I would think simply replacing ImageComponent's getEventPath code to point to the one in functions.php should fix the issue.

The other part of course is what you are doing with the API. If you are using the API for your own purposes, this may be all you need.

skrussell, 2017-05-22:
Well, thank you both @SteveGilvarry & @pliablepixels both for pointing me in the right direction on this.

At some point, I may look closer at the idea of converting my installation to using DEEP_STORAGE mode, but in the meantime, applying the following change on my own installation gives me the right output:
```
--- /usr/share/zoneminder/www/api/app/Controller/Component/ImageComponent.php.orig      2017-05-22 16:04:47.811432836 +0100
+++ /usr/share/zoneminder/www/api/app/Controller/Component/ImageComponent.php   2017-05-22 16:03:50.886724514 +0100
@@ -94,7 +94,10 @@
        // Take the StartTime of an Event and return
        // the path to its location on the filesystem
        public function getEventPath( $event ) {
-               return $event['Event']['MonitorId'].'/'.strftime( "%y/%m/%d/%H/%M/%S", strtotime($event['Event']['StartTime']) );
+               if ( $config['ZM_USE_DEEP_STORAGE'] == 1 )
+                       return $event['Event']['MonitorId'].'/'.strftime( "%y/%m/%d/%H/%M/%S", strtotime($event['Event']['StartTime']) );
+               else
+                       return $event['Event']['MonitorId'].'/'.$event['Event']['Id'];
        }
 }
 ?>
```

I'm not sure if this would be considered the right way to fix this, but it does give me the desired results.

connortechnology, 2017-05-23:
I have applied your patch to master, as it is obviously correct.  Thanks for pursuing this!

skrussell, 2017-05-23:
Awesome - thank you all for your help!