ZoneMinder/zoneminder #1623: Bad PATH_ZMS path in Ubuntu PPA install

forcetrainer, 2016-09-14:
When following the directions for install on Ubuntu 16.04, I was unable to get monitors working.  I found that the default path for PATH_ZMS was set as /cgi-bin/nph-zms, but the path should actually be /zm/cgi-bin/nph-zms.  This bad path was creating sockets errors in the logs and preventing any of the monitors from functioning.

knnniggett, 2016-09-14:
The value for PATH_ZMS is different for each distro and each version of each distro, Also, the correct value you should be using is well documented in the wiki.

If you think the default should be something different than contact the package manager directly as this does not affect the source code in our github forum.

closed 2016-09-14 by knnniggett

forcetrainer, 2016-09-14:
Argh - totally my fault.  Classic case of reading too quickly and missing a vital line.  Apparently my brain saw "Congratulations" in the install docs and I stopped there.