ZoneMinder/zoneminder #1627: Error decoding key frame

spyjo, 2016-09-15:
Hello,

Version 1.30 compiled on Debian Jessie

I have added an IP cam : Wanscam HW0043, and setup with RTSP, source type "remote".
On the camera there is a setting "key frame", which is actually at 20, with fps 10.
But in the logs, it seems every time the key frame appears, it generates an error:

09/16/16 00:49:55.294498 zmc_m1[9278].ERR-zm_remote_camera_rtsp.cpp/322 [Error while decoding frame 720]
09/16/16 00:49:55.297325 zmc_m1[9278].ERR-zm_remote_camera_rtsp.cpp/323 [8: 00 00 01 06 e5 01 c5 80]
09/16/16 00:49:57.306335 zmc_m1[9278].ERR-zm_remote_camera_rtsp.cpp/322 [Error while decoding frame 740]
09/16/16 00:49:57.310938 zmc_m1[9278].ERR-zm_remote_camera_rtsp.cpp/323 [8: 00 00 01 06 e5 01 c7 80]
09/16/16 00:49:59.298672 zmc_m1[9278].ERR-zm_remote_camera_rtsp.cpp/322 [Error while decoding frame 760]
09/16/16 00:49:59.301333 zmc_m1[9278].ERR-zm_remote_camera_rtsp.cpp/323 [8: 00 00 01 06 e5 01 c9 80]

If I change the source type with libvlc of ffmpeg this error disappear, but I would prefer still using the "remote" type because it uses less CPU.

Thanks

spyjo, 2016-09-16:
Ok not sure why but today I have restarted everything (camera + zoneminder) and now I don't have this error anymore. Closing for now

spyjo, 2016-09-16:
Ok I reopen finally.. This is weird, if I switch LOG_DEBUG on, the error disappears in the log page and I don't see it in log files. If I switch LOG_DEBUG off, the error appears in the log page. Not sure what to do.

knnniggett, 2016-11-23:
We are no longer devoting resources to maintain the Remote -> RTSP method due to lack of resources. It is possible it may go away in the future. Please switch to either the ffmpeg or libvlc source type since they offer much better compatibility. 

We will accept pull requests from you or others to address this, if it can be shown to resolve the issue w/o causing new problems.

closed 2016-11-23 by knnniggett