ZoneMinder/zoneminder #1653: Onvif probe is bound to the default interface

kfirsa, 2016-10-13:
Although the system has two interfaces the multicast probe (239.255.255.250:3702) is bound to the one having a default route.

My design is as follows:<ISP>--(R)---LAN--(ZM)---[CAM network]

verified with tcpdump.
accessing onvif profile based on ip works but the probe failed.

JoeAlamo, 2016-11-10:
I have also recently experienced this issue.

I am not aware of a code based fix (I'm sure that when setting up the UDP Multicast socket you should be able to join all available interfaces).

However, there is a temporary network level fix. Create a static route for traffic to 239.255.255.250/32 to go via your LAN/CAM interface.

This worked for me, I know it's not great.

ziotibia81, 2018-01-09:
I did some research. I'm not a multicast expert but seems that linux need a static route to multicast address for each interface that need multicast traffic. Windows does it for default.

more info here: [https://askubuntu.com/questions/501842/add-multicast-ip-address-into-kernel-route-table](https://askubuntu.com/questions/501842/add-multicast-ip-address-into-kernel-route-table)

closed 2018-06-22 by stale[bot]

klausseufzger, 2018-11-06:
I am running zoneminder 1.32.2 on Ubuntu 18.04.01 in a virtual machine with two interface (one home and one camera network). I also had the problem with ONVIF detection.

Adding a static route for 239.255.255.250 worked for me. Use the command 
`route add -net 239.255.255.250 netmask 255.255.255.255 eth1`
where eth1 is the interface of your camera network.

If you use a newer ubuntu version with netplan you can also add it the route to the interface. This is my netplan config:

```
network:
    ethernets:
#home network
        eth0:
            addresses: []
            dhcp4: true
#camera network
        eth1:
            addresses: [192.168.1.5/24]
            routes:
              - to: 239.255.255.250/32
                via: 192.168.1.5
            dhcp4: no
    version: 2
```

cweiland, 2023-05-25:
works on a ubuntu vm but not inside a lxc debian (using proxmox), why ?

connortechnology, 2023-05-25:
routing issue?  onvif uses multicast.... maybe multicast isn't enabled?

cweiland, 2023-05-27:
yes, transparent bridge on opnsense/pfsense are not so .... transparent
After some tuning, everything going well :)