ZoneMinder/zoneminder #1654: Event view shows actual image size, not the one when event was created

ricardomenzer, 2016-10-14:
I have a camera that was configured vertically (720x1280 WxH). After sometime, I decided it was better to put it the normal way (1280x720 WxH). But when I try to view an Event recorded with the previous configuration, it shows a stretched image in the 1280x720 factor.

I expect it to show the event in the original "resolution" used when it was created, not the one used by the camera at the viewing moment.

connortechnology, 2016-10-14:
Thanks for the report.

I actually have fixed this in my storageareas branch.  

Not sure when it will make it to a release.

SteveGilvarry, 2016-10-31:
Feature request to store event dimensions not just use monitor dimensions.

closed 2016-10-31 by SteveGilvarry