ZoneMinder/zoneminder #1660: zoneminder.com ssl certificate expired Oct 22 2016

jamiemaher, 2016-10-24:
Hello, 

 It looks like the ssl certificate for the wiki/forums/www has just expired over the weekend:

forums.zoneminder.com uses an invalid security certificate. The certificate expired on Saturday, October 22, 2016 4:59 PM. The current time is Monday, October 24, 2016 8:33 AM. Error code: SEC_ERROR_EXPIRED_CERTIFICATE

I couldn't connect to the main zm website because of this error either to report it, so hopefully this makes it back to the website admin.

kylejohnson, 2016-10-24:
Hi there.

I replaced the certificate, as well as moved the SSL termination point, late last week.  This isn't happening on my system.  Could you please show me the steps you took to see these errors?  Also, please let me know which IP addresses the three hostnames resolve to.

Thank you

jamiemaher, 2016-10-24:
It looks like zoneminder.com is resolving locally to 52.7.23.241 via our DNS server here.. this is the server returning an old expired certificate.

When I did a lookup with an external internet based tool it returned: 
52.7.93.24
54.85.114.40

kylejohnson, 2016-10-24:
Could you show me the output of `dig zoneminder.com`?

jamiemaher, 2016-10-24:
Sorry, looks like a false alarm..
I cleared the cache of the DNS server and it seems to be working now.

For some reason the local DNS server has some really funky old cached values:
52.7.23.241 for forums,wiki, and zoneminder.com
SOA: a.ns.joker.com, hostmaster.joker.com
NS: a.ns.joker.com
NS: b.ns.joker.com
NS: c.ns.joker.com

kylejohnson, 2016-10-24:
Good to hear.  I was going to recommend clearing your cache depending on the output of the `dig` command.

Just for reference, we did move registrars (from joker.com to AWS), so those values do make sense on a cached system.

closed 2016-10-24 by kylejohnson