ZoneMinder/zoneminder #1663: Axis 5635-E PTZ not working in zoneminder 1.30

IT-malaysia, 2016-10-27:
Recently, I bought Axis 5635-E camera & try to configure with latest version of zoneminder 1.30.
live view working fine but i am unable to configure PTZ with zoneminder. after long searching on google found result that AXIS V2 API version not supported this new camera...

please advise how to do this.

Regards,
Sahim

knnniggett, 2016-10-27:
This issue does not meet our criteria for a bug report and will be closed.

Please read our posting rules then take this to the user forum:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

closed 2016-10-27 by knnniggett

connortechnology, 2016-10-27:
@IT-malaysia That camera claims to be onvif compliant.  You might try the onvif PTZ code.

When you post to the forums, please provide more information as to how to configure this camera in ZM.  I have found a demo camera online and may be able to help, but you will need to minimize the work that we have to do.