ZoneMinder/zoneminder #1664: ZM1.3 Onvif Detection of Doorbird Intercom

Transalpler, 2016-10-30:
When trying to connect the Intecom to Zm via Onvif detection I get: "Unable to probe network cameras, status is '255'"

This is what I get when Probing via Command Line:

`wolfgang@Zoneminder:~$ sudo zmonvif-probe.pl probe
[sudo] Passwort für wolfgang:
http://192.168.27.80/onvif/device_service, 1.1, (Profile='Streaming', hardware='IntercomDoorBirdOne', name='DoorBird', location/country='Germany', location/city='Berlin', location/building='Joachimsthaler_Str_12')
`

tony5, 2016-11-19:
you have the URL wrong it is sudo zmonvif-probe.pl -v profiles "http://192.168.27.80/onvif/device_service" 1.2 myuser mypassword

but then I get the error Can't call method "GetProfiles" on an undefined value at /usr/bin/zmonvif-probe.pl line 238.

knnniggett, 2016-11-19:
We have since made improvements to the zoneminder onvif probe, if you still have an issue after you try again w/o the trailing "s" as @tony5 indicated, then please try again after rebuilding zoneminder from the master the branch. Otherwise, just wait until the next release.

Transalpler, 2016-11-19:
wolfgang@Zoneminder:~$  sudo zmonvif-probe.pl -v profiles "http://192.168.27.80/onvif/device_service" 1.2 myuser mypassword
[sudo] Passwort für wolfgang:
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"><faultcode>SOAP-ENV:Server</faultcode><faultstring>Error deserializing message: Bad top node html at /usr/share/perl5/ONVIF/Deserializer/MessageParser.pm line 89.
 at line 4 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 82.
.
Message was:
&lt;?xml version=&quot;1.0&quot; encoding=&quot;iso-8859-1&quot;?&gt;
&lt;!DOCTYPE html PUBLIC &quot;-//W3C//DTD XHTML 1.0 Transitional//EN&quot;
         &quot;http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd&quot;&gt;
&lt;html xmlns=&quot;http://www.w3.org/1999/xhtml&quot; xml:lang=&quot;en&quot; lang=&quot;en&quot;&gt;
 &lt;head&gt;
  &lt;title&gt;404 - Not Found&lt;/title&gt;
 &lt;/head&gt;
 &lt;body&gt;
  &lt;h1&gt;404 - Not Found&lt;/h1&gt;
 &lt;/body&gt;
&lt;/html&gt;
</faultstring><faultactor>urn:localhost</faultactor></Fault>wolfgang@Zoneminder:~$

knnniggett, 2017-05-18:
Closing all the Onvif issues out since the latest release contains improvements to the Onvif probe. Please create a new issue if problems persist.

closed 2017-05-18 by knnniggett