ZoneMinder/zoneminder #1671: Deleting event from watch view doesn't work

kylejohnson, 2016-11-05:
If you click on one of the `X` buttons in the `eventList` of the `watch` view, the request that is sent to the server has the `id` set to `undefined` and as such doesn't actually do anything.  Submitting a PR for this.

connortechnology, 2016-11-08:
I don't see how this code could work. event is defined at the time we are generating the button using the dom, but will not be defined when the button is clicked.  This works for the other buttons because it is passing the current value of event.Id, not the event object.  delete event needs to be changed to not take event as a parameter.

connortechnology, 2016-11-08:
yeah we should be passing e instead of event.

kylejohnson, 2016-11-08:
Passing `e` instead of `event` isn't the issue, though it is causing the `event.stop` error.

The issue is that for this button, the var `event` was being overwritten within `function( event )` and as such when we passed `event.Id` to `deleteEvent`, `event.Id` was not defined.  You can see that the other buttons don't have a `function( event )`.

> yeah we should be passing e instead of event.

My PR does fix the immediate issue, but then causes the `event.stop` error.  SO we should probably pass `e` as the first argument.

connortechnology, 2016-11-08:
yeah that's what I'm saying.  This shouldn't have been merged because it was just incomplete.   passing e will complete the fix.

kylejohnson, 2016-11-08:
Ah, fair enough.
BTW, originally I couldn't find where `event.stop` was defined.  Looks like it comes right from mootools: http://mootools.net/core/docs/1.6.0/Types/DOMEvent#DOMEvent:stop

knnniggett, 2016-11-08:
My bad. I thought I understood what was happening here and merged it. Clearly, I should have tested this before merging.

commit 735d2df3 (2016-11-07): event is redefined before being passed to deleteEvent.  Fixes #1671
commit 976c2eee (2016-11-07): Merge pull request #1672 from kylejohnson/1671-watch-view-event-delete

Fix event deletion from watch view.  Fixes #1671
closed 2016-11-10 by kylejohnson