ZoneMinder/zoneminder #1673: No function assigned to "Center" button in PTZ controls

SynAckFin, 2016-11-05:
When a camera's control capabilities has "can move" set but not "can Move Absolute", "can Move Relative" or "can Move Continuous" then no function is assigned to the center button on the ptz control console.
As a result an error message pops up.

The problem is in skins/classic/includes/control_functions.php at line 118:

        elseif ( $monitor->CanMoveAbs() )
        {
            $cmds['MoveRoot'] = "moveAbs";
            $cmds['Center'] = $cmds['PresetHome'];
                } else {
                        $cmds['MoveRoot'] = '';
        }
This should be:

       elseif ( $monitor->CanMoveAbs() )
        {
            $cmds['MoveRoot'] = "moveAbs";
            $cmds['Center'] = $cmds['PresetHome'];
        } else {
            $cmds['MoveRoot'] = '';
            $cmds['Center'] = "Stop";
        }

knnniggett, 2016-11-05:
Like the camera you are using, my airlink 747W can move relative but not absolute. When I click the center button, the camera goes to its home position with no errors in the log so its working fine here.

To troubleshoot this further, for starters we need a detailed the description of the problem:

- Provide make/model of the camera you are attempting to use
- Tell us what control script you are attempting to use. More information may be required if you are attempting to use a custom script.
- Copy the exact error message from the logs, verbatim.
- Verify the control script you are attempting to use has a "presetHome" subroutine
- The error message usually states the command it tried to execute. Paste that command into the command line and execute it as root, then provide the output of that (this often provides

SynAckFin, 2016-11-05:
This issue doesn't have anything to do with move absolute (or relative). That code line is only there to provide context of where the bug is and the fix that needs to be applied.

> Provide make/model of the camera you are attempting to use
> 

Camera model is irrelevant. This is a bug in control_functions.php
> Tell us what control script you are attempting to use.
> 

Control script is irrelevant. The bug is in control_functions.php 

To replicate do the following:

1. Create a new control.
2. Use any existing or custom Protocol (PelcoD, PelcoP, Visca etc)
3. In move tab select "Can Move" leave all others clear.
4. Assign new control to a monitor.
5. Open the monitor and press the centre button.

You will get the following error message popping up:

> Control response was status = undefined
> message = /usr/bin/zmcontrol.pl --command= --id=5=>Usage: //      zmcontrol.pl --id {monitor_id} --command={command} [various options] //  // Options: //      --autostop          - //      --xcoord [ arg ]    - X-coord //      --ycoord [ arg ]    - Y-coord //      --speed [ arg ]     - Speed //      --step [ arg ]      - //      --panspeed [ arg ]  - //      --panstep [ arg ]   - //      --tiltspeed [ arg ] - //      --tiltstep [ arg ]  - //      --preset [ arg ]    - // 
> 

You get the usage message because --command= doesn't have a command. It doesn't have a command because the Centre button (usually Stop) doesn't have an action assigned.

knnniggett, 2016-11-05:
The make/model of camera and the control script in question is quite relevant in this case. 

When creating a ptz control script, it is not good enough to only select "Can Move" on the Move tab. You also have to tell Zoneminder the kind of movement the camera supports by selecting one or more of the checkboxes below "Can Move". This is how all the other control scripts in ZoneMinder are setup, and is exactly why I asked you what script you are attempting to use.

I'm closing this out because the issue you have described is in fact not a bug, but working exactly the way it was intended. 

However, I would absolutely agree with the argument that 1) there should be better documentation describing this behaviour and 2) the Move tab should be modified to prevent the user from hitting save until more than just the "Can Move" checkbox is selected. We would be more than happy to accept a pull request from you or anyone else who wants to address either of those.

closed 2016-11-05 by knnniggett

connortechnology, 2016-11-06:
We could probably do an isset() to test for the case and provide a better error message... but that's about it.

I get this with one of my cameras and it has been on my todo list but I pretty much never use the ptz features of my cam, so it almost never shows up in my logs.

The long and short is... we can probably do better... and it actually doesn't matter which camera or protocol, although it might be that the protocol/camera in question CAN do what is needed... so we would like to know.  SynAckFIn please humour us.

SynAckFin, 2016-11-06:
Two no name Chinese PTZs using Pelco D and a HikVision DS-2DE4220IW-D

The protocol is irrelevant since zmcontrol.pl never reaches the point where it gets to load it. It doesn't reach this point because the web code calls it with the flag `--command=` and nothing after the "=".

The web code does this because it doesn't set up the control console properly in the file `skins/classic/includes/control_functions.php` when no movement type (absolute, relative, continuous) has been chosen for whatever protocol is being used.

It is a simple one line fix (`$cmds['Center'] = "Stop";`) as shown in the first post.

knnniggett, 2016-11-06:
@SynAckFin It's clear you've decided on your own how this part of ZoneMinder should work. Rather than admit the root cause of this problem is a simple failure to check the appropriate check boxes, you continue to insist we should change the code to match your idea of how things should work. We have spent more than enough time trying to help you understand this, and even offered ways to improve ZoneMinder which would make this more clear. Heck, had you told us what custom ptz script you are working with, we could have told you how to fix it.

Since the OP doesn't seem to care about anything other than his own way of doing this, I'm not going to waste any more time with this and am locking the thread for further conversation.