ZoneMinder/zoneminder #1682: Installation problem

BlackDragon381, 2016-11-14:
Hello!
I try installed ZoneMinder on Debian (Proxmox LXC container) by this manual [https://wiki.zoneminder.com/Debian_8_64-bit_with_Zoneminder_1.29.0_the_Easy_Way](https://wiki.zoneminder.com/Debian_8_64-bit_with_Zoneminder_1.29.0_the_Easy_Way).
Instead mysql-server i installed mariadb-server.
On step "systemctl start zoneminder.service" i have a error:
Failed to get D-Bus connection: Unknown error -1
How i may repair this problem?
In Proxmox VM all is ok!

knnniggett, 2016-11-15:
Please refer to our posting rules:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

Our github forum is not for installation issues. Please take this to the user forum.

knnniggett, 2016-11-15:
And do not double post. 
https://forums.zoneminder.com/viewtopic.php?f=36&t=25475

Does anyone ever bother to read instructions anymore?
Thread locked.

closed 2016-11-15 by knnniggett