ZoneMinder/zoneminder #1695: H.264 branch - mp4 video length does not match event length

pliablepixels, 2016-11-22:
the length of the event (as stored in the DB) does not match the actual mp4 video length. This creates issues when trying to insert cue points in the video player for where alarmed frames are.

@connortechnology is aware of this issue but the cause of the problem is yet unknown. I'm putting this up here so I can refer to it from zmNinja as a dependent issue.

liucougar, 2016-12-04:
i think at least one of the problems is caused by keyframes: missing keyframe at the beginning/end could cause the video to be off by several seconds (depends on the key frame settings)

SteveGilvarry, 2016-12-04:
Not an unknown issue zm begins events at arbitrary points in decoded
stream, videostream has to on keyframes. Needs to be coded to get the key
frame before events start, and offset the event to match the video.

pliablepixels, 2018-09-23:
I don't believe it is. I'm using video storage for a while now, and in general, the cue points seem to align.

connortechnology, 2018-09-23:
I think we can close.

closed 2018-09-23 by knight-of-ni