ZoneMinder/zoneminder #1696: debian / ubuntu packaging scripts install both sysv init and .service files

connortechnology, 2016-11-22:
@onlyjob 

This leads to very bad situations like 
https://forums.zoneminder.com/viewtopic.php?f=36&t=25462&p=97362&hilit=systemd

onlyjob, 2016-11-23:
Respectfully disagreed... That's policy to support both init systems. Debhelper scripts handle configs and I've never heard about such conflicts before. We don't even know which system had this problem and how it was (mis-)configured. I wouldn't bother about this unless reliable reproducer is found with packages from official Debian repository...

knnniggett, 2016-11-23:
What other applications actually install both init scripts from the same package? Is this policy documented anywhere? It certainly does not extend beyond Debian distros.

Since systemd is picky, it  refuses to acknowledge any process that it didn't start.... so by installing two different init scripts, that creates a problem for any application if it was started by the legacy init and then started again by systemd.

Additionally Debian's proprietary, backwards-compatibility layer always wants to use the legacy init script if the script simply exists on the filesystem.... so by including both init scripts, you aren't supporting both systemd and legacy systems, you are only supporting the legacy init system ....and if the user manages to override the default and start zoneminder with the systemd service file, it has no way to know zoneminder was already started via the legacy layer.

connortechnology, 2016-11-23:
Yeah I've seen this often.  I use monit to keep things running and I've had to change everything to use systemd instead of the sysv init scripts or else systemd won't be aware of the running process. 

Supporting both systems is great, but we need to remove/not install the sysv init script when systemd is in charge.

knnniggett, 2016-11-23:
I've not seen the behavior of actually installing both init scripts in the few debian packages applications I've worked with, such as mysql and apache. Redhat of course doesn't do this.

onlyjob, 2016-11-23:
Under systemd on Debian it is not possible to start application by "legacy init" because systemd takes over. For most applications it works well enough through systemd compatibility. One person experienced problem on the broken system -- let's not make drama out of it. @knnniggett, you are mistaken about init support in Debian. Systemd uses .service files when they are available and uses init scripts only when there are no native systemd service files. You may complain about systemd design or imperfect support for init scripts but we probably not going to work towards improving systemd... 

Also please don't say "Debian's proprietary" -- there is nothing proprietary in Debian. It is all about freedom and transparency.

SysV and Systemd are both supported on Debian (even though systemd is the default).
Here is lintian error when only systemd .service file is installed (but not init.d):

 https://lintian.debian.org/tags/init.d-script-not-included-in-package.html

knnniggett, 2016-11-23:
@onlyjob I'm not sure where the claim of drama is coming from. I'm simply pointing about a behavior in Debian, that isn't present in other distros.  I'm sorry you disagree with the word "proprietary", but I'd rather not argue over semantics. Call it what you want, but the functionality in question is only in Debian based distros. I'm not calling it a bad thing either. I'm calling it an unknown thing, as its not part of systemd (documentation).

My previous statement was not a complaint against systemd (I drank the coolaid - I love systemd), nor was it a complaint against supporting both sys v initi and systemd simultaneously. 

The question originally asked by @connortechnology is the one I'm asking too. Why are both init scripts installed simultaneously? And I was serious when I asked What other Debian packaged applications actually install both init scripts from the same package and is this policy documented anywhere? I'm not trying to incite drama. I'm trying to learn why you made the decision you did. I'm sorry if my written statements do not come across that way. 

I see the lintian error, but why does it complain it is missing? Do all deb packages install with both? Can the legacy script be blank or just be a redirect to systemctl? You seemed to acknowledge Debian's attempt to allow both is imperfect, so maybe the solution involves an imperfect workaround.

Hey, I need to run.

knnniggett, 2016-11-24:
I have a new data point for this issue.

When a restart command is sent through zoneminder, zmpkg.pl checks if systemd is running by checking the name of pid 1. If pid 1 = systemd then it calls zmsystemctl.pl, logs the redirect event, which in turn calls the systemctl command. We know from previous experience, this is required. Otherwise systemd is not aware zoneminder was restarted and flags the service as failed all the while the web console happily reports a running state.

In any case, the last time I tested this on an Ubuntu, the redirect through zmsystemctl was not happening because pid 1 was not systemd, even though systemd was installed. 

This is what I believe the root cause of the issue reported in the forum recently because I could previously demonstrate this behavior.

This evening I ran the same test again, and to my surprise, restarting zoneminder on Ubuntu is now properly redirecting through zmsystemctl.pl. I don't have Debian installed or I would have used that.

The previous result must have been during the transition period from sys v init to systemd. I assume Debian had a similar transition.

To extend the test even further, I did the usual tricks, which would have tripped up systemd in the past: Start zoneminder via systemctl, but restart it using the legacy service tool, or start zoneminder via systemctl and then restart it from the web console. All these worked, and I was not able to get the system into a state where the web console said it was running but systemd said it was stopped or failed, nor was able to get the service file to start one copy of zoneminder and the init script to start another.

So based on this data, I think it is a good bet our forum user was using an older Ubuntu or Debian which had not fully transitioned to systemd. If we see this reported again, we should as @onlyjob suggested verify what version they are running and get them to describe the steps they took to get it to fail.

@connortechnology I know you mentioned you've seen similar behaviour but can you check the version of Ubuntu you were running? With 16.04, the legacy service tool does not even appear to exist... it's just a link to systemctl and gives the exact same output now.

connortechnology, 2016-11-24:
Having thought about it, since init.d/zoneminder uses zmpkg.pl which in turn calls systemctl which will in turn call zmpkg.pl, it should all work out.  However I suspect that because our startups are so slow, and each perl script does so much before even doing anything that there might be some concurrency issue which is actually causing the problem.  My cases might be exacerbated by my use of monit to monitor the zmdc socket and try to start zm if not running.  So I'm closing this out, but we may need to keep this issue in the back of our minds.

connortechnology, 2016-11-24:
Oh btw, my issues are predominantly with 16.04.  Previous version was 14.04 but that was upstart, not systemd.

closed 2016-11-24 by connortechnology