ZoneMinder/zoneminder #1698: 1.30.0 debian/ubuntu bulk frames problem

ilcalmo, 2016-11-23:
Hi,
Installed 1.30.0 from debian testing repositories and also in Ubuntu (iconnor ppa).
I use mocord method.
Opening events and clicking on related events "Frames" you can see the frame list.
If you click on frame 100 (or any bulk frame such as 200 or 300) you see the image related to frame 100, if you click Prev or Next to see the 99th or 101st, you do not see image! In the syslog file I get the message:

`web_php[656]: ERR [No image path]`

while in the apache2 error log file I get the messages: 
`[:error] [pid 656] [client 192.168.19.106:61082] PHP Notice:  Undefined property: Event::$StorageId in /usr/share/zoneminder/www/includes/Event.php on line 31, referer: http://192.168.19.110/zm/?view=frame&eid=6&fid=100

[:error] [pid 656] [client 192.168.19.106:61082] PHP Notice:  Undefined property: Frame::$Id in /usr/share/zoneminder/www/includes/Frame.php on line 73, referer: http://192.168.19.110/zm/?view=frame&eid=6&fid=100

[:error] [pid 656] [client 192.168.19.106:61082] ERR [No image path], referer: http://192.168.19.110/zm/?view=frame&eid=6&fid=99&scale=100`

So only images related to frame numbers multiple of 100 are displayed.

If you play the full event, the event is displayed correctly, so the images are all in the right place.

In 1.29.0 everything worked fine.

marcmerlin, 2018-04-20:
Oh my, this problem has been going on for 1.5 years? (I just recently updated to 1.30 and now very much regret it I guess)
Now, the missing frames are just a small-ish bug, since in real life they were never recorded to start with because bulk made them go away.
I filed a bigger bug here: https://github.com/ZoneMinder/zoneminder/issues/2084

mccrobie2000, 2018-05-05:
I also use mocord and get the same thing for viewing with bulk frames.  The older versions passed a path to view.php.  The newer versions seems to pass an event-id and frame-id.  Unfortunately, bulk frames are not individually stored in the Frames table.  A marker for the first frame in the bulk "set" is stored.  This is why the first frame in the bulk set can be viewed, but not the subsequent frames.  (You can "SELECT * FROM Frames WHERE EventId=blah" to see how the frames are stored.)

I suspect this never worked but was masked when using the path to the individual bulk frame in the older versions.

I've tried to mask the error also by passing EventId and FrameId in the query string formed in Frame.php, but the new image.php asks Frame.php to "find_one".  This "find_one" function just assumes the frame is in the Frames table by EventId and FrameId - which returns no rows.

Fortunately, it appears this check in image.php is a sanity check only and can be commented out.  Doing all three changes gets me my bulk image - finally.

Somewhere along the line, one of these functions in the chain (frame.php Frame.php view.php) need to understand how bulk frames are stored in the database (not as individual frames, but rather as a marker for the set of bulk frames.)

I'll include my changes in another comment once I clean-up my debug statements.

mccrobie2000, 2018-05-05:
diff --git a/web/includes/Frame.php b/web/includes/Frame.php
index d31ddf162..a380d572e 100644
--- a/web/includes/Frame.php
+++ b/web/includes/Frame.php
@@ -70,7 +70,7 @@ class Frame {
   }
 
   public function getImageSrc( $show='capture' ) {
-    return $_SERVER['PHP_SELF'].'?view=image&fid='.$this->{'Id'}.'&show='.$show;;
+    return $_SERVER['PHP_SELF'].'?view=image&eid='.$this->{'EventId'}.'&fid='.$this->{'FrameId'}.'&show='.$show;;
   } // end function getImageSrc
 
        public static function find( $parameters = array(), $limit = NULL ) {
diff --git a/web/skins/classic/views/frame.php b/web/skins/classic/views/frame.php
index 96289f7bc..4f189909e 100644
--- a/web/skins/classic/views/frame.php
+++ b/web/skins/classic/views/frame.php
@@ -36,7 +36,7 @@ $event = dbFetchOne( $sql, NULL, array($eid) );
 if ( !empty($fid) ) {
     $sql = 'SELECT * FROM Frames WHERE EventId = ? AND FrameId = ?';
     if ( !($frame = dbFetchOne( $sql, NULL, array($eid, $fid) )) )
-        $frame = array( 'FrameId'=>$fid, 'Type'=>'Normal', 'Score'=>0 );
+        $frame = array( 'EventId'=>$eid, 'FrameId'=>$fid, 'Type'=>'Normal', 'Score'=>0 );
 } else {
     $frame = dbFetchOne( 'SELECT * FROM Frames WHERE EventId = ? AND Score = ?', NULL, array( $eid, $event['MaxScore'] ) );
 }
diff --git a/web/views/image.php b/web/views/image.php
index 82f0ed046..396d7a479 100644
--- a/web/views/image.php
+++ b/web/views/image.php
@@ -65,9 +65,9 @@ if ( empty($_REQUEST['path']) )
     if ( ! empty($_REQUEST['eid'] ) ) {
       $Event = new Event( $_REQUEST['eid'] );
       $Frame = Frame::find_one( array( 'EventId' => $_REQUEST['eid'], 'FrameId' => $_REQUEST['fid'] ) );
-      if ( ! $Frame ) {
-        Fatal("No Frame found for event(".$_REQUEST['eid'].") and frame id(".$_REQUEST['fid'].")");
-      }
+#      if ( ! $Frame ) {
+#        Fatal("No Frame found for event(".$_REQUEST['eid'].") and frame id(".$_REQUEST['fid'].")");
+#      }
       $path = $Event->Path().'/'.sprintf("%'.0".ZM_EVENT_IMAGE_DIGITS.'d',$_REQUEST['fid']).'-'.$show.'.jpg';
     } else {
 # If we are only specifying fid, then the fid must be the primary key into the frames table. But when the event is specified, then it is the frame #

mccrobie2000, 2018-05-05:
These changes only mask the issue and don't fix it.

[zoneminder-bulk.txt](https://github.com/ZoneMinder/zoneminder/files/1976342/zoneminder-bulk.txt)

marcmerlin, 2018-07-05:
This fixed the problem for me too, thank you for the patch.
Is there a chance it can be included?

asantaga, 2018-07-20:
@mccrobie2000  is the a git issue for this for the dev's to pull?

marcmerlin, 2018-09-18:
Dear devs, is anyone looking at this? Can you add @mccrobie2000 's workaround if a better fix can't be added or now?
More generally can this bug stay open until it's fixed instead of requiring us to write updates just to prevent it from auto closing?

marcmerlin, 2018-11-18:
can this bug stay open until it's fixed instead of requiring us to write updates just to prevent it from auto closing?

marcmerlin, 2019-01-17:
still not stale

closed 2019-01-24 by close-issue-app[bot]

SteveGilvarry, 2019-01-24:
Lol stupid bot

marcmerlin, 2020-07-05:
I first upgraded to ZM 1.32 which broke half my cameras in various ways, and no one on the forums could tell why
A later update to ZM 1.34.8 didn't fix anything either.
After many months of having a non working install, I migrated off and moved to shinobi