ZoneMinder/zoneminder #1704: Live scalable grid view (pull request)

Goblenus, 2016-12-11:
Live scalable cameras grid view. Support fullscreen mode.
[image]
[image]
[image]
[image]
[image]
[image]

Can use arrows to change grid view columns.

knnniggett, 2017-01-13:
This looks nice, but please move the php function calls out of gridview.php and into web/includes/functions.php

I personally don't use montage, but I'm wondering if, instead of creating a new view, we merge this capability into the standard montage. @connortechnology @kylejohnson

connortechnology, 2017-01-13:
Yes we already have too many views.

kylejohnson, 2017-01-13:
I've got an early version of this as part of the `console` view in my updated ui branch.  You can toggle between the table and the grid using the `Table | Grid` button in the page header.

I'm still figuring out what to display in the grid (e.g. do we want the event counts there, or just the simple picture?), but at this point I don't think it makes sense to have an entirely new view for this feature.

[image]

Goblenus, 2017-01-13:
I don't need to see any events or smth like that at monitor page. The main task of this page to give me current state, nothing more. There are a lot of others pages, which can provide me that information.

Goblenus, 2017-01-13:
@knnniggett well, can you give me an idea what are you talking about...I mean, you tell put my code into web/includes/functions.php, and I'd like to know how do you expect to call it? Any input/output parameters? Which page will provide this view? I need basic understanding...

knnniggett, 2017-01-13:
@Goblenus You don't have to worry about how functions.php gets loaded. That is done for you. All you have to do is move your function definitions into that file, and they will be found when called. That's it. No special understanding is necessary.

knnniggett, 2017-01-13:
Oh, I just noticed something else. 
The small bits of javascript at the bottom of gridview.php need to be moved into a file named `web/skins/classic/views/js/gridview.js`

Similar to the php function definitions, you don't have to worry about the details behind how gridview.js will get loaded. The zoneminder "framework" will find the file and load it for you.

knnniggett, 2017-04-15:
Thank you for submitting this pr. However, I'm going to close this out as we intend to implement the feature described in this issue as part of the new UI. Please coordinate with @kylejohnson to help him get that off the ground. I'm sure he would welcome the help.

closed 2017-04-15 by knnniggett