ZoneMinder/zoneminder #1732: v.1.30.1 Error: The view for EventsController::index() was not found.

dwrobel, 2017-01-02:
Following my comments on [rpmfusion](https://bugzilla.rpmfusion.org/show_bug.cgi?id=4393#c8) I've tried to validate API as per [Validating-if-APIs-work-on-ZM](https://github.com/pliablepixels/zmNinja/wiki/Validating-if-APIs-work-on-ZM):

1. /zm/api/host/getVersion.json is working and returns:
`{
    "version": "1.30.1",
    "apiversion": "1.0"
}`

2. /zm/api/monitors.json is working and returns:
`{
    "monitors": [
        {
            "Monitor": {
                "Id": "1",
                "Name": "Monitor-1",
                "ServerId": "0",
                "Type": "Local",
                "Function": "Monitor",
                "Enabled": "1",
                "LinkedMonitors": "",
                "Triggers": "",
                "Device": "\/dev\/video0",
                "Channel": "0",
                "Format": "45056",
                "V4LMultiBuffer": false,
                "V4LCapturesPerFrame": "1",
                "Protocol": "",
                "Method": "v4l2",
                "Host": "",
                "Port": "80",
                "SubPath": "",
                "Path": "",
                "Options": "",
                "User": "",
                "Pass": "",
                "Width": "720",
                "Height": "480",
                "Colours": "3",
                "Palette": "0",
                "Orientation": "0",
                "Deinterlacing": "0",
                "RTSPDescribe": false,
                "Brightness": "-1",
                "Contrast": "-1",
                "Hue": "-1",
                "Colour": "-1",
                "EventPrefix": "Event-",
                "LabelFormat": "%N - %d\/%m\/%y %H:%M:%S",
                "LabelX": "0",
                "LabelY": "0",
                "LabelSize": "1",
                "ImageBufferCount": "50",
                "WarmupCount": "25",
                "PreEventCount": "25",
                "PostEventCount": "25",
                "StreamReplayBuffer": "1000",
                "AlarmFrameCount": "1",
                "SectionLength": "600",
                "FrameSkip": "0",
                "MotionFrameSkip": "0",
                "AnalysisFPS": "0.00",
                "AnalysisUpdateDelay": "0",
                "MaxFPS": "0.00",
                "AlarmMaxFPS": "0.00",
                "FPSReportInterval": "1000",
                "RefBlendPerc": "6",
                "AlarmRefBlendPerc": "6",
                "Controllable": "0",
                "ControlId": "0",
                "ControlDevice": "",
                "ControlAddress": "",
                "AutoStopTimeout": "0.00",
                "TrackMotion": "0",
                "TrackDelay": "0",
                "ReturnLocation": "-1",
                "ReturnDelay": "0",
                "DefaultView": "Events",
                "DefaultRate": "100",
                "DefaultScale": "100",
                "SignalCheckColour": "#0000c0",
                "WebColour": "red",
                "Exif": false,
                "Sequence": "1"
            }
        }
    ]
}`

3. /zm/api/events doesn't work and reports:
[Error: The view for EventsController::index() was not found.](https://dwrobel.fedorapeople.org/tmp/zoneminder/events.html).

this prevents me from using zmNinja with that installation - previously on the same hardware using Fedora 21 + ZM v1.29 works without any issues.

If some more detailed information/logs would be helpful - please let me know.

connortechnology, 2017-01-02:
Isn't it supposed to be /zm/api/events.json ?Yup, it is.

dwrobel, 2017-01-02:
>Isn't it supposed to be /zm/api/events.json ?Yup, it is.
Thank you, event.json works indeed.
So the problem is in the https://github.com/pliablepixels/zmNinja/wiki/Validating-if-APIs-work-on-ZM page.

closed 2017-01-10 by connortechnology