ZoneMinder/zoneminder #1736: zmdc.pl tries starting zmc on non-existent device, even when monitor is disabled

jjakob, 2017-01-04:
After disabling a monitor that has a nonexisting source, zoneminder still tries to start the zmc process, leading to errors and filling up the log with unnecessary noise (this happens every 30 seconds).
The source used to be an USB camera that is no longer plugged in, hence its video device doesn't exist.
This happens even with the function set to "none".

Zoneminder version: 1.30.0
OS version: Ubuntu 16.04.1 LTS (Xenial Xerus)

Log:
In modect mode, disabled:

```
DATE/TIME
COMPONENT	SERVER	PID	LEVEL	MESSAGE	FILE	LINE
2017-01-04 18:43:11.183380	zmdc		1387	ERR	'zmc -d /dev/video0' exited abnormally, exit status 255	zmdc.pl	
2017-01-04 18:43:11.177013	zmc_dvideo0		22537	FAT	Failed to stat video device /dev/video0: No such file or directory	zm_local_camera.cpp	680
2017-01-04 18:43:11.159761	zmc_dvideo0		22537	INF	Starting Capture version 1.30.0	zmc.cpp	250
2017-01-04 18:43:11.116827	zmc_dvideo0		22537	INF	No direct match for the selected palette (	zm_local_camera.cpp	412
2017-01-04 18:43:11.108488	zmc_dvideo0		22537	ERR	Automatic format selection failed. Falling back to YUYV	zm_local_camera.cpp	347
2017-01-04 18:43:11.105992	zmc_dvideo0		22537	ERR	Automatic format selection failed to open video device /dev/video0: No such file or directory	zm_local_camera.cpp	1166
2017-01-04 18:43:11.065920	zmdc		1387	INF	'zmc -d /dev/video0' starting at 17/01/04 18:43:11, pid = 22537	zmdc.pl	
2017-01-04 18:43:11.063230	zmdc		22537	INF	'zmc -d /dev/video0' started at 17/01/04 18:43:11	zmdc.pl	
2017-01-04 18:43:11.036110	zmdc		1387	INF	Starting pending process, zmc -d /dev/video0	zmdc.pl

```
In none mode, disabled:

```
2017-01-04 18:49:05.136440	zmdc		1387	ERR	'zmc -d /dev/video0' exited abnormally, exit status 255	zmdc.pl	
2017-01-04 18:49:05.130358	zmc_dvideo0		27497	ERR	No monitors found	zmc.cpp	246
2017-01-04 18:49:05.068900	zmdc		27497	INF	'zmc -d /dev/video0' started at 17/01/04 18:49:05	zmdc.pl	
2017-01-04 18:49:05.068900	zmdc		1387	INF	'zmc -d /dev/video0' starting at 17/01/04 18:49:05, pid = 27497	zmdc.pl	
2017-01-04 18:49:05.052510	zmdc		1387	INF	Starting pending process, zmc -d /dev/video0	zmdc.pl
```

connortechnology, 2017-01-08:
I think I've seen this behaviour too although when I went to reproduce it I wasn't able to.

To be clear, "disabling" a monitor, only disables the motion detection, not the capturing, so a disabled monitor should still run the zmc process. We really aught to make this more clear in the UI.

Setting the function to none, SHOULD have the effect that the zmc process is no longer run.

jjakob, 2017-01-13:
I rebooted the whole thing and now there are no more errors...

It might have had something to do with the sequence of events regarding the missing camera, If I recall correctly, the camera was disconnected while the system (virtual machine) was off, so it booted and was still set to Modect and enabled, then I went to disable it/set the function to None and got the above mentioned errors. Rebooting seems to have fixed the issue for now.

closed 2018-06-22 by stale[bot]