ZoneMinder/zoneminder #1737: Zoneminder Hostname doesn't allow Special Characters

bridges86406, 2017-01-05:
Using Zoneminder NT
Version 1.30.0-nt2 compiled by NASTools.

When trying to enter Host Path (admin:Giant1!@192.0.0.0) the system will generate error that states that the Host must be valid.  

If I remove the "!" it will validate and let me add the monitor, however, the camera won't authenticate.

First play with this program on my ReadyNAS 3138 so it took me nearly 30 mins of playing to realize that may be the issue.  Removed the exclamation mark and it worked perfectly once I changed the password for the camera.

knnniggett, 2017-01-05:
We don't know what the NT means, but we are already aware of this issue. There may already be an open issue concerning it. Last time we looked into this it was the Apache web server escaping the characters in the password. I'm not sure there is anything we can change in the zoneminder source, other than to change how we send monitor streams, which is on the drawing board.

knnniggett, 2017-05-18:
This was fixed with the uridecode functionality which is part of 1.30.4

connortechnology, 2017-05-18:
No it wasn't.  This is in the monitor edit popup.  Not in zms auth.

hemna, 2017-08-04:
I can't even enter an IP address in the Remote Host Name field without it complaining
"Names may only contain alphanumeric characters plus hyphen and underscore."

I'm running 1.29.0

closed 2018-06-22 by stale[bot]

jmssil, 2021-03-25:
Hi @JoeD84 , have you been able to solve this? Is it perhaps fixed in a recent version of ZoneMinder?

JoeD84, 2021-03-28:
Hi @jmssil. I'm not sure, but I remember I tried to edit the DB directly and add an escaped  version of the password directly into the configuration entry. But I'm not sure if this was the solution.

jmssil, 2021-03-28:
@JoeD84, in my case I believe the problem was caused by using this address: `rtsp://admin:<password>@<ipaddress>:554/cam1/mpeg4` which I saw in [https://wiki.zoneminder.com/Honeywell](https://wiki.zoneminder.com/Honeywell).

I removed the `:554/cam1/mpeg4` part and am now using `rtsp://admin:<password>@<ipaddress>` and it works.

The password, according to camera's security rules, has symbols, e.g. `(`, `%` and `$`.

So I assume this version of ZoneMinder now accepts passwords with symbols. I'm running version v1.34.23.

Instead of closing this ticket due to inactivity it would be useful to keep it open until it is fixed, stating which version solved the issue.

ssb-kod, 2025-11-21:
For those who still are having this issue which is still present on my configuration with v1.36.35 running in a jail on TrueNAS core 13.3. The solution for me was to convert the ^ character in my password to a URL percent encoding as per this website ... https://www.w3schools.com/tags/ref_urlencode.ASP   ... which has a calculator as well as a list for special characters; which zone minder seems to not like. So the ^ is now %5E . Now takes the password when editing "source" values without complaints/ popups.