ZoneMinder/zoneminder #1739: feature-h264-videostorage HLS playback

mcbp223, 2017-01-07:
Hello,

I am trying hard to get a HLS feed for stored events, but I don't seem to get the timestamps right, there is a noticeable discontinuity between the mp4 segments. Is it possible for you to use the segmenting features from ffmpeg, since they make each segment start exactly from where the previous one ended? They also have recently added more options in v.3, including mkdir (use_localtime_mkdir).
I'm using ppa.launchpad.net/iconnor/zoneminder-feature-h264-videostorage for testing, with 10 second segments, and I'm generating the m3u8 playlist with this view:
```php
<?php

if ( !canView( 'Events' ) ) {
	$view = "error";
	return;
}

if(isset($_REQUEST['mp4tots'])) {
	header('Content-Type: video/MP2T');
	header('Content-Disposition: attachment; filename="'.basename($_REQUEST['mp4tots']).'.ts";');
	header('Content-Transfer-Encoding: binary');
	// different options pertaining to mpegts
	// -c copy
	// -bsf h264_mp4toannexb
	// -start_at_zero
	// -copyts
	// -mpegts_copyts 1
	// -initial_offset '.$_REQUEST['offset'].'
	// -vf "setpts=0.5*PTS"
	// -vf "settb=expr=0.1"
	system('ffmpeg -i '.$_REQUEST['mp4tots'].' -c copy -bsf h264_mp4toannexb -output_ts_offset '.$_REQUEST['offset'].' -f mpegts -');
	// system('ffmpeg -i '.$_REQUEST['mp4tots'].' -c copy -bsf h264_mp4toannexb -map 0 -f segment -segment_format mpegts -initial_offset '.$_REQUEST['offset'].' -segment_time 200 -');
	return;
}

//require_once('includes/Event.php');

$sqlfilter = 'MonitorId = '.$_REQUEST['channel'].' and EndTime > "'.$_REQUEST['begin'].'" and StartTime < "'.$_REQUEST['end'].'"';

$rows = dbFetchAll( 'SELECT min(StartTime) as firstStartTime, max(ceiling(Length)) as targetDuration FROM Events where '.$sqlfilter );

$firstStartTime = date_create($rows[0]['firstStartTime']);
$targetDuration = $rows[0]['targetDuration'];
if(empty($targetDuration))
	$targetDuration = 0;

$rows = dbFetchAll( 'select * from Events as E where '.$sqlfilter );
/*
if(empty($rows)) {
	Error("no recording found");
	header ("HTTP/1.0 404 Not Found");
	return;
}
*/
//header('Content-type: application/vnd.apple.mpegurl');
//header('Content-type: application/x-mpegURL');
header('Content-type: text/plain');
// pay attention to the newlines, so the output doesn't get mangled
//$DATE_FMT = "Y-m-d\TH:i:s.uP";
$DATE_FMT = "Y-m-d\TH:i:s.000P";

echo '#EXTM3U'.PHP_EOL;
echo '#EXT-X-VERSION:3'.PHP_EOL;
echo '#EXT-X-MEDIA-SEQUENCE:0'.PHP_EOL;
echo '#EXT-X-ALLOW-CACHE:YES'.PHP_EOL;
echo '#EXT-X-TARGETDURATION:'.$targetDuration.PHP_EOL;
echo '#EXT-X-PROGRAM-DATE-TIME:'.date_format($firstStartTime, $DATE_FMT).PHP_EOL;

$continueOffset = 0;

foreach($rows as $row) {
	echo PHP_EOL;
	$startTime = date_create($row['StartTime']);
	$offset = abs($startTime->getTimestamp() - $firstStartTime->getTimestamp());
	
	if(true && abs($continueOffset - $offset) > 1) {
		$continueOffset = $offset;
		echo '#EXT-X-DISCONTINUITY'.PHP_EOL;
		echo '#EXT-X-PROGRAM-DATE-TIME:'.date_format($startTime, $DATE_FMT).PHP_EOL;
	}

	$segmentPath = getEventDefaultVideoPath($row);
	if(!isset($_REQUEST['segtype']) || $_REQUEST['segtype'] != 'mp4')
		$segmentPath = 'index.php?view=m3u8&offset='.$continueOffset.'&mp4tots='.urlencode($segmentPath);

	echo '#EXTINF:'.$row['Length'].PHP_EOL;
	echo $segmentPath.PHP_EOL;
	$continueOffset += $row['Length'];
}
echo PHP_EOL;
echo '#EXT-X-ENDLIST'.PHP_EOL;
?>
```
To test the playback, I'm using this sample video-js player:
```html
<!doctype html>
<html>
  <head>
    <link  href="//vjs.zencdn.net/5.10/video-js.css" rel="stylesheet">
    <script src="//vjs.zencdn.net/5.10/video.js"></script>
    <script src="https://npmcdn.com/videojs-contrib-hls@^3.0.0/dist/videojs-contrib-hls.js"></script>
  </head>
  <body>
    <video id="my_video_1" class="video-js vjs-default-skin" controls
        xpreload="auto" width="352" height="288" poster="083-analyse.jpg"
        data-setup="{}">
        <source src="/zm/index.php?channel=1&begin=2017-01-02 12:40&end=2017-01-02 12:55&view=m3u8" 
				xsrc="http://d2zihajmogu5jn.cloudfront.net/bipbop-advanced/bipbop_16x9_variant.m3u8"
                type="application/x-mpegURL" 
                xtype='application/vnd.apple.mpegurl'/>
    </video>
	<script>
		var player = videojs('my_video_1');
		//player.play();
	</script>
  </body>
</html>
```
Note that video-js playback for this example is acceptable, but vlc and mpv have trouble syncing because of the timestamps.

connortechnology, 2017-01-08:
I'm not entirely sure what is possible.  

We really use ffmpeg in a minimal way, using the library, not the executable.  Also feature-h264-videostorage is not where we are doing our active development.  There is a definite problem with our .mp4 timestamps.  I use it in modect mode and the video is always 6-8 seconds late.  

I would like to work through this with you, but I'd like you to switch to using my storageareas ppa, as that is where I have done my work, and will continue to work.  

ppa.launchpad.net/iconnor/zoneminder-storageareas

mcbp223, 2017-01-09:
I got stuck initially at "Upgrading DB to 1.30.10 from "1.30.1", which is traceable to "ALTER TABLE Config MODIFY DefaultValue TEXT;" from the update sql script. I am using mariadb-server-10.0.28. I've tried several times, apt-purge / fresh restart / apt-get install and it always gets stuck, so I'm afraid that you will have reports coming from other people too. I tried issuing the statement in phpmysql and it got stuck too, in mysql command line and it got stuck, however it worked once in mysql command line, after a fresh restart. I had to comment out this statement inside the update script, so I could get past it.

Back to HLS, I will have to update my little php script since getEventDefaultVideoPath() changed the meaning of what it returns. I'll keep you updated once I get an understanding on how you are doing the segmenting.
My approach, before seeing how you did it, would be to do the processing after the fact; that is, let the segmenting process write everything to disk, then come with the motion detection on the pre-created on-disk segments in another process, kind of à la zmaudit, and either insert the associated rows into the database or delete the segment, based on the monitoring strategy.

connortechnology, 2017-01-09:
It has been on the todo list to be able to submit a video file for analysis.  Would be very useful in tuning zones, etc as well to re-analyze the stored event.  

I havn't had any complaints about that update script... there aren't too many people using my storagareas branch but it isn't zero either.  I'll do some testing.

mcbp223, 2017-01-13:
There is a bit of an amelioration if I call av_write_frame(oc, NULL) to flush buffers right at the beginning of the destructor of VideoStore. I wonder if enqueuing the output context allocation and stream creation to a separate thread could help with suppresing the delay.
Also there is a small bug in the timeline view in the storageareas branch, there is something wrong with the paths, if you can take a look.

connortechnology, 2017-01-13:
Interesting.  That makes some sense.  I have added an av_interleaved_write_frame call there whcih should do the same trick.

timeline view needs a lot of love.

connortechnology, 2017-01-13:
I think we have enough threads already.  

I suspect that you are more familiar with the ffmpeg libs and video storage in general.  I'm quite sure you will be able to help us get this code working better.

mcbp223, 2017-01-15:
Nevertheless, I did try to read the input in a separate thread, but no joy, the missing frames are somewhere else to be found. I am not familiar with the libs, but I'm trying hard.
Quick question, would it be possible, the way you see the videostorage feature, to drive the segment name change from zmc instead of from zma?

connortechnology, 2017-01-15:
Possibly.  

I think what is generally happening is that the zmc process is in the ffmpeg libs doing h264 deoding and whatnot... and zma triggers the new event and it takes several seconds for zmc to notice.  I think we need to be storing several seconds of pre-event frames and writing them out at the start of each event.  

Currently we are only buffering since the last keyframe.  We need to do more.

connortechnology, 2017-01-15:
Hey do you know what you are doing regarding transcoding ?  We need to transcode whatever sound the camera sends to AAC.  I have completely failed to achieve this.

mcbp223, 2017-01-15:
Given the complexity of the subject, I think that only a massive copypaste effort would magically solve all the transcoding problems. The file ffmpeg.c has 4600 lines worth of code that is not in the libs, but is very important at driving the process in the correct way. So now this is what I am up to. I have isolated this file and a few other required internal files, and succeeded at compiling them under the zmc project and giving it a run, but with the config.h file generated from the ffmpeg project for now. Now I want to plug some callbacks at the outputs somehow, and inform zma when there is a new image for analysis or a new segment. I'm thinking to try to set this up with three outputs for each input, one for images for analysis at the required fps, one for segments, and one for live view, which ffmpeg is said to be able to do smoothly.

SteveGilvarry, 2017-01-15:
I was doing the similar although with cut and paste from with FFmpeg 3.1 version of ffmpeg.c, 3.2 has some better usage of their new send and receive functions, but lots of changes in 3 in general and not many distros using yet. I haven't had the time lately so if you get something working would be good.

mcbp223, 2017-01-15:
Steve, what do you think about moving the segment name change logic from zma to zmc? Maybe this could give you more control on the segment duration, now their timings are too interrelated with the analysis process.
The reason why I was thinking to use the processing pipeline that's in ffmpeg.c was to decouple the analysis and the segmentation passthrough, since their implementation of one input multiple outputs seems to be made for this. And they seem to do it in a very elaborate way, as ffmpeg.c has over 4500 lines just to drive the libs, and the segmenter for example, libavformat/hlsenc.c has over 1000 lines of code, uncommented. However, if you could see a better fix for making the segment cuts without missing frames by looking at their segmenter implementation, I'd rather have it.

connortechnology, 2017-01-16:
I think zmc should be doing the segmenting.  The main reason being that we would no longer have to share all the frame img buffers.  zma just needs access to the current, frame image... and if someone was doing record... we wouldn't even need a zma process.
But we need to fix/improve the signalling between zma and zmc.  zma needs to send signals and they need to be picked up really quick by zmc.  zmf needs to go away.

The problem is that theoretically... the writing of data out can block... which on local cameras, or remote jpeg snapshot cameras... could delay grabbing the image.  

So maybe we need to be more multi-threaded in zmc.  the ffmpeg camera uses a separate thread .. maybe the local and remote_http versions need a seperate thread as well for grabbing the image. 

but I do agree that zma needs to be simplified.

Also.. currently we have a queue that queues frames since the last keyframe. This needs to be improved to queue a structure that contains the source raw fframe (avpacket, or whatever), and pointer to a raw decoded frame image (for zma purposes, or live streamining), a jpeg version (for live streaming), etc.  These pointers should only be populated if needed.... and this queue needs to queue a certain # of seconds.

mcbp223, 2017-01-16:
some quick questions:

- I am trying to do the processing with the code that's in ffmpeg.c, file that I will try to host verbatim inside the project, do you have any preference for a folder name under src for it, and a few other files internal to the ffmpeg project? And is it allright to compile against ffmpeg 2.8.10 that's in xenial-updates? I would like to use it as a library, but unfortunately this code is not in the libraries, and you can't load an executable as a library in linux without complications. Steve said that in version 3.2 they have implemented stuff that improve the dataflow, so that would be a target once that version gets mainstream. I've seen also some new hooks in it that would allow for reduced need to copypaste (you can replace AVFormatContext.io_open() with your own).

- is it ok to use features from C++11, and add set (CMAKE_CXX_STANDARD 11) in CMakeLists.txt? They have this move semantics that I would like to use when passing around objects that need to release resources once they go out of scope. Enqueuing images and other messages is very doable.

- what IDE's are people using around here, and what filediff tools? I'm using netbeans and meld for now, but I'm unhappy with netbeans because I can't launch the debugger under a different user, such as www-data, and also it doesn't pass environment variables to the program being debugged.

- when you say that zma needs to send signals that need to be picked up real quick by zmc, what do you mean? As I understand, the data would mostly flow one way, from zmc to zma, and zma would rarely need to signal anything to zmc, maybe just the termination?

- did you try segmenting with the ffmpeg command line from a camera with sound, to verify the AAC transcoding into the segments, that you mentioned before?

connortechnology, 2017-01-16:
If it is ffmpeg code, then stick it in an ffmpeg dir.  Yes 2.8.10 should be our target for now.

Yes, C++11 is fine and I think I've already turned it on in storageareas. Maybe I havn't merged that yet... 

I use vim.

zma does motion analysis.  So it needs to tell zmc when to start writing to disk and when to stop.

No I have not tried any command line ffmpeg use.

mcbp223, 2017-01-17:
Now I am having trouble at sending the data produced by ffmpeg to the correct FfmpegCamera instance, since the ffmpeg thread will be only one per zmc process, and there can be more than one input stream, each to be associated to one FfmpegCamera instance. I don't seem to be able to find where the input url is inside AVFormatContext that I get at the outputs. I am thinking to use as an identifying key the root part of the output path, the one that includes the monitor-id, such as /usr/share/zoneminder/www/events/2, where 2 is the monitor-id. To this, I can append the variable part, such as "/%Y-%m-%d/%H-%M/%S.%ff" to obtain the final deep path filename, to which I'll selectively add the extensions; they are needed by ffmpeg to select the output codec. But I need the root part separately, because there we can save the m3u8 playlist for live view, and it's location doesn't have to depend on the datetime.

For analysis, I assume that the bmp output codec is what you want, to avoid other conversions. The filtergraphs inside ffmpeg will take care of the scaling and framerate;  those can be different for each output, for example for analysis you can use half resolution images at 5fps, these settings are aready in the interface anyway, and for mp4 storage just copy everything from source as it is. Can you tell me how to save a bmp buffer to a zoneminder Image buffer, so I can run some tests with zma? I did some runs with two outputs, bmp images for analysis, for now saved to disk, and hls segmenting, and it solves my problem. The good thing with hls is that with one output you have in fact two, one is storage and one is live view.

mcbp223, 2017-01-18:
Ok regarding the key thing, I can set it in the metadata associated to each output, so I can get at it in the callbacks. The other questions regarding the paths remain, we need a separate location for a file live.m3u8 for each monitor, to allow for live streaming through this.

Would it be a stretch of imagination to consider storing the segment filenames in the Frames table, so you can have more than one per event? This would make it more like an attachements table.

SteveGilvarry, 2017-01-18:
I would have thought segmenting for hls playback would be an on demand type
thing in zms or its replacement. So event stores whatever container we like
single file per event, then that is run through segmenting code for
playback?

mcbp223, 2017-01-18:
@SteveGilvarry I don't know, I thought that streaming segments comes too cheap not to use it both for storage, playback and live streaming, this way you won't need segmenting on the fly for playback, just directly serve files. People using zoneminder in mocord or record would find it easier to seek back and forth through segments, video-js supports this; the standard recommends a 2 seconds segment length for quick loading, I think for mpeg-dash the default is one second.

BioSehnsucht, 2017-01-19:
How easy is it to convert segmented files back to a monolithic file if you need to (i.e., something happened, providing footage to police and/or social media)? Could a "download time range" function be put in so the user of ZoneMinder doesn't have to run some archaic ffmpeg command to assemble it?

mcbp223, 2017-01-19:
@BioSehnsucht this should be an option, not a change, maybe an additional type of input in the interface, at least until benchmarks come out. But "download time range" would work anyway, though I'm not sure if there is any place left for a new button in the interface. As for inode usage I don't know, my samples so far are around 200kB in size for 2 second segments at D1 resolution, with the I-frame set to come at 2 seconds as well. The lengths must be left tunable in the implementation, there is a setting in the interface that will be taken into account.  I do have an old Solaris 10 with ZFS, but I'm not sure if I could ever get the newer versions of ffmpeg to compile on it; I'll do some tests maybe through NFS when I get to.

SteveGilvarry, 2017-01-19:
Mike I will try and explain what I was thinking about doing and why, like the saying there is more than one way to skin a cat(What the cat did to them I don't know). My thoughts were to stay close to native outputs from cameras for least cost, and get a single file per event. I know segmenting won't add much so no worries there, and from what you are saying segmenting at 2 seconds I will get 30 files for every minute, plus an m3u8. Not as bad as jpegs but still  I really like how I get a single event file now.

I am also thinking about non h264 cameras, say older or newer that don't fit into the pretty narrow HLS requirements for codecs. So then we would need to be transcoding them on record to be streaming ready format. Whereas if we modify on playback only we could handle any format that FFmpeg could, yes probably means transcoding and segmenting on some cameras. But in this instance we are only doing it on playback, so assuming no one is watching all their recorded video realtime. What I do need to do is spend sometime understanding HLS, MPEG-DASH and MSE, to ensure the on demand model works, no point if we need all segments and complete m3u8 before streaming can begin, as the delay to convert will be ridiculous. So this is more analogous to Live streaming methods than pre-segmented complete files.

I was then thinking we had a model that could evolve with streaming standards, so my archived mpeg events would playback as HLS now, and later when say VP15 comes along we could switch streaming to that, always using the latest and greatest streaming option.

But not saying don't like what you are doing, pretty much same as me ripping ffmpeg.c apart, but I am focusing on stream copy functions and actually integrating them to zm. Just been super busy at work and now travelling on holidays so time is against me. The issues with zma controlling events does hit me also, but I haven't seen the delays that @connortechnology does on event change. Mine is more like event change taking place off I frame so I miss a few frames when compared to event jpegs, so need a avpacket buffer for pre-event value and a way to align event start with I Frame.

We could add a segmenting option on the storage tab per camera and control segment lengths etc.

mcbp223, 2017-01-19:
Steve, yes, my use case is that I need scrolling back and forth through the event video, just like you do it on youtube, and segmenting was the obvious solution. I would be very interested in getting 2 second chunks from a larger file on the fly, this would just simply solve my problem.
Once you get the time do some test runs with this code that I;ll post of skinning the cat using the ffmpeg engine, and not only the libraries, you're gonna like how easy it is to change options with just a few command line switches, instead of painstakingly preparing your buffers and calling the right functions with the right parameters at the right size and don't forget to release resources. It strikes me why they don't put the ffmpeg.c code in the libs, it would very much simplify the life of other devs using them. I mean really someones got to make the case to the fine guys at ffmpeg, libavengine would sound cool anyway.

BioSehnsucht, 2017-01-19:
Reformatting from regular H264 to HLS without re-encoding should be doable generally speaking (if using ffmpeg, just codec copy during the operation), I don't know if there's any gotchas though (like it only works if it was originally encoded with certain I-frame intervals or such). Could cache the segmented output for a user-configurable time so it only has to process once (though it should be a quick effectively realtime process), but not double disk space usage over time (because it would eventually get cleaned up if not looked at recently, leaving just the complete raw file)

As for my earlier comments about ZFS, come to think of it, probably even storing segments is a huge step up from current ZM practice of jpeg-per-frame. So I wouldn't worry too much about filesystem impact, comparatively speaking segments are probably dozens of times better than jpeg files. Steve's arguments for storing raw stream and segmenting on the fly do make sense though, but filesystem thoughts are overkill.

mcbp223, 2017-01-19:
Damn, there is a Pandora's box inside ffmpeg, look at the export_mvs flags2 option [https://ffmpeg.org/ffmpeg-codecs.html](url)

connortechnology, 2017-01-24:
Am a bit time restricted at the moment.  Maybe tomorrow.

BioSehnsucht, 2017-02-06:
I currently have ZM installed via the zm repo for CentOS7. If I wanted to try out this functionality, is the easiest way to do so to clone @mcbp223 's repo/branch and then build custom packages per http://zoneminder.readthedocs.io/en/stable/installationguide/redhat.html, remove the zmrepo ones, install custom ones? Or is there a 'quick and dirty' option I could use instead (i.e. './configure; make clean; make; make install') ?

mcbp223, 2017-02-07:
Hi, @BioSehnsucht my patch is against the [storageareas branch](https://github.com/connortechnology/ZoneMinder/tree/storageareas), and it is not compatible with master from official, you would have to take into account that there are changes in the database schema if you are using master on the system you are testing on. For my own tests I have installed prebuilt packages for the origin branch from the ppa, set up a repo with the patch applied, issued cmake with these flags, in my case they apply to an ubuntu setup, for CentOS7 some flags may be different:
`
cmake -DCMAKE_INSTALL_PREFIX=/usr -DZM_CONFIG_DIR=/etc/zm -DZM_RUNDIR=/var/run/zm -DZM_SOCKDIR=/var/run/zm -DZM_TMPDIR=/tmp/zm -DZM_CGIDIR=/usr/lib/zoneminder/cgi-bin -DZM_CONTENTDIR=/var/cache/zoneminder -DZM_WEBDIR=/usr/share/zoneminder/www .
` 
then
` 
make
` 
then you can either replace only src/zmc where the package installed it, and restart the zoneminder service, or test run it in place, along with zma in two separate terminals, with the zoneminder service stopped, with commands like
`
sudo -u www-data LOG_LEVEL=9 LOG_LEVEL_TERM=9 src/zmc -m 2 
` 
and 
`
sudo -u www-data LOG_LEVEL=9 LOG_LEVEL_TERM=9 src/zma -m 2 
` 
if you want to test for example the monitor number 2, where you can play with different log levels depending on what you want to debug. If you don't run zma it will not record, you'd be only able to see /usr/share/zoneminder/www/events/2/live.m3u8 file; I use mpv to play it.

I have not tested building the packages with zmrepo nor other build systems.

BioSehnsucht, 2017-02-10:
@mcbp223 Whew, that sounds like way more work than I was hoping for :D 

I imagine rather than managing DB schema changes coming from an already installed system, simply pointing it to a different DB and a fresh start would be easier? Or do I still need to manually run the schema patches either way (i.e., the schema changes won't happen automatically if starting with a fresh database / new install)? I don't currently have a "production" ZM system going so losing footage or even DB config is mainly just inconvenient at this time, rather than a critical issue.

Hopefully I'll find time to play with it this weekend.

mcbp223, 2017-02-10:
I believe that you can build from source following instructions from [here](http://zoneminder.readthedocs.io/en/stable/installationguide/redhat.html), just git clone from the repo you're testing. The installer should take care of upgrading the database schema to the version required. Just a note about the patch I pushed, it will launch a ffmpeg process to concatenate the segments, and it doesn't use an absolute path for it, so it must be reachable by environment path.

SteveGilvarry, 2017-02-10:
@mcbp223 is the hls streaming stuff in your repo, didn't see it in ffmpeg_engine branch? I am experimenting with movflags options to create the a fragmented mp4 instead of individual file segments. Keeps the single file I am looking for and mpeg-dash and hls both support it. Just need to check the files produced playback OK standalone also.

mcbp223, 2017-02-10:
@SteveGilvarry yes, if patch is activated with zm-method=ffmpeg-engine in the options tab, it creates a file, live.m3u8 in www/events/<monitor-id>, that's refreshed every few seconds; clients who can play the protocol know when to reload it. As of now the patch creates mp4 segments, not ts. I kept the logic that's in the application for now, with single file per event, by concatenating all pertaining segments at the end of the event. For my own needs I would like to have them stored as segments though, and be able to make a playlist between any start time and end time, irrespective of how motion events are grouping them. I haven't experimented with movflags options yet.

knight-of-ni, 2018-09-23:
I'm going to close this out because the conversation has gone long cold. The latest release supports recording direct to mp4, but I have not read this thread sufficiently to know if that solves the HLS recording issue or not. Feel free to create a new issue if further development is necessary.

closed 2018-09-23 by knight-of-ni