ZoneMinder/zoneminder #1740: 1.30.1 Release

knnniggett, 2017-01-10:
This is a place holder to document blockers for the upcoming 1.30.1 release.

- #1713, now closed, identifies an issue with the mmap files being orphaned in /dev/shm. We need to agree upon an approach to solve this.
- packpack automated builds. While there is no technical reason why we can't release before this is implemented, it is consuming my time at the moment. So from a time perspective I am focusing on completing this before my schedule frees up to work on the next release.

Please add additional blockers to this list.

knnniggett, 2017-01-12:
#1734 requires we bump our version to 1.30.2

knnniggett, 2017-01-24:
#1717 is the continuing saga to fix the recent mmap issue

knnniggett, 2017-01-25:
#1758 is a blocker

knnniggett, 2017-02-04:
I will create 1.30.2 Release Candidate 1 shortly

knnniggett, 2017-02-05:
1.30.2 RC1 has been released. Let the testing begin.
https://github.com/ZoneMinder/ZoneMinder/releases/tag/1.30.2-rc.1

connortechnology, 2017-02-05:
You are so good at doing the release text.  Bravo.

Just a couple of nitpicking points... I don't required you to change anything, just pointing them out in case you do want to change them.  

#1: There actually are new features in here.  The hikvision script and the keekoon scripts...and the display of shm memory... minor.. but technically new features.

#2: The other thing is when you say the developers don't get paid from financial contributions... that's not exactly true, as in the past we have applied some donations to bounties, and I think should continue to do.

knnniggett, 2017-02-05:
Thanks for the feedback. When I wrote that, I was struggling to remember what we did.
Yes, I considered the bounties and how to word the release text. Without making it too wordy... I wanted to make it say we don't earn a steady paycheck.

I'll go update it.

knnniggett, 2017-02-18:
@connortechnology I received a non-detailed report that the motion zones cause a crash of some sort when the image is rotated. Sounds like the motion zone polygons might not be rotating with the image and cause an out of bounds problem. This should be pretty easy to verify. Have you got a moment, to test for this? I'm tied up with other priorities at the moment.

knnniggett, 2017-02-20:
@connortechnology 
https://forums.zoneminder.com/viewtopic.php?f=36&t=25856

connortechnology, 2017-02-20:
I did a lot of work on Rotation several years ago.  I doubt it all got into master.

knnniggett, 2017-03-30:
1.30.2 has been released

closed 2017-03-30 by knnniggett

josh4trunks, 2017-04-25:
I was wondering... is there a reason the 1.30.2 release does not have some of the same downloads available as version 1.30.0 and earlier?

I was trying to update the FreeBSD port, which currently uses the below URL, but it was not working for version 1.30.2 (like it did for previous versions).
https://codeload.github.com/zoneminder/ZoneMinder/tar.gz/v1.30.2

Thanks

knnniggett, 2017-04-25:
We had to drop the "v" from our release tags in order to be compatible with our new automated build system.

The url you posted relies on us to create a tag with that name. You can see the tags we create on our releases page. Since the new tag is 1.30.2, the url becomes:
https://codeload.github.com/zoneminder/ZoneMinder/tar.gz/1.30.2

josh4trunks, 2017-04-25:
makes sense, thanks I'll try that