ZoneMinder/zoneminder #1748: Event(s) not deleted from disk when deleted in Zoneminder

doublehp, 2017-01-12:
This is a followup of
https://github.com/ZoneMinder/ZoneMinder/issues/748

In short: zmaudit.pl does not clean it all.

***

I just ran /usr/bin/zmaudit.pl manualy.

According to ZM web interface, the oldest event for camera 8 is 2016-dec-14 th.

According to filesystem ...

546# ls /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8
16 17
547# ls /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8/16/12/
08 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31
548# ls /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8/16/12/08/
18
549# ls /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8/16/12/08/18/
42
550# ls /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8/16/12/08/18/42/
55
552# ls /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8/16/12/08/18/42/55/ | wc
615 615 11070

So I have an event for dec 8th that is out if all ZM dbs. It may never be deleted by any later script.

This is an issue for me; I have writen a script that relies on the filesystem to determine the oldest even of a camera. This kind of error by ZM will produce false resuts for me.

Several cameras have similar residues.

knnniggett, 2017-01-14:
It's working fine here. 

This is going to take quite a bit more information about your system to get to the bottom of this.

- What distro and version of distro are you running?
- How did you install zoneminder? zmrepo? ppa? from-source???
- If you are running something other than the official 1.30.0 release then we need to know about it.
- What is the value of DIR_EVENTS?
- What is the value of OPT_FAST_DELETE?
- What is the value of AUDIT_MIN_AGE?
- Can you reproduce this issue every time or just some of the time?
- How exactly are you deleting the events in question from the database? Does the problem occur if you select an event from the web console and hit delete?
- Please show the actual `ls -lh` content of one of the event folders in question
- Please describe in detail how you have your events folder is set up. 
- Is your events folder filesystem something other than ext4?
- Run zmaudit with Debug turned on and post the results both from what you see on the command line and what shows in the zm log. This will likely be on the order of 1000+ events so please create a pastebin/gist and link to it.

I know I'm asking for a lot of information, and I will probably ask for more. However, this is what it is going to take to truly understand how you've got your system set up and what is going on.

I also need to state that this is going to require full cooperation from you. This has more to do with those who have come before you, as many seem to think we can just wave a magic wand and fix the issue with no work on their part. This isn't going to be solved that way.

As a side note, we now have an API which is intended to allow others to write third party applications. Moving forward, this is what we recommend you use. Since the API is relatively new, it might very well not be able to do exactly what you want, but we can deal with that in a separate issue.

doublehp, 2017-01-15:
debian/ stretch

aptitude

v1.30.0

events

ticked

86400 (seems to be a few days, no more than 8 from memory)

random; some events are left; most are removed. Now you ask this ... Ihave an idea; see below. I have found 6 old events between 1st and 15th of december, which should have been removed. I have filters to remove old events when they are more than 30 days, and when disk is full. When I reported this bug, I had 6 events over 4 days that were "alone". All event were already deleted for the next days (when I removed 15th dec, events for 16th and 17th were already removed: 18th had a few hundred events; while 15th had only one). I also may remove some events manually. I have a manual filter to find event with more than 1000 images; once a week, I look at them, and remove events that use disk for nothing. - I have that cron line forums recommend to have, to run daily the cleaning script.

I don't understand the question. I never talk to the DB directly. I only use the ZM web interface. Automated filters remove >30d events, and when disk touches 95%. Manually remove events that have >1000 frames (30mn falling snow is very nice in HD; completely useless with cheap cam). And that cron line: */10 *  * * *   root    /usr/bin/zmaudit.pl >/dev/null 2>&1

NOW is irrelevant; I remove them to stop corrupting my stats; I will keep them when it occurs again (munin will spot them easily: one cam will have the oldest event age up to 40, while other cams will stick to "oldest=30" - then I know i have to check the filesystem). But permissions are fine (owner and rights).

I moved the event folder; it was too big for the main disk, so I moved it to SSHFS; during the move, I checked the perms, and fixed them. This occured in october-novemeber. Left-over events I am talking about were from december.  Details of migration are irrelevant; I found bugs in NFS and samba (both are crap); so I am now using sshfs.

Now, yes. May change in the future, because I am not happy with sshfs. 

Now is irrelevant, but, next time I see left-over events, I will.

***

NB: main filters are set to run continuously; and this usually works fine. Disk never exceeded 95%; and most cams don't show events older than 30d as I set.

Left-over events I talk about are only visible in filesystem; they are not visible in any event webpage.

Problem with local ext4: ZM is running inside a Virtualbox, forwarding the physical disk to guest seemed to bug; had too many errors, moved to network FS.
https://forums.virtualbox.org/viewtopic.php?f=1&t=80795

Problem with NFS: can't set lock (not sure if it was ZM or an other service needing the same mount point.
http://forums.debian.net/viewtopic.php?f=5&t=130379
http://forums.debian.net/viewtopic.php?uid=198372&f=5&t=130412&start=0

Problem with Samba: could not even configure it to allow ZM to write on it. Maybe I am stupid; spent one or two full days on it ...

sshfs worked in 2h. But sshfs also has hangs. I regularly need to restart ZM, and remount the disk.

But other services using the mount point don't complain, and don't have side effects.

Left-over events could match system freeses (VM guest), or sshfs bugs.

I have also written a script to track lost events; I changed it to stop deleting them, and report them to me. Next time I have one, I will receive an email, and can come back here.

Even if this is due to my system instability, you could:
- keep event in internal DB as long as files are not SUCCESSFULLY deleted. When sshfs is broken, you may fail to see the files associated to event, because ls won't return any output; but in such a case, ls will return a FALSE value. Wait one day, the system will fixitself, and when files come back, a new attempt to delete them will succeed. Check the existance of parent folder. If camera folder does not exists, then, filesystem is gone away for a while; it will come back soon.

I have seen in zoneminder webinterface that some events don't have any frame. The events I talk about in this bug do have frames in !!! Many image files.

- you could also ... cross check if all event folders match an event entry in the DB. Then, let admin set a policy for them (either ignore, report, or remove). You don't need to perform the check agains the whole disk, or all events. Just ask the filesystem to check the 10 or 100 oldest events. If one or two are unfoundable in your DB, apply the policy. If most of them are unfoundable, then, there is a bigger problem. Seeking for just 10 or 100 events (or, 10 per cam) does not take long. 

I seek for lost events with something like this:
find "${events_folder}" -mindepth 5 -maxdepth 5 -ctime +40

***

Tips for me for next time i see the bug:
- "ls -lh content of one of the event folders in question"
- "Run zmaudit with Debug turned on and PASTEBIN the results both from what you see on the command line and what shows in the zm log. 1000+ lines expected"

doublehp, 2017-01-19:
Luck day; got one. 

```
508# cd /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8/16/12
509# ls -lh
total 48K
drwxr-xr-x  8 root root 4.0K Jan 16 00:11 17
drwxr-xr-x 15 root root 4.0K Dec 21 22:49 21
drwxr-xr-x 11 root root 4.0K Dec 24 20:58 22
drwxr-xr-x 12 root root 4.0K Dec 23 22:41 23
drwxr-xr-x 11 root root 4.0K Dec 24 22:47 24
drwxr-xr-x 11 root root 4.0K Dec 25 22:10 25
drwxr-xr-x 14 root root 4.0K Dec 26 18:06 26
drwxr-xr-x 14 root root 4.0K Dec 27 17:16 27
drwxr-xr-x 12 root root 4.0K Dec 28 19:28 28
drwxr-xr-x  5 root root 4.0K Dec 29 14:35 29
drwxr-xr-x  9 root root 4.0K Dec 30 18:16 30
drwxr-xr-x 11 root root 4.0K Dec 31 23:48 31
510# ls -lh 17
total 24K
drwxr-xr-x 5 root root 4.0K Jan 16 00:11 17
drwxr-xr-x 5 root root 4.0K Jan 16 00:11 18
drwxr-xr-x 4 root root 4.0K Jan 16 00:11 19
drwxr-xr-x 6 root root 4.0K Jan 16 00:11 20
drwxr-xr-x 3 root root 4.0K Jan 16 00:11 21
drwxr-xr-x 3 root root 4.0K Dec 18 21:55 23
511# ls -lh 17/17/
total 12K
drwxr-xr-x 3 root root 4.0K Dec 17 17:33 31
drwxr-xr-x 3 root root 4.0K Dec 17 17:34 34
drwxr-xr-x 3 root root 4.0K Dec 17 17:42 42
512# ls -lh 17/17/31/
total 4.0K
drwxr-xr-x 2 root root 4.0K Dec 17 17:31 25
513# ls -lh 17/17/31/25/
total 80K
-rw-r--r-- 1 root root 79K Dec 17 17:31 00057-analyse.jpg
514#
514# ls -lh 17/20/47/03/
total 22M
-rw-r--r-- 1 root root  97K Dec 17 20:47 00087-capture.jpg
-rw-r--r-- 1 root root  98K Dec 17 20:47 00088-capture.jpg
-rw-r--r-- 1 root root  99K Dec 17 20:47 00089-capture.jpg
-rw-r--r-- 1 root root  99K Dec 17 20:47 00090-capture.jpg
-rw-r--r-- 1 root root 100K Dec 17 20:47 00091-capture.jpg
-rw-r--r-- 1 root root 100K Dec 17 20:47 00092-capture.jpg
-rw-r--r-- 1 root root 100K Dec 17 20:47 00093-capture.jpg
-rw-r--r-- 1 root root 101K Dec 17 20:47 00094-capture.jpg
-rw-r--r-- 1 root root 101K Dec 17 20:47 00095-capture.jpg
-rw-r--r-- 1 root root 102K Dec 17 20:47 00096-capture.jpg
-rw-r--r-- 1 root root 102K Dec 17 20:47 00097-capture.jpg
[...]
516# ll
total 48
drwxr-xr-x  8 root root 4096 Jan 16 00:11 17
drwxr-xr-x 15 root root 4096 Dec 21 22:49 21
drwxr-xr-x 11 root root 4096 Dec 24 20:58 22
drwxr-xr-x 12 root root 4096 Dec 23 22:41 23
drwxr-xr-x 11 root root 4096 Dec 24 22:47 24
drwxr-xr-x 11 root root 4096 Dec 25 22:10 25
drwxr-xr-x 14 root root 4096 Dec 26 18:06 26
drwxr-xr-x 14 root root 4096 Dec 27 17:16 27
drwxr-xr-x 12 root root 4096 Dec 28 19:28 28
drwxr-xr-x  5 root root 4096 Dec 29 14:35 29
drwxr-xr-x  9 root root 4096 Dec 30 18:16 30
drwxr-xr-x 11 root root 4096 Dec 31 23:48 31
```

Several days are missing: 18-20. Day 17 has events during 6 hours. Each hour contains several events. Not giving details.

```
# cat /var/log/zm/zm_debug.log.4946
01/19/17 09:23:05.828811 web_php[4946].DBG [LogOpts: level=DBG/DBG, screen=OFF, database=INF, logfile=DBG->/var/log/zm/zm_debug.log.4946, weblog=INF, syslog=INF] at /usr/share/zoneminder/www/includes/logger.php line 168
```

```
502# zmaudit.pl >/tmp/log 2>&1
503# cat /tmp/log
Database event '11/122521' does not exist in filesystem but too young to delete.Database event '18/122340' does not exist in filesystem but too young to delete.Deleted 9 log table entries by time
```

Event page in web interface:

```
Id 	Name 	Monitor 	Cause 	Time(^) 	Duration 	Frames 	Alarm
Frames 	Total
Score 	Avg.
Score 	Max.
Score 	
102694 	Event-102694 	NVT_ZM-01_cam-02_annke_jardin-est_38 	Motion 	12/21 05:57:04 	00:03:30 	753 	285 	829 	2 	17 	
102777 	Event-102777 	NVT_ZM-11_cam-03_jidetech cours_42_depuis_gouttiere 	Motion 	12/21 08:05:57 	00:02:19 	408 	89 	1136 	12 	31 	
```

So, they are sorted by date (all cams together), oldest first; and oldest know is from dec 21st.

Dec 17th is clearly ... a lost date.

I had ticked debug, but left level 1; you did not ask any particular level.

Assuming linear occurence of events ... since 21st is still here, 17th should have been removed 3d ago. During the last 3 days, system had 6 crashes (freese/reboot - i monitor system uptime), and ZM service was restarted 14 times (if you substract the 6 crashes, that's 8 service restart for live system: ZM service crashed 8 times - i monitor age of ZM PID)

How age of PID is calculated:
```
pid=$(/bin/systemctl status zoneminder.service | grep "Main PID" | awk '{print $3}')
        out="$(ls -lt --full-time /proc/${pid}/cmdline 2>/dev/null | tail -n 1 | awk '{print $6 " " $7 }')"
        [ "$out" != "" ] && {
        outs=$(($(date +%s)-$(date +%s --date "$out")))
        outh=$((($(date +%s)-$(date +%s --date "$out"))/(3600)))
                true
        } || {
                outs=0
                outh=0
        }
```

So I am not excluding the possibility you may remove events from DB before removing files; if crash happens during removal, it would explain my issue. => remove files before updating DB. Or cross check that all event folders match an event in DB.

What I can't know is if this issue may happen during service crash, or system freese. I would need a higher level or jounalising to know this; check exactly on my filesystem at which moment you tried to remove other events for the same day; and following day. Unless you can have this in your debug logs. You seem to keep logs only 7 days; this problem occured already 3d ago; answer me before 3d, and tell me where is the debug info about file removal.

doublehp, 2017-01-19:
I can get the warning faster next time; actual script is waiting 3 days to remove and report issue (disabled removal for now); I could make it 2 or 1d.

knnniggett, 2017-01-19:
- uncheck OPT_FAST_DELETE.

- I didn't specify the level of debug because it is irrelevant. The perl scripts only support debug on or off. 
So again, turn on debug and show us the 1000+ lines *zmaudit* produces in the zm log when it runs and it finds the orphaned records in question.

-Your event folder filesystem isn't right, which is why I asked you to tell us exactly how you set it up. We don't need to know what you've tried in the past, but how it is set up right now, including the filesystem, how you are mounting it, and what mount parameters you used. Everything in the events folder should be owned by the web account user. You have to use a linux friendly file-system and if that file-system is remote then use a 100% compatible file sharing mechanism... iscsi would be the fastest, and NFS would be good. Don't use Samba... it's intended for Windows file sharing and Windows does not support symlinks. Don't use sshfs, or weddav, or any other userspace filesharing mechanism.

> I don't understand the question. I never talk to the DB directly. I only use the ZM web interface. Automated filters remove >30d events, and when disk touches 95%. Manually remove events that have [...]

You told us you wrote a third party app, and the only thing you told us about it is that is scans the event folder. Other than that, we have zero information telling us what it does, hence the question about how you are deleting events. We can't possible know this until you tell us.

doublehp, 2017-01-19:
Fast unticked; debug ticked. Here is command in console:

```
# zmaudit.pl
Database event '11/122521' does not exist in filesystem but too young to delete.Deleted 2 log table entries by time
```

I don't understand where is the log file you want. After running the command, /var/log/zm/zmaudit.log was not updated (last change 8h ago). The only recent file I can find is

```
# ll /var/log/zm/zm_debug.log*
-rw-r--r-- 1 www-data www-data   220 Jan 19 18:25 /var/log/zm/zm_debug.log.2253
-rw-r--r-- 1 www-data www-data   440 Jan 19 18:24 /var/log/zm/zm_debug.log.2545
-rw-r--r-- 1 www-data www-data  8238 Jan 19 09:27 /var/log/zm/zm_debug.log.26795
-rw-r--r-- 1 www-data www-data  7431 Jan 19 18:25 /var/log/zm/zm_debug.log.26893
-rw-r--r-- 1 www-data www-data 34035 Jan 19 10:04 /var/log/zm/zm_debug.log.27085
-rw-r--r-- 1 www-data www-data     0 Jan 19 18:25 /var/log/zm/zm_debug.log.29595
-rw-r--r-- 1 www-data www-data   220 Jan 19 09:23 /var/log/zm/zm_debug.log.4946
-rw-r--r-- 1 www-data www-data   218 Jan 19 09:28 /var/log/zm/zm_debug.log.994
# cat /var/log/zm/zm_debug.log.2545
01/19/17 18:20:41.821237 web_php[2545].DBG [LogOpts: level=DBG/DBG, screen=OFF, database=INF, logfile=DBG->/var/log/zm/zm_debug.log.2545, weblog=INF, syslog=INF] at /usr/share/zoneminder/www/includes/logger.php line 168
01/19/17 18:24:01.820023 web_php[2545].DBG [LogOpts: level=DBG/DBG, screen=OFF, database=INF, logfile=DBG->/var/log/zm/zm_debug.log.2545, weblog=INF, syslog=INF] at /usr/share/zoneminder/www/includes/logger.php line 168
# cat /var/log/zm/zm_debug.log.29595
[...]
# cat /var/log/zm/zm_debug.log.29595 | pastebinit
http://slexy.org/view/s20E78La9F
```
About storage disk, I did not change any ZM setting. While service was down, I moved the folder from  /var/cache/zoneminder/ to /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/ , and replaced the first name by a symling pointing to the second. This folder is now located on an SSHFS mount point. Had a few issues about permissions and locks; now works fine 99.999% of the time. On ssh server side, that disk is ext4.

SSHFS Mount options:
vol@server:/mnt/Leon_Host_Big2 on /mnt/Leon_Host_Big2 type fuse.sshfs (rw,relatime,user_id=0,group_id=0,allow_other)

Server for physical disk:
/dev/sdb3 on /mnt/Leon_Host_Big2 type ext4 (rw,relatime,data=ordered)

Never heard of ISCSI; NFS failed (spent days trying to get it work; does not support locks; links above); Samba failed. 

The normal deletion procedure is to use ZM internals: default filters via web interface. Since this did not worked properly, forums told me to add a line in cron. From there, ZM could delete most events corectly; but randomly leaves some around.

I wrote scripts to:
- count events per hour for munin (based on file system)
- measure age of daemon for munin (based on /proc)
- find events older than the limit set in the web interface (it's fuzzy; most cams are allowed to retain events for 30 days; but when disk reaches 92%, I  remove events for the busy cam). (base on filesystem)

But, Munin reported events older than 30d, what is impossible because of ZM filters. When digging, I found that the filesystem had records and events for dates that were out of scope of ZM, and those events were not listed in the even page. My script is designed to delete such events; but to help you debug it all, I commented the deletion line, so that they are just detected, but kept for now.

In the end, my script only make read only operations on the filesystem, they do not write or delete anything for now. It's a heavy bunch of ps, ls, find, grep. The process one is running on the VB guest running ZM ; filesystem oriented scripts run faster on the VB host (direct access to ext4; saves SSHFS layer).

... reading a bit about iscsi: not possible; my purpose is to use a space from a shared disk; I can not make that disk exclusively used by that VB guest; the VB host handles the FS, and then VB guests use a network FS. The disk is used by other hosts as well.

knnniggett, 2017-02-01:
There is nothing posted here to indicate this is anything other than a configuration issue, and I am not willing to troubleshot further due to OP's attitude. Feel free to continue this in the user forum.

closed 2017-02-01 by knnniggett