ZoneMinder/zoneminder #1750: Add runstate to filters

damnthetorpedos, 2017-01-13:
I would like to see runstate as an option for a filter. 

Others have asked for similar here:
https://forums.zoneminder.com/viewtopic.php?f=30&amp=&t=21459
and
https://forums.zoneminder.com/viewtopic.php?f=29&amp=&t=21702

It would be in the drop down along with items such as: "Monitor ID, Monitor Name, ID, Name, Cause, Notes, etc."

damnthetorpedos, 2017-05-14:
I tested this on 1.30 and it appears to work as expected. I didn't do extensive testing. I'll be testing this on 1.30.2 next. Is there anything I can do to help or get this worked in? I could increase the bounty, if that helps. I don't have any useful programming experience to help with.

P.S. I don't like feature creep and bloat, so I'd understand if this was an optional feature, or patch. If it has to be hacked in, then so be it, but I figure I should check.

connortechnology, 2017-05-15:
I think it just needed a little polish... I will take a look at it and refresh my memory today. Should be trivial to get it merged to master soonish.

closed 2018-06-22 by stale[bot]