ZoneMinder/zoneminder #1760: comparison between pointer and zero

abishai, 2017-01-31:
ZoneMinder can't be compiled with CLANG 4.0 It bumped severity of such comparison to error. 
src/zm_monitor.cpp contains comparison between (unsigned char*) mem_ptr and 0 in several places.
```
src/zm_monitor.cpp:197:18: error: ordered comparison between pointer and zero
('unsigned char *' and 'int')
    if ( mem_ptr > 0 )
	 ~~~~~~~ ^ ~
```

connortechnology, 2017-02-02:
The line should probably like this:
if ( mem_ptr > (void *)0 ) {

abishai, 2017-02-02:
Maybe, !=0 is OK too.

connortechnology, 2017-02-02:
No because failure to mmap results in -1.

abishai, 2017-02-03:
I added fix to my FreeBSD PR. Compiles on my test system with your fix.

mastertheknife, 2017-03-23:
What about comparing it to NULL?
NULL is typically defined as (void*)0, so mem_ptr > NULL should work.

arvidn, 2017-11-01:
for the record, ``mmap()`` returns ``MAP_FAILED`` on error (which may or may not be the same value as ``(void*)-1``). [posix spec.](http://pubs.opengroup.org/onlinepubs/009695399/functions/mmap.html#tag_03_379_04)

connortechnology, 2017-11-03:
It seems like we should follow posix.. defined constants are both readable... and defined.. hence... good.. Mmkay?

closed 2018-06-22 by stale[bot]

ali-babaei, 2018-09-26:
> The line should probably like this:
> if ( mem_ptr > (void *)0 ) {

on:
Telegram Android Client

I had following error:

jni/./gifvideo.cpp:151:28: error: ordered comparison between pointer and zero ('AVStream *' and 'int')
    if (info->video_stream <= 0) {
        ~~~~~~~~~~~~~~~~~~ ^  ~

Thanks @connortechnology, your solution worked for me too.

Just commented here for everybody facing this issue later!