ZoneMinder/zoneminder #1762: kill a non-numeric process ID at /usr/bin/zmdc.pl line 558

doublehp, 2017-02-01:
Google tells me nobody ever had this issue, am I really alone in the world ?

Message from syslogd@hostname at Dec 29 07:01:35 ...
 n't kill a non-numeric process ID at /usr/bin/zmdc.pl line 558, <CLIENT> line 1.

Several times a day, everyday. In my console. Dispite message ... message does not appear in /var/log/messages:
grep "kill a non-numeric process"  /var/log/zm/

So:
1: if there is an issue, send it to disk logs, not only to console. Walls should be reserved for critical problems; they are messing the terminal, and should be used with caution. If it's critical, log it to disk. If it's not, don't bother logged-in people.
2: fix the root issue

 v1.30.0

PRETTY_NAME="Debian GNU/Linux stretch/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

knnniggett, 2017-02-01:
There is no data in this post to indicate there is any kind of we can fix in the source code.

While new bugs pop up from time to time, this is not one of them. The fact that Google doesn't reveal other issues is a big first clue that you should be looking into how your system is configured.

Hint:  As you can see from what you posted the message is coming from syslogd, not zoneminder, so you need to look into why syslogd is trying to kill or restart zoneminder. Hwoever, this is not the place to troubleshoot this. Please take this and any other configuration issues to the user forum.

connortechnology, 2017-02-01:
There should be a corresponding log line in wherever you have set to log at the warning level that says something like:
[process] has not stopped at blah blah Sending KILL to pid [pid] 

what does it say for [pid] ? probably nothing.

Sounds to me like there is a bug somewhere, in zmdc.pl where it is losing track of processes or something.  Since it happens frequently we should be able to get to the bottom of it.

doublehp, 2017-02-01:
"The fact that Google doesn't reveal other issues" means it could be very recent, or that nobody looks at console, or that most distributions disable walls, or that syslog does not broadcast for other people ...

Is bug in system ? I don't know; syslog reports that application /usr/bin/zmdc.pl at line 558 produced an error; so, this line is:
kill( 'KILL', $$process{pid} );

ok, so, strictly, the error originates from kill; does it mean the error is in kill ? I don't think so; is the error in syslog ? 

Am I the only one complaining about it ? Not exactly: this post mentions a different line:
https://forums.zoneminder.com/viewtopic.php?t=24549
He seems to have it from /var/log ... 

The message has been CATCHED BACK by syslog, because zoneminder was unable to handle it internally. 

I have not set logs any particular way; it's just default.

But since you just assume your code is just perfect and stable, and that new bugs are almost impossible to discover, fine, I will fix this for myself, and stop bothering BTS. Nobody here seems to care about bugs or user feedback. I have found more than a dozen defects in ZM, and nobody cares. It's not better on forum. I have kags, crashes, and white pages, but fine, if it's impossible to meet new bugs, I will just ignore all problems I meet, and, probably move to an other software when I am bored.

Seems recurrent with dev using Github; never saw dev refusing to fix bugs in projects using other BTS.

I had half a dozen other issues to talk about, I think I will just save my time chutting up, and forget about ZM. I will just tell my familly to pay for real working solution, like HIK-Vision or Dahua.

connortechnology, 2017-02-01:
@doublehp I am willing to help with whatever problems you are having.  Please read and respond to my post above.

knnniggett, 2017-02-01:
@doublehp 
It's your entitled attitude which precludes you from getting help. We have a clearly defined set of posting rules:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

And simply posting a single error message, with zero context, does not in any way mean you've got a bug. No one here has made any statement our code is perfect. Rather we stated that, if you do want help, then you need to do it in the user forum as it does not meet the threshold of a bug.

closed 2017-02-01 by knnniggett