ZoneMinder/zoneminder #1763: Crash of nph-zms

mohahn, 2017-02-03:
I'm using OpenMediaVault 3 based on Debian Jessie and have installed ZM 1.29 via default Debian repo. Only thing is I'm using Nginx as webserver with installed fcgiwrap

```
root@zeus:/var/log# uname -a
Linux zeus 4.8.0-0.bpo.2-amd64 #1 SMP Debian 4.8.15-2~bpo8+2 (2017-01-17) x86_64 GNU/Linux

root@zeus:/var/log# cat /etc/nginx/openmediavault-webgui.d/zm.conf
location /zm/cgi-bin {
        gzip off;
        alias /usr/lib/zoneminder/cgi-bin;
        include /etc/nginx/fastcgi_params;
        fastcgi_param SCRIPT_FILENAME $request_filename;
        fastcgi_pass unix:/var/run/fcgiwrap.socket;
}
location /zm {
        gzip off;
        alias /usr/share/zoneminder/www;
        index index.php;
        location ~ \.php$ {
                if (!-f $request_filename) { return 404; }
                expires epoch;
                include fastcgi_params;
                fastcgi_param SCRIPT_FILENAME $request_filename;
                fastcgi_index index.php;
                fastcgi_pass unix:/var/run/php5-fpm.sock;
        }
        location ~ \.(jpg|jpeg|gif|png|ico)$ {
                access_log off;
                expires 30d;
        }
}

```
At present I've attached only one Hikviksion DS-2CD2342WD-I and setup two monitors pointing to same device:

Monitor 1: Remote, Modect, Source HTTP/Simple, 640x360px, 32 bit
Monitor 2: Remote, Nodect, Source RTP/RTSP, 1920x1080px, 32 bit, linked to Monitor 1

Most time all is fine, analysis and recording works and I can view events.
But, there are some events that cannot be viewed in browser. I get only a broken image in browser for them. I've no idea what could cause this error. It does not seem to be related to length of recorded event or daytime. But when an event cannot be viewed, then this applies to both monitors (means I cannot watch this event for both - maybe as they are linked?). Most interesting thing is, after opening frame layer, I can view each frame of this event seperately without problem. So it is recorded (as each frame exists with a valid picture) but cannot be streamed. Restart of ZM doesn't fix it.

```
root@zeus:/var/log# tail -n 30 /var/log/syslog
Feb  3 07:47:43 zeus [4932]: ERR [Got signal 6 (Aborted), crashing]
Feb  3 07:47:43 zeus [4932]: ERR [Signal address is 0x2100001344, from 0x7f8f71a26067]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 0: /usr/lib/zoneminder/cgi-bin/nph-zms(_Z14zm_die_handleriP9siginfo_tPv+0x78) [0x55a17ce5a028]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0xf890) [0x7f8f74b99890]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 2: /lib/x86_64-linux-gnu/libc.so.6(gsignal+0x37) [0x7f8f71a26067]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 3: /lib/x86_64-linux-gnu/libc.so.6(abort+0x148) [0x7f8f71a27448]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 4: /usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x15d) [0x7f8f72313b3d]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 5: /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x5ebb6) [0x7f8f72311bb6]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 6: /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x5ec01) [0x7f8f72311c01]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 7: /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x5ee19) [0x7f8f72311e19]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 8: /usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZSt19__throw_logic_errorPKc+0x66) [0x7f8f723678b6]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 9: /usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSs12_S_constructIPKcEEPcT_S3_RKSaIcESt20forward_iterator_tag+0x25) [0x7f8f72373805]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 10: /usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSsC2EPKcRKSaIcE+0x36) [0x7f8f72373c46]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 11: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN7Monitor4LoadEjbNS_7PurposeE+0x282) [0x55a17cdff952]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 12: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN10StreamBase11loadMonitorEi+0x2e) [0x55a17ce5c28e]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 13: /usr/lib/zoneminder/cgi-bin/nph-zms(main+0xdd5) [0x55a17cdc28c5]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 14: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f8f71a12b45]]
Feb  3 07:47:43 zeus [4932]: ERR [Backtrace 15: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x1e2d7) [0x55a17cdc32d7]]
Feb  3 07:47:43 zeus [4932]: INF [Backtrace complete, please execute the following command for more information]
Feb  3 07:47:44 zeus [4932]: INF [addr2line -e /usr/lib/zoneminder/cgi-bin/nph-zms 0x55a17ce5a028 0x7f8f74b99890 0x7f8f71a26067 0x7f8f71a27448 0x7f8f72313b3d 0x7f8f72311bb6 0x7f8f72311c01 0x7f8f72311e19 0x7f8f723678b6 0x7f8f72373805 0x7f8f72373c46 0x55a17cdff952 0x55a17ce5c28e 0x55a17cdc28c5 0x7f8f71a12b45 0x55a17cdc32d7]
Feb  3 07:47:45 zeus web_php[22993]: ERR [socket_sendto( /var/run/zm/zms-428999s.sock ) failed: No such file or directory]
Feb  3 07:47:45 zeus web_php[25210]: ERR [getCmdResponse stream error: socket_sendto( /var/run/zm/zms-428999s.sock ) failed: No such file or directory - checkStreamForErrors()]
```

Can someone help?

mohahn, 2017-02-03:
Ok, I found a workaround and you will not believe what I'm going to tell you, as it makes this problem more spooky. Setting one option in web frontend helps for viewing these broken events. IMO it's not a fix but a workaround.
I've enabled only one option: LOG_DEBUG. After restarting ZM I can watch the broken events. And this behavior is repeatable. When disabling LOG_DEBUG again (and restarting ZM) - events are broken, no matter how often I restart ZM with disabled debug log. Enabling LOG_DEBUG again - all is fine.

And debug log show nothing helpful. First log is the broken one and second is for an event which is ever fine, no matter if debugging is enabled or not. As you can see - no differences. So what can cause an enabled debug log to fix this?

```
02/03/17 08:14:02.825200 undef[7078].DB1-zm_logger.cpp/234 [LogOpts: level=DB1/DB1, screen=OFF, database=INF, logfile=DB1->/var/log/zm/zm_debug.log.07078, syslog=INF]
02/03/17 08:14:02.825316 undef[7078].INF-zm_config.cpp/139 [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
02/03/17 08:14:02.935041 zms[7078].DB1-zm_logger.cpp/234 [LogOpts: level=DB1/DB1, screen=OFF, database=INF, logfile=DB1->/var/log/zm/zm_debug.log.07078, syslog=INF]
02/03/17 08:14:02.935143 zms[7078].DB1-zm_utils.cpp/262 [Detected a x86\x86-64 processor with SSSE3]
02/03/17 08:14:02.935244 zms[7078].DB1-zms.cpp/102 [Query: source=event&mode=jpeg&event=170&frame=1&scale=100&rate=100&maxfps=30&replay=single&auth=XXX&connkey=613998&rand=1486106042]
02/03/17 08:14:02.935780 zms[7078].DB1-zm_user.cpp/172 [Attempting to authenticate user from auth string 'XXX']
02/03/17 08:14:02.936734 zms[7078].DB1-zm_user.cpp/235 [Checking auth_key 'XXX' -> auth_md5 'XXX']
02/03/17 08:14:02.936799 zms[7078].DB1-zm_user.cpp/241 [Authenticated user 'admin']
02/03/17 08:14:02.940061 zms[7078].DB1-zm_monitor.cpp/2738 [Got 1 for v4l_captures_per_frame]
02/03/17 08:14:02.940475 zms[7078].DB1-zm_monitor.cpp/390 [monitor purpose=0]
02/03/17 08:14:02.940533 zms[7078].DB1-zm_monitor.cpp/398 [mem.size=46081760]
02/03/17 08:14:02.940577 zms[7078].DB1-zm_monitor.cpp/449 [Monitor Terrasse-MTN has no zones, adding one.]
02/03/17 08:14:02.940683 zms[7078].DB1-zm_zone.cpp/51 [Initialised zone 0/All - 1 - 640x360 - Rgb:0000ff, CM:3, MnAT:15, MxAT:0, MnAP:50, MxAP:75000, FB:3x3, MnFP:50, MxFP:50000, MnBS:10, MxBS:0, MnB:0, MxB:0, OF: 0, AF: 0]
02/03/17 08:14:02.959489 zms[7078].DB1-zm_monitor.cpp/459 [Monitor Terrasse-MTN has function 3]
02/03/17 08:14:02.959538 zms[7078].DB1-zm_monitor.cpp/460 [Monitor Terrasse-MTN LBF = '%N - %Y-%m-%d %H:%M:%S %z', LBX = 0, LBY = 0, LBS = 1]
02/03/17 08:14:02.959569 zms[7078].DB1-zm_monitor.cpp/461 [Monitor Terrasse-MTN IBC = 50, WUC = 25, pEC = 25, PEC = 25, EAF = 1, FRI = 1000, RBP = 6, ARBP = 6, FM = 0]
02/03/17 08:14:02.959601 zms[7078].DB1-zm_monitor.cpp/3004 [Loaded monitor 7(Terrasse-MTN), 0 zones]
02/03/17 08:14:02.959686 zms[7078].DB1-zm_monitor.cpp/543 [Unable to map file /dev/shm/zm.mmap.7 (46081760 bytes) to locked memory, trying unlocked]
02/03/17 08:14:02.959742 zms[7078].DB1-zm_monitor.cpp/546 [Mapped file /dev/shm/zm.mmap.7 (46081760 bytes) to locked memory, unlocked]
02/03/17 08:14:02.959989 zms[7078].DB1-zm_stream.cpp/306 [We have obtained a lock on /var/run/zm/zms-613998.lock fd: 10]
02/03/17 08:14:03.244942 zms[7078].DB1-zm_event.cpp/1088 [Got QUERY command, sending STATUS]
02/03/17 08:14:04.498736 zms[7078].DB1-zm_event.cpp/1088 [Got QUERY command, sending STATUS]
02/03/17 08:14:05.762125 zms[7078].DB1-zm_event.cpp/1088 [Got QUERY command, sending STATUS]
02/03/17 08:14:07.015967 zms[7078].DB1-zm_event.cpp/1088 [Got QUERY command, sending STATUS]
02/03/17 08:14:08.270199 zms[7078].DB1-zm_event.cpp/1088 [Got QUERY command, sending STATUS]
02/03/17 08:14:09.874229 zms[7078].DB1-zm_event.cpp/1088 [Got QUERY command, sending STATUS]
02/03/17 08:14:11.157090 zms[7078].DB1-zm_event.cpp/1088 [Got QUERY command, sending STATUS]
02/03/17 08:14:12.400063 zms[7078].DB1-zm_event.cpp/1088 [Got QUERY command, sending STATUS]
02/03/17 08:14:20.230533 zms[7078].ERR-zm_event.cpp/1329 [Unable to send raw frame 67: Broken pipe]
02/03/17 08:14:20.803266 zms[7078].DB1-zm_logger.cpp/239 [Terminating Logger]
```

```
02/03/17 08:15:59.734158 undef[7202].DB1-zm_logger.cpp/234 [LogOpts: level=DB1/DB1, screen=OFF, database=INF, logfile=DB1->/var/log/zm/zm_debug.log.07202, syslog=INF]
02/03/17 08:15:59.734301 undef[7202].INF-zm_config.cpp/139 [No Server ID or Name specified in config.  Not using Multi-Server Mode.]
02/03/17 08:15:59.842261 zms[7202].DB1-zm_logger.cpp/234 [LogOpts: level=DB1/DB1, screen=OFF, database=INF, logfile=DB1->/var/log/zm/zm_debug.log.07202, syslog=INF]
02/03/17 08:15:59.842370 zms[7202].DB1-zm_utils.cpp/262 [Detected a x86\x86-64 processor with SSSE3]
02/03/17 08:15:59.842467 zms[7202].DB1-zms.cpp/102 [Query: source=event&mode=jpeg&event=172&frame=1&scale=100&rate=100&maxfps=30&replay=single&auth=XXX&connkey=194345&rand=1486106159]
02/03/17 08:15:59.843017 zms[7202].DB1-zm_user.cpp/172 [Attempting to authenticate user from auth string 'XXX']
02/03/17 08:15:59.843600 zms[7202].DB1-zm_user.cpp/235 [Checking auth_key 'XXX' -> auth_md5 'XXX']
02/03/17 08:15:59.843696 zms[7202].DB1-zm_user.cpp/241 [Authenticated user 'admin']
02/03/17 08:15:59.850733 zms[7202].DB1-zm_monitor.cpp/2738 [Got 1 for v4l_captures_per_frame]
02/03/17 08:15:59.851135 zms[7202].DB1-zm_monitor.cpp/390 [monitor purpose=0]
[... cut ...]
02/03/17 08:16:05.150378 zms[7202].DB1-zm_event.cpp/1088 [Got QUERY command, sending STATUS]
02/03/17 08:16:06.945949 zms[7202].DB1-zm_event.cpp/1088 [Got QUERY command, sending STATUS]
02/03/17 08:16:08.191957 zms[7202].DB1-zm_event.cpp/1088 [Got QUERY command, sending STATUS]
02/03/17 08:16:09.446273 zms[7202].DB1-zm_event.cpp/1088 [Got QUERY command, sending STATUS]
02/03/17 08:16:09.677925 zms[7202].ERR-zm_event.cpp/1329 [Unable to send raw frame 39: Broken pipe]
02/03/17 08:16:10.026667 zms[7202].DB1-zm_logger.cpp/239 [Terminating Logger]
```

abishai, 2017-02-03:
Is fcgiwrap has enough childs to process requests? Default value is 1.

mohahn, 2017-02-03:
Yes it has, exactly 10 at the moment. But even it's only 1, then it does not explain why it works for some videos (i.e. events) and for some not. And it does not explain why LOG_DEBUG fixes this issue.

knnniggett, 2017-02-03:
Zms is crashing, which (only) handles the streaming, so it makes perfect sense that everything else is working. The backtrace hints that it is failing when it tries to load monitor 4 so find out which of your two monitors has an id of 4.

We need to see:
- How you populated every field for this monitor. Show everything, but in particular any field that might be particularly long, have non-standard ascii characters (such as a password embedded in a url), or might be unusual in any other way
- With only the broken monitor enabled & debug set to level 3, show us a complete set of log entries from the moment zoneminder was started up until just *after* the crash occurs. Don't cherry pick log entries. Context is very important.
- Use vlc to verify you have the resolution correct, down to the exact pixel. Many cameras stream at a slightly different resolution than what the marketing materials says. I don't see how this could cause a backtrace, but verify it anyway just in case.

Lastly, you are running an old version of ZoneMinder. If it isn't obvious what the issue is and we cannot duplicate it, then the next step will be for you to build and install a development version from the master branch. I have previously fixed an issue very similar to this (debug even made it go away just like your case).

mohahn, 2017-02-03:
Hi. The _funny_ thing, there is no monitor with ID 4. Sorry, I don't want to annoy you, but have you really read, what I've written in first post?

- There are two monitors (ID 6+7), one configured as RTSP stream the other streams MJPEG.
- The crash occurs on *both* monitors.
- I've said: I _cannot_ log it, because with LOG_DEBUG enabled there is no crash!?
- And I've written that the crash occurs only for some recorded events, not for all (approximately 90% of all events are fine)

I'm sorry but I don't know how to provide you more needful informations. All available logs are posted above and here are the settings for one monitor that differ from standard 1.29 settings:

Name: Terrasse-MTN
Source Type: Remote
Function: Modect
Remote Protocol: HTTP
Remote Hostname: admin:ABCDEF@192.168.179.101
Remote Host Path: /Streaming/Channels/102/httppreview
Target colorspace: 32bit
Capture Width: 640
Capture Height: 360

I tried to prevent self-compiling and use distributions package as ZM suggests to do so. So it seems this is unresolvable for me and is resolved itself as enabled debugging is a workaround.

knnniggett, 2017-02-03:
@mohahn I read it multiple times, and yes you are right I told you turn on debug. That was knee jerk since 99.99999% of the time that is exactly what is needed. ...so there. I fixed it for you, andI added another item as well.  We need more data from you and the way to do that is for you provide what is being asked in each bullet point. Getting to the bottom of this will likely take a lot of back and forth discussion.

Enabling debug is just a work around. This could be masking a bigger problem, which is why I am asking for information. I can explain to you why enabling debug makes things work, but I doubt you'd care.

Please post screenshots of your monitor config. Do not re-type by them by hand in this case. 
Is that password the actual password you are using or did you change it before posting?

In any case, if you are saying you are satisfied with the workaround and don't wish to help further then no harm no foul. I will close this issue out. If there really is an underlying issue then someone else will experience it and then we will know (and I will reopen this issue).

mohahn, 2017-02-03:
As compiling of ZM 1.30 based on the [documentation](https://zoneminder.readthedocs.io/en/stable/installationguide/ubuntu.html#harder-way-build-package-from-source) fails with

```
#./do_debian_package.sh `lsb_release -a 2>/dev/null | grep Codename | awk '{print $2}'`  `date +%Y%m%d`01 local stable
[...]
This package has a Debian revision number but there does not seem to be
an appropriate original tar file or .orig directory in the parent directory;
(expected one of zoneminder_1.30.2-2017020301.orig.tar.gz, zoneminder_1.30.2-2017020301.orig.tar.bz2,
zoneminder_1.30.2-2017020301.orig.tar.lzma,  zoneminder_1.30.2-2017020301.orig.tar.xz or zoneminder_1.30.2-2017020301-local.orig.orig)
continue anyway? (y/n) y
 dpkg-buildpackage -rfakeroot -d -us -uc -S -sa
dpkg-buildpackage: Warnung: Verwendung eines root-werde-Befehls, obwohl bereits root
dpkg-buildpackage: Quellpaket zoneminder
dpkg-buildpackage: Quellversion 1.30.2-2017020301-local
dpkg-buildpackage: Quelldistribution 2017020301
dpkg-buildpackage: Quellen geändert durch root <root@zeus>
 dpkg-source --before-build zoneminder_1.30.2-2017020301-local.orig
 fakeroot debian/rules clean
dh clean --parallel --buildsystem=cmake --builddirectory=dbuild \
            --with systemd,sphinxdoc,apache2,linktree
   dh_testdir -O--parallel -O--buildsystem=cmake -O--builddirectory=dbuild
   dh_auto_clean -O--parallel -O--buildsystem=cmake -O--builddirectory=dbuild
   debian/rules override_dh_clean
make[1]: Entering directory '/install/zoneminder_1.30.2-2017020301-local.orig'
dh_clean dbuild/scripts/zmupdate.pl.1
rm -f -r docs/_build docs/installationguide
make[1]: Leaving directory '/install/zoneminder_1.30.2-2017020301-local.orig'
 dpkg-source -b zoneminder_1.30.2-2017020301-local.orig
dpkg-source: Fehler: unmöglich, mit Quellformat »3.0 (native)« zu bauen: native Paketversion darf keine Revision haben
dpkg-buildpackage: Fehler: Fehler-Exitstatus von dpkg-source -b zoneminder_1.30.2-2017020301-local.orig war 255
debuild: fatal error at line 1376:
dpkg-buildpackage -rfakeroot -d -us -uc -S -sa failed
```

and I have no time to dig into it, I have to come back to this ticket. So I'll give you screenshot and logs on upcoming monday. Password will be defaced (they consists only of letters and numbers a-zA-Z0-9).

PS: And "but I doubt you'd care" is pure speculation. I'm a developer also ;-)

kylejohnson, 2017-02-03:
Let us try and keep this civil.

I'm guessing that the syslog tail is from when you try to view the event?  Could you also provide the relevant nginx and fcgi log lines at this time?

Can you also try unlinking the monitors and then performing the test (with new events) again?  The truth is that some of this code is very old, and maybe there are gremlins which only manifest when LOG_DEBUG is enabled while using linked monitors.  This can help us narrow down the issue.

knnniggett, 2017-02-03:
@mohahn In case you noticed the emojis appear in this thread I want to point out something that we've had to deal with recently. We had a user show up from out of nowhere last December who, rather than try to work with the project team, started trashing zoneminder in his blogs, spammed our forums with his product, and even thought it was okay to use our private messaging system to send unsolicited messages to our users. He has since been banned from our user forum, but as you can see he has resulted in childish tactics to seek his revenge. He has even quoted this very issue in his blog, citing that I am telling you that bugs are normal and I am trying to argue with you. I can assure you that this is a complete fabrication on his part. The questions I am asking to try to get a better understanding of the problem so we can fix it.  Maybe our immature friend is super smart and can fix these sorts of problems with the mere stroke of a magic wand, while I am just a mere human and need a lot of conversation to figure out how to solve the problem. I get the sense that you probably realize this, and I am writing this more for the sake of our friend since he will likely log in with an alt account to troll us further since his github account has now been banned from this forum as well. Again, sorry for the distraction.

abishai, 2017-02-03:
https://www.reddit.com/r/ShinobiCCTV/comments/5rv9oh/zoneminder_issues_are_like_manga/
Hahahaha. He is so funny

knnniggett, 2017-02-03:
@abishai Yeah, I read that, but I don't want the real issue in this thread to get side tracked. Feel free to discuss this in our irc channel. ;-)

mohahn, 2017-02-03:
OMG I'm famous ;-)
But really frustrating is, that you suggest to compile it from source but there is no clear documentation. I found the linked doc for Ubuntu only and this fails as written above. My intend is: if it would be fixed in 1.30 I can stop spending my time from debugging. Unfortunately I don't get it installed.
So if you have no hint how to install it, so that I can test it first, I promise to give you more informations. As I'll try some of @kylejohnson suggests and am extremly busy for the next 2 weeks, it'll take a bit longer than monday..

knnniggett, 2017-02-03:
@mohahn The last time I followed the documentation to build a deb on Debian Jessie, it worked with no problems at all, but I am far more experienced with rpm rather than Debian. Consequently, I've got a question in our slack channel for another to take a look so please be patient.  

Since you are using a spin of Debian, it would be interesting to see what the value of `lsb_release -a 2>/dev/null | grep Codename | awk '{print $2}''` is on your system. 

Also, please build the latest master, not the latest stable release.

UPDATE: Actually, that does seem to be the problem. 
Here is where I last built zoneminder for Jessie: http://zoneminder.blogspot.com/p/zoneminder-on-odroid-xu4.html

If you look at the resulting package filename in my example, the distro name "jessie" is part of the filename. 

However, in your example the distro name is missing from your attempt... so try replacing `lsb_release -a 2>/dev/null | grep Codename | awk '{print $2}''` with "jessie" just like the documentation says.

mohahn, 2017-02-03:
`lsb_release -a 2>/dev/null | grep Codename | awk '{print $2}'` *does* return "jessie". That's all I know so far, but testing (with latest master) has to wait until monday.

SteveGilvarry, 2017-02-03:
@mohahn I will check a Jessie master build for you over the weekend, I normally use the build script on Ubuntu without issue. Will probably just be a fresh install so might not pickup exact config. Sorry for your troubles, zms is my least favourite component and I am working to replace it.

connortechnology, 2017-02-04:
I use that Ubuntu
1604 build for Jessie. Or debian distro is really for wheezy

mohahn, 2017-02-04:
I want to mention that my base system is a relative fresh install without any modifications. I've setup OpenMediaVault 3 (based on Debian 8), installed MySQL server and Zoneminder from Debian repo. That's all what I've done no other funky features. But keep cool and wait before investing time; I'll try build with master on monday again.

connortechnology, 2017-02-04:
In my branch, I have removed the content from /distros/ubuntu1604/source/format .  The file has to still be there, but having it empty allows the build to continue.

mohahn, 2017-02-06:
@connortechnology when truncated file the build does exit immediately without failure (build-package is not executed) and there is no .deb file. When the file contains only "3.0" (instead of "3.0 (native)") the build also fails with

```
zoneminder-build-deps (1.29.0+dfsg-1) wird eingerichtet ...
This package has a Debian revision number but there does not seem to be
an appropriate original tar file or .orig directory in the parent directory;
(expected one of zoneminder_1.30.2-2017020601.orig.tar.gz, zoneminder_1.30.2-2017020601.orig.tar.bz2,
zoneminder_1.30.2-2017020601.orig.tar.lzma,  zoneminder_1.30.2-2017020601.orig.tar.xz or zoneminder_1.30.2-2017020601-local.orig.orig)
continue anyway? (y/n) y
 dpkg-buildpackage -rfakeroot -d -us -uc -S -sa
dpkg-buildpackage: Warnung: Verwendung eines root-werde-Befehls, obwohl bereits root
dpkg-buildpackage: Quellpaket zoneminder
dpkg-buildpackage: Quellversion 1.30.2-2017020601-local
dpkg-buildpackage: Quelldistribution 2017020601
dpkg-buildpackage: Quellen geändert durch root <root@zeus>
 dpkg-source --before-build zoneminder_1.30.2-2017020601-local.orig
dpkg-source: Fehler: Quellpaketformat »3.0« wird nicht unterstützt: Can't locate Dpkg/Source/Package/V3.pm in @INC (you may need to install the Dpkg::Source::Package::V3 module) (@INC contains: /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.20.2 /usr/local/share/perl/5.20.2 /usr/lib/x86_64-linux-gnu/perl5/5.20 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.20 /usr/share/perl/5.20 /usr/local/lib/site_perl .) at (eval 9) line 1.

dpkg-buildpackage: Fehler: Fehler-Exitstatus von dpkg-source --before-build zoneminder_1.30.2-2017020601-local.orig war 2
debuild: fatal error at line 1376:
dpkg-buildpackage -rfakeroot -d -us -uc -S -sa failed
Do you want to keep the checked out version of Zoneminder (incase you want to modify it later) [y/N]y
The checked out copy is preserved in zoneminder_release
Done!
```

mohahn, 2017-02-07:
Apart from compiling problems for 1.30, I think I can delimit the crash a bit. As exporting web log from ZM produces only empty files without any error (option "all" is checked), it's getting annoying a bit :-(
I've remove one monitor (RTSP), so only the MJPEG stream exists. I've attached screens and a log. What I've noticed is (see screen zm-events.jpg), that all events longer than 25sec are broken when I want to view them (except with LOG_DEBUG enabled as I mentioned earlier).

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[zm-syslog.txt](https://github.com/ZoneMinder/ZoneMinder/files/756736/zm-syslog.txt)

SteveGilvarry, 2017-02-07:
The last two recognisable ZM functions in the backtrace.
https://github.com/ZoneMinder/ZoneMinder/blob/254fcbcef790c0db11cd845548cd0383b7660e90/src/zm_stream.cpp#L46
https://github.com/ZoneMinder/ZoneMinder/blob/254fcbcef790c0db11cd845548cd0383b7660e90/src/zm_monitor.cpp#L2695
The relation to 25 second events is at least some clue. Makes me think some resource limitation.

connortechnology, 2017-02-07:
Crashing when loading a monitor can be caused by doing atoi on a null column... can be annoying to track down which one.  
Can also be caused by not being able to allocate memory.  Doesn't explain why turning on debug log fixes it though.

knnniggett, 2017-02-07:
I tend to agree this is memory related.

When debug is turned on, the memory allocation order changes. So turning on Debug doesn't fix the problem, it just masks the problem. The difference is the area of memory being overwritten/oveflowed just happens to no longer be used by the zms. 

I ran into the same kind of problem with #1153 .
It took 2 months to identify exactly how the problem was occurring and then solve it.

closed 2018-06-22 by stale[bot]