ZoneMinder/zoneminder #1768: %EI1% and %EIM% email tokens don't work in H264-videostorage branch

vpoluyaktov, 2017-02-11:
If you are using **H264-videostorage** branch (direct recording of h264 cameras into MP4 format) and in a monitor **Storage** settings set **Save JPEGs** in **Disabled** value, **%EI1%** and **%EIM%** email tokens won't work because a jpeg that needs be attached to an email (or a message) just doesn't exist. 
To fix the issue a frame can be extracted directly from a .mp4 file on the fly using ffmpeg utility (in zmfilter.pl)
 
**Relevant log lines:**
```
2017-02-10 16:15:37.693190	zmfilter		19334	ERR	Can't send email: /usr/share/zoneminder/www/events/5/17/02/10/16/14/00/00689-capture.jpg: not readable	zmfilter.pl
```

vpoluyaktov, 2017-02-11:
Fix branch created https://github.com/vpoluyaktov/ZoneMinder/tree/1768-Fix-email-tokens-for-h264-storage

knight-of-ni, 2018-09-23:
Marking this as solved since #1814 was merged.

closed 2018-09-23 by knight-of-ni