ZoneMinder/zoneminder #1781: Adjust score calculation algorithm

vajonam, 2017-02-15:
I am not convinced that the score is calculated correctly. When I specify the MAX_ALARMED, MAX_FILTER or MAX_BLOB size, those should be used to calculate the score rather the area of the whole polygon.

I might have a very large zones (less zones for faster), but smaller max (30-40% of the zone.area() )values which means the scoring should be higher, this is important as more higher scores are less likely to be false positives. 

https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_zone.cpp#L273

https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_zone.cpp#L368

https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_zone.cpp#L667

johncolby, 2018-04-20:
@connortechnology I agree that the score algorithm could be improved.

A more basic change would be letting scores have more precision one way or another (e.g. 1000* instead of 100*), so as to avoid the problem of many events being generically squashed to a score of 1 (see below, as also mentioned by @mastertheknife in #1782 ).

[image]

Then, @vajonam's suggestion could be accommodated by letting user have a display option for the Y-axis like "Display as..." "% of zone pixels" or "% of max threshold pixels" or "log scale" or whatever.

closed 2018-06-26 by stale[bot]

vajonam, 2018-08-14:
so what do I need to do to get this in? you new a new PR against current master/HEAD?

connortechnology, 2018-08-22:
Yes update it.  In the meantime I will make another attempt at understanding it.

vajonam, 2018-08-22:
So what I am proposing is that when we have any alarm and the max_alarm_pixes is not 0. i.e the user has specified that the max alarm aera in percent or pixels (gets converted to pixels anyway), then use that to calculate the score rather than to total size of the zone. 

```
if (max_alarm_pixels != 0)  // max size is specified use that to calculate the score
    score = (100*alarm_pixels)/max_alarm_pixels;
  else // max is not specified use total size of the zone to calculate the score
    score = (100*alarm_pixels)/polygon.Area();

```
So if we have a zone that 100% of the screen, but we say the max_alarm_pixels is 20%, then the score will be calculated on the 20% rather than the size of the screen. this means that we have an object that is 10% of the screen size that triggers an alarm, it will have a score of 50, rather than than 10, which means its important event to the user.

connortechnology, 2018-08-22:
Yeah I spent some time thinking about it and I think I agree.  Update the PR and I'll merge it.

vajonam, 2018-08-22:
Updated with PR #2186