ZoneMinder/zoneminder #1786: Dockerfile Ubuntu 16.04 Xenial support

MarkosVakondios, 2017-02-17:
As requested by SteveGilvarry in #1774

mharrend, 2017-02-26:
Hi, I have a Zoneminder Docker setup using Xenial which can be found here:
https://github.com/mharrend/docker-zoneminder

Maybe you can check it out.

knnniggett, 2018-01-14:
We've moved our dockerfiles here:
https://github.com/ZoneMinder/zmdockerfiles

We will continue to troubleshoot docker related issue there.

@MarkosVakondios We would be willing to give you direct commit access to the new zmdockerfiles repo if you are interested in maintaining it. Let me know.

closed 2018-01-14 by knnniggett

MarkosVakondios, 2018-01-15:
@knnniggett: Sure, I will be glad to. Please be aware that it's the first time I will be maintaining a github repo. Any pointers apart from general Googling would be appreciated to avoid elementary mistakes.

viktorsmari, 2018-10-03:
Why don't we have a single Dockerfile and a `docker-compose.yml` in the main repo, like most open source projects do?
Even though it is flaky or a bit broken, we can note in the readme that things are 'experimental'.