ZoneMinder/zoneminder #1796: ZoneMinder 1.30.2 and problems with PATH_ZMS

tecufanujacu, 2017-02-25:
I'm using ZoneMinder 1.30.2 with OpenMediaVault 2.2.13 and I have a problem, I can't open a windows with a livestrem of the camera and see a livestream of an event in the same time, in the window of the one opened for second I can't see nothing.

In the log i have these two warnings:

> getCmdResponse stream error: Socket does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors()

> Socket does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.

I have already changed the PATH_ZMS from /cgi-bin/nph-zms to /zm/cgi-bin/nph-zms otherwise I wasn't be able to see nothing.

What can I do?

knnniggett, 2017-02-25:
This is not a bug. Please take this to the user forum.

closed 2017-02-25 by knnniggett

tecufanujacu, 2017-02-25:
I already tried everything is proposed in the user forum but without success, I only have this problem with ZoneMinder 1.30.2, where should I find in the user forum? Could you point to me to the right thread?

knnniggett, 2017-02-25:
PATH_ZMS issues are not bugs in the zoneminder software. We agree it can be difficult to understand what the problem is, and someday we would love to zms just go away for something easier.

However, as we've stated in our posting guidelines, our github forum is not the place to troubleshoot one's system. We have a large user forum just for that purpose. Create a new thread, and we will be glad to help you there.