ZoneMinder/zoneminder #1797: ZoneMinder - Cross-Site Scripting (XSS)

bestshow, 2017-02-28:
Product：ZoneMinder
Download: https://github.com/ZoneMinder/ZoneMinder
Vunlerable Version: 1.30.2 and probably prior
Tested Version: 1.30.2
Author: Haojun Hou in ADLab of Venustech

Advisory Details:
Haojun Hou in ADLab of Venustech discovered a Cross-Site Scripting (XSS) in “ZoneMinder”, which can be exploited to execute arbitrary code.
The vulnerability exists due to insufficient filtration of user-supplied data in “postLoginQuery” HTTP POST parameter passed to “ZoneMinder-master/web/skins/classic/views/js/postlogin.js.php” url. An attacker could execute arbitrary HTML and script code in browser in context of the vulnerable website.
The exploitation example below uses the "alert()" JavaScript function to see a  pop-up messagebox:
Poc:
Post: postLoginQuery=><script>alert(1);</script><
To
http://localhost/.../ZoneMinder-master/web/skins/classic/views/js/postlogin.js.php

connortechnology, 2017-03-01:
It is not exactly clear how to reproduce this.  A better example would help.  However, I propose that we use urlencode on the passed POST data for now...

For future releases, I propose that we move all skin content out of the direct doc root so that we can't post to it directly.

bestshow, 2017-03-02:
Using "POST" method pass the parameter "postLoginQuery" which value is "><script>alert(1);</script><" to the affected the URL  "ZoneMinder-master/web/skins/classic/views/js/postlogin.js.php",and  you will see the pop-up messagebox.

connortechnology, 2017-03-13:
Please note that this requires a misconfiguration of apache.  Only index.php should be executable as a php script.

mnoorenberghe, 2017-03-18:
> Please note that this requires a misconfiguration of apache. Only index.php should be executable as a php script.

Why do you think that? Every setup I've used enables PHP for all files with the .php extension. See line 1 of Ubuntu's configuration: https://www.apt-browse.org/browse/ubuntu/xenial/main/i386/libapache2-mod-php7.0/7.0.4-7ubuntu2/file/etc/apache2/mods-available/php7.0.conf

Changing the suffix is a better way to stop it from being executed e.g. .php.inc but I'm not even sure that will fix the core problem here as it may still be exploited from the regular login form (haven't tested yet).

mnoorenberghe, 2017-03-18:
I'm making a patch for this btw.

carnil, 2017-03-21:
This has been assigned CVE-2017-7203

SteveGilvarry, 2017-03-21:
Slight issue being 1.30.2 is not released, it should be 1.30.2-rc.1, simply because we are reviewing the patch for this now and 1.30.2 won't have issue.

commit ea5342ab (2017-03-21): Properly escape postLoginQuery. Fixes #1797
commit eaca58bb (2017-03-21): Merge pull request #1815 from mnoorenberghe/postlogin

Properly escape postLoginQuery. Fixes #1797
closed 2017-03-21 by connortechnology