ZoneMinder/zoneminder #1798: API: Long-standing connection for alarm updates

mickeyl, 2017-03-04:
Is there any interest in adding an API for a long-standing connection that streams a notification line whenever any of the monitors emit an alarm?

(Background: I'm working on an app that offers remote iOS/Android PUSH notification alarms for some models (e.g. MOBOTIX), and I think it would be interesting to support Zoneminder installations as well).

connortechnology, 2017-03-04:
That would be neat.  Might be a little tricky because zm is multiple 
processes... I would be tempted to use a pipe...

pliablepixels, 2017-03-10:
The problem with a long-standing API that does not terminate the connection is that it will take up 1 slot of the max 6 per domain problem Chrome has.

connortechnology, 2017-03-11:
I think the idea is to use this in other applications rather than the browser.  Plus... we need to solve that connection limit problem anyways.

mickeyl, 2017-03-11:
Exactly. I'm working on an iOS app that talks to Zoneminder installations in order to allow polling-free-alarm-notifications that are bridged to PUSH notification services, such as Google Cloud Messaging or Apple Push Notification Service. I'm successfully using asker's [zmeventnotification.pl](https://github.com/pliablepixels/zmeventserver), but I think it would be very good if something like that would be available in the main project.

pliablepixels, 2017-03-11:
I think there are going to practical challenges:

1. A browser is not always just a browser - it can also be an embedded browser (like in zmN). Any client that is also displaying streams and has an embedded web view may land up in the connection issue (based on [which browser instance](http://blog.olamisan.com/max-parallel-http-connections-in-a-browser) is used) which the long polling API will contribute to.

2. CakePHP and PHP are stateless, so making CakePHP do a long standing connection, I suppose, will anyway require jugglery (There are examples on the net that mess with jquery or node to achieve this)

3. Websockets are bi-directional while long polling is uni-directional. If we are putting in a service, why not architect it as bi-directional to future proof it?

I

connortechnology, 2017-03-11:
I agree it should be bi-directional.  I'm just thinking that locally we implement it as a pipe, and then make a quick perl script that just exposes the pipe as a network socket. Or something like that.

pliablepixels, 2017-03-11:
yup that would work, but then I suppose we have the [same issues of a raw tcp connection](http://stackoverflow.com/a/16947118/1361529) - that it's not based on message boundaries making the client more complex. I mean this was the way its been done for a long time till web sockets came along, so it will work, just that I think we have better solutions now.

And I suppose doing it the pipe/socket way will be faster. Not sure if its the best way though, but that's what zmtrigger does too, so....

closed 2018-06-22 by stale[bot]