ZoneMinder/zoneminder #1801: Multiple SQLis

gioulisapo, 2017-03-06:
Hello, 
My colleauges at Cisco inc. have discovered multiple SQL injections on the software. How do wish to proceed? Is it prefereable that I simply post the details of the advisory here or should we establish an alernate communication route. 

Thank you,
(Credit For Discovery:

Jerzy Kramarz 
Michail Sarantidis
Rafael Gil Larios
Giovani Cattani
Anton Garcia)

SteveGilvarry, 2017-03-06:
Were they found against current master? There are some recent patches to prevent emulated prepared statements so that may have fixed them. These are part of 1.30.2 version that is currently in rc1 status. If you believe they are still evident one of the team will provide you with a secure communication channel.

gioulisapo, 2017-03-06:
Hello, they were found againsta the master branch. We will test them again against 1.30.2 and get back to you.

connortechnology, 2017-03-06:
You can email me at isaac@zoneminder.com.

gioulisapo, 2017-03-14:
The vulnerable code is unchanged, so the advisory should work on the new branch as well. How should we proceed with the advisory details.

SteveGilvarry, 2017-03-14:
@gioulisapo Email address above from @connortechnology

SteveGilvarry, 2017-03-22:
@connortechnology confirm these have been received as I assume we are on the clock for them being released.

connortechnology, 2017-03-22:
I have received them.  At first glance they appear to be repeats of the existing CVE's, but the content is a little hard to grok.  I will be sharing them with you today.

gioulisapo, 2017-03-27:
Hello, 
Do you have any news on the final verdict on the advisory? Is it a duplicate. Thank you for your cooperation.

connortechnology, 2017-03-27:
I believe the issues are duplicates.  I also believe that they do not affect current master or 1.30.2-rc1.

I would like someone else to confirm.

gioulisapo, 2017-04-05:
Hello, just wanted to make sure that this issue is indeed a duplicate so that it can be marked as Closed. Can someone else confirm.

closed 2018-06-22 by stale[bot]