ZoneMinder/zoneminder #1802: Main page event counter mistakes

flexiti, 2017-03-08:
Event counters live their own lives.
Should something reset?

[image]

connortechnology, 2017-03-08:
I'm not sure what you are trying to show us.  You are pointing out an event that happened at 10:29am ... and the current time is 1:31pm.

So?

flexiti, 2017-03-08:
if the last 24 hours is ok, this explains 11 events , thanks  
btw. which in this case means month and how much is it?  30 days? 31 days? :)

however,   

I think (my opinion) **day** is more associated with the current day than the last 24 hours.
I'm thinking that it would be more convenient: 
- just how much of this hour 
- how much on this day, 
- how much this month etc.

closed 2018-06-22 by stale[bot]