ZoneMinder/zoneminder #1804: Unable to add PPA Iconnor Ubuntu 16.04

tridiumcontrols, 2017-03-12:
Running into an issue with adding iconnors PPA in ubuntu 16.04. 
```
max@UbuntuNVR:~$ sudo add-apt-repository ppa:iconnor/zoneminder
[sudo] password for max: 
Cannot add PPA: 'ppa:~iconnor/ubuntu/zoneminder'.
ERROR: '~iconnor' user or team does not exist.
```

I tried adding the manual way to the sources list. 
```
deb http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial main 
deb-src http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial main 

```
and I Get this. 

```
max@UbuntuNVR:~$ sudo nano /etc/apt/sources.list
max@UbuntuNVR:~$ sudo apt-get update
Hit:1 http://archive.ubuntu.com/ubuntu xenial InRelease
Get:2 http://archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB]                           
Get:3 http://archive.ubuntu.com/ubuntu xenial-security InRelease [102 kB]                          
Ign:4 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial InRelease           
Ign:5 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial Release
Ign:6 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main Sources
Ign:7 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main amd64 Packages
Ign:8 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main all Packages
Ign:9 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main Translation-en
Ign:6 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main Sources
Ign:7 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main amd64 Packages
Ign:8 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main all Packages
Ign:9 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main Translation-en
Ign:6 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main Sources
Ign:7 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main amd64 Packages
Ign:8 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main all Packages
Ign:9 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main Translation-en
Ign:6 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main Sources
Ign:7 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main amd64 Packages
Ign:8 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main all Packages
Ign:9 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main Translation-en
Ign:6 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main Sources
Ign:7 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main amd64 Packages
Ign:8 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main all Packages
Ign:9 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main Translation-en
Err:6 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main Sources
  Unable to connect to ppa.launchpad.net:http:
Ign:7 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main amd64 Packages
Ign:8 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main all Packages
Ign:9 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial/main Translation-en
Fetched 204 kB in 3s (65.5 kB/s)
Reading package lists... Done
W: The repository 'http://ppa.launchpad.net/iconnor/zoneminder/ubuntu xenial Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://ppa.launchpad.net/iconnor/zoneminder/ubuntu/dists/xenial/main/source/Sources  Unable to connect to ppa.launchpad.net:http:
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

talkinggoat, 2017-03-12:
sudo add-apt-repository ppa:iconnor/zoneminder Stable releases of ZoneMinder.

This is to provide newer official packages for older ubuntu dists.
 More info: https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder
Press [ENTER] to continue or ctrl-c to cancel adding it

gpg: keyring `/tmp/tmp6z93tt8h/secring.gpg' created
gpg: keyring `/tmp/tmp6z93tt8h/pubring.gpg' created
gpg: requesting key 776FFB04 from hkp server keyserver.ubuntu.com
gpg: /tmp/tmp6z93tt8h/trustdb.gpg: trustdb created
gpg: key 776FFB04: public key "Launchpad PPA for Isaac Connor" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
OK

Try it again... it works here.

tridiumcontrols, 2017-03-12:
Just tried it again, same result. Tried on a fresh install of Ubuntu 16.04, Weird. what am I missing? 

`Audis-MacBook-Pro-2:~ audi$ ssh max@192.168.1.4
max@192.168.1.4's password: 
Welcome to Ubuntu 16.04.2 LTS (GNU/Linux 4.4.16-1-pve x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
Last login: Sun Mar 12 05:29:40 2017 from 192.168.1.243
max@UbuntuNVR:~$ sudo add-apt-repository ppa:iconnor/zoneminder
[sudo] password for max: 
Cannot add PPA: 'ppa:~iconnor/ubuntu/zoneminder'.
ERROR: '~iconnor' user or team does not exist.
max@UbuntuNVR:~$ 
`

closed 2017-03-13 by tridiumcontrols

pxlong, 2017-10-19:
@tridiumcontrols  Hi, I also have this error, how do you fix it? Thanks

connortechnology, 2017-10-19:
get rid of the ~

Berg000, 2018-12-13:
how can you solve that ,can you give me a guide?