ZoneMinder/zoneminder #1805: Build failure at zm_image.h

adam900710, 2017-03-12:
When building (either using AUR from archlinux or build manually), v1.30 always fails at zm_image.h:
```
[  5%] Building CXX object src/CMakeFiles/zm.dir/zm_camera.cpp.o
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/6.3.1/include-fixed/limits.h:168:0,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/6.3.1/include-fixed/syslimits.h:7,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/6.3.1/include-fixed/limits.h:34,
                 from /usr/include/zconf.h:418,
                 from /usr/include/zlib.h:34,
                 from /tmp/yaourt-tmp-adam/aur-zoneminder/src/ZoneMinder-1.30.0/src/zm_image.h:38,
                 from /tmp/yaourt-tmp-adam/aur-zoneminder/src/ZoneMinder-1.30.0/src/zm_camera.h:26,
                 from /tmp/yaourt-tmp-adam/aur-zoneminder/src/ZoneMinder-1.30.0/src/zm_camera.cpp:21:
/tmp/yaourt-tmp-adam/aur-zoneminder/src/ZoneMinder-1.30.0/src/zm_image.h:125:25: error: expected identifier before numeric constant
  enum { CHAR_HEIGHT=11, CHAR_WIDTH=6 };
                         ^
/tmp/yaourt-tmp-adam/aur-zoneminder/src/ZoneMinder-1.30.0/src/zm_image.h:125:25: error: expected '}' before numeric constant
/tmp/yaourt-tmp-adam/aur-zoneminder/src/ZoneMinder-1.30.0/src/zm_image.h:125:25: error: expected unqualified-id before numeric constant
```

Related lines are in definition of  class Image:
```
public:
        enum { CHAR_HEIGHT=11, CHAR_WIDTH=6 };
                     ^
        enum { LINE_HEIGHT=CHAR_HEIGHT+0 };
```

ZoneMinder Version: 1.30.0 stable release

Linux Distribution and Version: Archlinux, so no version at all

Related gcc version: 6.3.1
I tried to build with clang, by exportng `CC=clang` and `CXX=clang++`, then `cmake` and `make`, but still fails with same error message, I assume it just didn't replace the compiler.

adam900710, 2017-03-12:
Clang-3.9.1 fails either, at the same position, similar error message.
```
Scanning dependencies of target zm
[  1%] Building CXX object src/CMakeFiles/zm.dir/zm_box.cpp.o
[  3%] Building CXX object src/CMakeFiles/zm.dir/zm_buffer.cpp.o
[  5%] Building CXX object src/CMakeFiles/zm.dir/zm_camera.cpp.o
In file included from /home/adam/ZoneMinder/src/zm_camera.cpp:21:
In file included from /home/adam/ZoneMinder/src/zm_camera.h:26:
/home/adam/ZoneMinder/src/zm_image.h:125:25: error: expected identifier
        enum { CHAR_HEIGHT=11, CHAR_WIDTH=6 };
                               ^
/usr/include/limits.h:147:22: note: expanded from macro 'CHAR_WIDTH'
#  define CHAR_WIDTH 8
                     ^
1 error generated.
make[2]: *** [src/CMakeFiles/zm.dir/build.make:111: src/CMakeFiles/zm.dir/zm_camera.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:292: src/CMakeFiles/zm.dir/all] Error 2
make: *** [Makefile:128: all] Error 2
```

Also, current head, 2686946ce9c0292979335d35c97ee0e7f13e2ae1 also fails to be built, both gcc6 and clang fails.

So it's not a GCC-6 bug.

adam900710, 2017-03-12:
OK, clang error output is indeed better than gcc6.

Class Image has a public anonymous enum, which defines CHAR_WIDTH=6, while global C header limits.h defines CHAR_WIDTH to 8, then causes this bug.

Not familar with C++, so no idea to fix, but the bug really seems dumb...

adam900710, 2017-03-12:
Renaming CHAR_WIDTH to ZM_CHAR_WIDTH will fix in a dirty C way.
(Sorry, I'm not a C++ fan but a C coder)
Pull request submitted https://github.com/ZoneMinder/ZoneMinder/pull/1806

knnniggett, 2017-03-12:
So the file in question `/usr/include/limits.h` is part of glibc-headers package, which on my system is at version 2.24, and does not define CHAR_WIDTH. Just for the sake of knowing, I'm curious what version you've got on your system.

We are in a release candidate right now, but I think this is important enough to make it into 1.30.2.

I am wondering, however, if it would be more appropriate to do `#ifndef` instead. 
@connortechnology thoughts?

Perhaps its 6 of one, half dozen of another. Getting my C hat on usually takes me a whole day, because well it just isn't like Perl.

knnniggett, 2017-03-12:
To answer my own question, Arch Linux is using glibc 2.25 and sure enough CHAR_WIDTH got added to limits.h e.g.
```
/* The integer width macros are not defined by GCC's <limits.h> before
   GCC 7, or if _GNU_SOURCE rather than
   __STDC_WANT_IEC_60559_BFP_EXT__ is used to enable this feature.  */
#if __GLIBC_USE (IEC_60559_BFP_EXT)
# ifndef CHAR_WIDTH
#  define CHAR_WIDTH 8
# endif
```

connortechnology, 2017-03-12:
I think the zm_ prefix is the right thing to do but I will look closer at the code tonight

connortechnology, 2017-03-12:
zm_image is talking about the # of pixels in a character in the font.  I'm not sure why these are enums instead of just defines.

I would be be tempted to replace them with a #define and rename them to either ZM_CHAR_WIDTH or even ZM_CHARACTER_WIDTH to make it a little more obvious that we aren't talking about the same thing as limits.h

mnoorenberghe, 2017-03-17:
This was fixed by #1806

closed 2017-03-18 by SteveGilvarry