ZoneMinder/zoneminder #1807: Writing frame [av_interleaved_write_frame()] failed: Invalid argument(-22)

shaun-ba, 2017-03-12:
ZoneMinder Version -  1.30.2   (h264 branch from PPA (isaacs))

Linux Distribution and Version - Ubuntu 16.04.2 LTSD

Relevant log lines:
```
2017-03-12 16:14:54.688680	zmc_m11		16661	INF	/build/zoneminder-SBIOmr/zoneminder-1.30.2~20170221163027/src/zm_videostore.cpp:238:DEBUG: pts: -9223372036854775808, dts: -22876189922261, data: 0x7f90a8907010, size: 206962, sindex: 0, dflags: 0000, s-pos: -1, c-duration: 0	zm_videostore.cpp	238
2017-03-12 16:14:54.686147	zmc_m11		16661	WAR	/build/zoneminder-SBIOmr/zoneminder-1.30.2~20170221163027/src/zm_videostore.cpp:312: Writing frame [av_interleaved_write_frame()] failed: Invalid argument(-22)	zm_videostore.cpp	312
```

Just can't get the branch to work,

connortechnology, 2017-03-13:
Honestly I think you have more issues than are apparent here.  Anything happening in zm_videostorage really shouldn't prevent live view.  

So first off, we are going to need debug levels turned up and see the logs. 

 I might encourage you to try out my storageareas ppa because I've done more work there than in h264.

shaun-ba, 2017-03-13:
Happy to try this ppa but why isn't there a branch for it on the main repo?

I did install the video storage branch over a 1.30.0 install as didn't
think that would be an issue. I didn't see any additional steps needed in
the docs to checkout this branch.

I will enable debug logs also and see what is there

shaun-ba, 2017-03-15:
@connortechnology Live stream works on restart, but still getting these errors filling logs up and events are just showing "new event", with 00:00 duration.

Debug logs show no errors or anything of note

shaun-ba, 2017-03-15:
I reinstalled a container from scratch on the latest h264 PPA and events are now recorded however still getting 1000s of the WARNs:

/build/zoneminder-SBIOmr/zoneminder-1.30.2~20170221163027/src/zm_videostore.cpp:312: Writing frame [av_interleaved_write_frame()] failed: Invalid argument(-22)

connortechnology, 2017-03-15:
That generally happens when there is an audio stream that isn't AAC. Turn off storing audio

connortechnology, 2017-03-15:
People don't like me working in the main fork. So I work in my fork. Which is the right thing to do.

SteveGilvarry, 2017-03-15:
Try this and see if it gives an error.
ffmpeg -i rtsp://camera_url -vcodec copy -acodec copy a.mp4

shaun-ba, 2017-03-16:
@connortechnology Cool stuff from master or your own code? everything you work on is with the intention of pulling to the main repo at some point?

thanks!

closed 2018-06-22 by stale[bot]