ZoneMinder/zoneminder #1808: Distorted Time Error 000.0

talkinggoat, 2017-03-13:
I starting having an issue where ZM changes the time format for the cameras. It seems to be happening between 4:30 and 5:00 am. It continued to record, presumably noting, until 5:00am. In the timeline, it says the event was ~1700-1800s, when the events should be ~600s.  

This problem effects all the cameras. 

The network does go down at 3:00 am, to perform backups, but only for about 30 minutes. By 3:23, it is back online and the time is working correctly, on ZM. 

In the logs, the last entry happens at 04:35:53, then picks back up at 04:56:57, saying the time has changed. 

Anyone have any ideas what might be going on, why it looses almost 10 minutes?

**ZoneMinder Version (`zmaudit.pl -v`):**

1.30.2

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**
NAME="Ubuntu"
VERSION="16.04.1 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.1 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
UBUNTU_CODENAME=xenial

**Relevant log lines:**
```
Mar 13 04:30:00 zoneminder zma_m2[2313]: INF [Back_Yard: 671695 - Closing event 24288, section end]
Mar 13 04:30:00 zoneminder zma_m7[2363]: INF [Driveway: 672739 - Closing event 24289, section end]
Mar 13 04:30:00 zoneminder zma_m5[2335]: INF [Kitchen: 673764 - Closing event 24287, section end]
Mar 13 04:30:00 zoneminder zma_m8[2389]: INF [Monitor-8: 656853 - Closing event 24285, section end]
Mar 13 04:30:00 zoneminder zma_m6[2355]: INF [Front_Yard_South: 670957 - Closing event 24286, section end]
Mar 13 04:30:00 zoneminder zma_m1[17082]: INF [Sunroom: 94546 - Closing event 24284, section end]
Mar 13 04:30:00 zoneminder zma_m4[2327]: INF [Garden_and_Shed: 467262 - Closing event 24290, section end]
Mar 13 04:30:00 zoneminder zma_m8[2389]: INF [Monitor-8: 656853 - Opening new event 24291, section start]
Mar 13 04:30:00 zoneminder zma_m2[2313]: INF [Back_Yard: 671695 - Opening new event 24292, section start]
Mar 13 04:30:00 zoneminder zma_m5[2335]: INF [Kitchen: 673764 - Opening new event 24293, section start]
Mar 13 04:30:00 zoneminder zmc_m8[17078]: INF [Deleting videoStore instance]
Mar 13 04:30:00 zoneminder zma_m7[2363]: INF [Driveway: 672739 - Opening new event 24294, section start]
Mar 13 04:30:00 zoneminder zmc_m2[17063]: INF [Deleting videoStore instance]
Mar 13 04:30:00 zoneminder zma_m1[17082]: INF [Sunroom: 94546 - Opening new event 24296, section start]
Mar 13 04:30:00 zoneminder zma_m6[2355]: INF [Front_Yard_South: 670957 - Opening new event 24295, section start]
Mar 13 04:30:00 zoneminder zmc_m1[17062]: INF [Deleting videoStore instance]
Mar 13 04:30:00 zoneminder zma_m4[2327]: INF [Garden_and_Shed: 467262 - Opening new event 24297, section start]
Mar 13 04:30:00 zoneminder zmc_m4[17070]: INF [Re-starting video storage module]
Mar 13 04:30:00 zoneminder zmc_m5[2333]: INF [Re-starting video storage module]
Mar 13 04:30:00 zoneminder zmc_m4[17070]: INF [Opening video storage stream events/4/17/03/13/04/30/00/24297-video.mp4 format: -1959795394#012]
Mar 13 04:30:00 zoneminder zmc_m6[2347]: INF [Re-starting video storage module]
Mar 13 04:30:00 zoneminder zmc_m5[2333]: INF [Opening video storage stream events/5/17/03/13/04/30/00/24293-video.mp4 format: -871549634#012]
Mar 13 04:30:01 zoneminder zmc_m4[17070]: INF [VideoStore startTime=-309766099#012]
Mar 13 04:30:01 zoneminder zmc_m6[2347]: INF [Opening video storage stream events/6/17/03/13/04/30/00/24295-video.mp4 format: 844756286#012]
Mar 13 04:30:01 zoneminder zmc_m5[2333]: INF [VideoStore startTime=949343938#012]
Mar 13 04:30:01 zoneminder zmc_m6[2347]: INF [VideoStore startTime=949416116#012]
Mar 13 04:30:01 zoneminder zmc_m8[17078]: INF [Re-starting video storage module]
Mar 13 04:30:01 zoneminder zmc_m8[17078]: INF [Opening video storage stream events/8/17/03/13/04/30/00/24291-video.mp4 format: 1754355006#012]
Mar 13 04:30:01 zoneminder zmc_m2[17063]: INF [Re-starting video storage module]
Mar 13 04:30:01 zoneminder zmc_m8[17078]: INF [VideoStore startTime=951555028#012]
Mar 13 04:30:01 zoneminder zmc_m7[17074]: INF [Re-starting video storage module]
Mar 13 04:30:01 zoneminder zmc_m2[17063]: INF [Opening video storage stream events/2/17/03/13/04/30/00/24292-video.mp4 format: 778458430#012]
Mar 13 04:30:01 zoneminder zmc_m7[17074]: INF [Opening video storage stream events/7/17/03/13/04/30/00/24294-video.mp4 format: -68541122#012]
Mar 13 04:30:01 zoneminder zmc_m2[17063]: INF [VideoStore startTime=951278969#012]
Mar 13 04:30:01 zoneminder zmc_m7[17074]: INF [VideoStore startTime=947482587#012]
Mar 13 04:30:01 zoneminder zmc_m1[17062]: INF [Re-starting video storage module]
Mar 13 04:30:01 zoneminder zmc_m1[17062]: INF [Opening video storage stream events/1/17/03/13/04/30/00/24296-video.mp4 format: 333215038#012]
Mar 13 04:30:01 zoneminder zmc_m1[17062]: INF [VideoStore startTime=951375257#012]
Mar 13 04:30:05 zoneminder zma_m6[2355]: INF [Front_Yard_South: 671000 - Analysing at 9.90 fps]
Mar 13 04:30:09 zoneminder zmc_m4[17070]: INF [Garden_and_Shed: 86000 - Capturing at 5.00 fps]
[... 207 lines cut ...]
Mar 13 05:00:48 zoneminder zma_m1[17082]: INF [Sunroom: 100154 - Closing event 24301, shutting down]
Mar 13 05:00:48 zoneminder zmdc[2248]: INF ['zma -m 1' exited normally]
Mar 13 05:00:48 zoneminder zmwatch[2408]: ERR [Error getting last capture time for  Back_Yard]
Mar 13 05:00:48 zoneminder zmwatch[2408]: INF [Restarting analysis daemon for  Back_Yard]
Mar 13 05:00:48 zoneminder zmdc[2248]: INF [Starting pending process, zma -m 1]
Mar 13 05:00:48 zoneminder zmdc[2248]: INF ['zma -m 1' starting at 17/03/13 05:00:48, pid = 18573]
Mar 13 05:00:48 zoneminder zmdc[18573]: INF ['zma -m 1' started at 17/03/13 05:00:48]
Mar 13 05:00:48 zoneminder zmdc[2248]: INF ['zma -m 2' sending stop to pid 2313 at 17/03/13 05:00:48]
Mar 13 05:00:48 zoneminder zma_m2[2313]: INF [Got signal 15 (Terminated), exiting]
Mar 13 05:00:48 zoneminder zma_m2[2313]: INF [Back_Yard: 677269 - Closing event 24302, shutting down]
Mar 13 05:00:49 zoneminder zma_m1[18573]: INF [In mode 4/1, warming up]
Mar 13 05:00:49 zoneminder zma_m1[18573]: INF [Sunroom: 026 - Opening new event 24305, section start]
Mar 13 05:00:49 zoneminder zmdc[2248]: INF ['zma -m 2' exited normally]
Mar 13 05:00:49 zoneminder zmc_m1[18554]: INF [Opening video storage stream events/1/17/03/13/05/00/48/24305-video.mp4 format: 1622512958#012]
Mar 13 05:00:49 zoneminder zmc_m1[18554]: INF [VideoStore startTime=8622454#012]
Mar 13 05:00:49 zoneminder zmdc[2248]: INF [Starting pending process, zma -m 2]
Mar 13 05:00:49 zoneminder zmdc[18575]: INF ['zma -m 2' started at 17/03/13 05:00:49]
Mar 13 05:00:49 zoneminder zmdc[2248]: INF ['zma -m 2' starting at 17/03/13 05:00:49, pid = 18575]
Mar 13 05:00:49 zoneminder zma_m2[18575]: INF [In mode 4/1, warming up]
Mar 13 05:00:49 zoneminder zma_m2[18575]: INF [Analysis fps (10.00) is lower than capturing fps (16.25), disabling adaptive skip feature]
Mar 13 05:00:52 zoneminder zma_m2[18575]: INF [Back_Yard: 026 - Opening new event 24306, section start]
Mar 13 05:00:52 zoneminder zmc_m2[18555]: INF [Opening video storage stream events/2/17/03/13/05/00/52/24306-video.mp4 format: 980415806#012]
Mar 13 05:00:52 zoneminder zmc_m2[18555]: INF [VideoStore startTime=8942166#012]
root@zoneminder:/var/log#
```

connortechnology, 2017-03-13:
I don't think this has anything to do with ZM.  Your system is literally changing it's time.

talkinggoat, 2017-03-13:
You are saying the time change in the logs, between Mar 13 04:35:53 and Mar 13 04:56:57 isn't an absence in the logs, but a jump forward in time? I could see how that would happen, although I don't know why it would loose 20 minutes, during the day, but why would ZM suddenly stop indexing time, correctly? For instance, from there, until 09:58, in the timeline for every camera and every event, time appears as 000:00/71571:41, instead of 00:00/10:00.

talkinggoat, 2017-03-15:
Same problem, last night, except I had it log the time and date, just before the time sync. There was virtually no change. ZMC stopped reporting in the log, at 04:40:44. at 05:03:06, systemd reported the time was changed, but there was no real change in the time; that is a normal output. It is at this time ZM stopped recording correctly and there is a gap in the log. ZMC doesn't start reporting again until 05:03:10, at which point, time is distorted on the timeline. I restarted the server at 09:00, but it didn't fix the erroneous time. At this point, my ZM h.264 install is non-functional.

SteveGilvarry, 2017-03-16:
@talkinggoat OK if we continue this in your forum thread, since h264 is still not mainline would rather talk there and keep issues for master bugs. Hopefully we can work out what is happening.

closed 2017-03-16 by SteveGilvarry