ZoneMinder/zoneminder #1821: Path options are not validated and block out entire webui leading to unrecoverable errors

rozza-m, 2017-03-18:
Using 1.29.0

The text fields for paths (images, etc) don't have any validation and are immediately accepted when submitted.

If these paths aren't valid, the webUI is blocked out with no means to recover (for example, if they've been left blank).

This would seem to have been solved with the following commit https://github.com/ZoneMinder/ZoneMinder/pull/1412 but the problem still occurs.

knnniggett, 2017-03-19:
1.29.0 was released on Feb 3, 2016.

#1412 did indeed fix the problem, but it wasn't merged until April 10, 2016, which is after 1.29.0 was released. You should upgrade to the latest version to receive the fix. If a newer version is not available in your distros repository then you can use the instructions in our readthedocs to build a package for your system.

closed 2017-03-19 by knnniggett

rozza-m, 2017-03-19:
My mistake then, sorry about that. Thanks for the quick response.